intel/intel-vaapi-driver #11: [BSW]GPU HANG: ecode 8:0:0x7f5f7f7f, in ba [916], reason: Ring hung, action: reset

seanvk, 2017-01-29:
migrated from Bugzilla [#98317](https://bugs.freedesktop.org/show_bug.cgi?id=98317)
status ASSIGNED severity *critical* in component *intel* for *---*
Reported in version *unspecified* on platform *x86-64 (AMD64)*
Assigned to: Pengfei

Original attachment names and IDs:
- _[bsw gpu hang error info.txt](https://bugs.freedesktop.org//attachment.cgi?id=127396)_ (ID 127396)

On 2016-10-19 01:25:40 +0000, william wrote:

> Created attachment 127396
> dump log from /sys/class/drm/card0/error
> 
> When using libva-stack from 01.org to encoding, occasionally get this GPU hang issue.
> Fully log from dmesg as below shows:
> [177082.427310] [drm] stuck on render ring 
> [177082.438483] [drm] GPU HANG: ecode 8:0:0x7f5f7f7f, in ba [916], reason: Ring hung, action: reset 
> [177082.448325] [drm] GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace. 
> [177082.458809] [drm] Please file a _new_ bug report on bugs.freedesktop.org against DRI -> DRM/Intel 
> [177082.468840] [drm] drm/i915 developers can then reassign to the right component if it's not a kernel issue. 
> [177082.479831] [drm] The gpu crash dump is required to analyze gpu hangs, so please always attach it. 
> [177082.489967] [drm] GPU crash dump saved to /sys/class/drm/card0/error 
> [177082.499497] drm/i915: Resetting chip after gpu hang 
> [177088.423173] [drm] stuck on render ring 
> [177088.434265] [drm] GPU HANG: ecode 8:0:0x7f5f7f7f, in ba [916], reason: Ring hung, action: reset 
> [177088.444257] [drm:i915_context_is_banned] *ERROR* gpu hanging too fast, banning! 
> [177088.454664] drm/i915: Resetting chip after gpu hang 
> 
> Linux Kernel version: 4.2.0

On 2016-10-19 13:41:26 +0000, haihao wrote:

> Someone reported GPU hang with high media workload on BSW and provided a workaround in the gfx mailing list. Could you give a try?
> 
> https://lists.freedesktop.org/archives/intel-gfx/2016-September/105710.html

On 2016-10-20 01:49:58 +0000, william wrote:

> (In reply to haihao from comment # 1)
> > Someone reported GPU hang with high media workload on BSW and provided a
> > workaround in the gfx mailing list. Could you give a try?
> > 
> > https://lists.freedesktop.org/archives/intel-gfx/2016-September/105710.html
> 
> Okay, I will let my customer to try it, thanks!

On 2016-11-04 03:41:16 +0000, haihao wrote:

> Does the fix in kernel work for you ?

On 2016-11-16 07:01:38 +0000, william wrote:

> (In reply to haihao from comment # 3)
> > Does the fix in kernel work for you ?
> 
> No, We have tried but have no effect.Is there any tools to debug this type(GPU hang) of issue?

On 2016-11-17 07:04:49 +0000, haihao wrote:

> Normally it is hard to identify the root cause for GPU hang issue without any details. 
> 
> You are using Kernel 4.2.0, could you try the latest rc kernel? If you can produce this issue with the latest rc kernel, please provide the steps to reproduce this issue and /sys/kernel/debug/dri/0/i915_error_state.

xhaihao, 2017-06-12:
BDW+ should use 48bit ppgtt. According to the information provided by the reporter, user was using Linux 4.2.0 which didn't enable 48bit ppgtt on BDW+.

xhaihao, 2017-06-12:
Closed the bug as 'not our bug'. Please feel free to reopen the bug if you still experience the issue after using a newer Linux kernel.

closed 2017-06-12 by xhaihao