intel/intel-vaapi-driver #3: [NM70][H264] h264 encoding does not works, return empty frames

seanvk, 2017-01-29:
migrated from Bugzilla [#99573](https://bugs.freedesktop.org/show_bug.cgi?id=99573)
status NEW severity *critical* in component *intel* for *---*
Reported in version *unspecified* on platform *x86-64 (AMD64)*
Assigned to: haihao

On 2017-01-28 01:45:54 +0000,  wrote:

> Hello, when i'm trying to encode something with ffmpeg it gets only empty frames.
> Full log: http://pastebin.com/5WtF5b3A
> [h264_vaapi @ 0x55fd7098cf80] Output buffer: 0 bytes (status 00000000). 
> 
> 
> 
> So, i tried to run h264encode from /test/encode of libva and it seem to have same problem:
> 
> Loading data into surface 15.....Complete surface loading
>       \00000059(000000 bytes coded)
> 
> Full log: http://pastebin.com/e1uhn848
> Also, drm.debug=0x06 does not shows anything special.
> Dmesg and vainfo output: http://pastebin.com/vwn1ufUv
> 
> My hardware: 
> IntelÂ® Dual-core CeleronÂ® 1037U
> Gigabyte GA-C1037UN-EU
> x86_64
> libva latest from Arch Linux (1.7.3)
> kernel 4.9.6-1-ARCH
> It is headless server.
> 
> Is there any way to get vaapi h264 encoding working?

xhaihao, 2017-02-03:
Intel® Dual-core Celeron® 1037U doesn't support encoding. I am asking the reporter provide the model name of 1037U and will provide a patch to disable encoding on 1037U.

Nalorokk, 2017-02-03:
cpu family	: 6
model		: 58
model name	: Intel(R) Celeron(R) CPU 1037U @ 1.80GHz

I hope that you are 100% sure that this CPU can't do encoding. It is pretty hard to find out what is capabilities of integrated GPU, from Intel Ark.

xhaihao, 2017-02-03:
Thanks for the info. Yes, 1037U doesn't support encoding.

xhaihao, 2017-02-03:
@Nalorokk,

Could you help to try with https://github.com/xhaihao/intel-vaapi-driver/commit/a437679ffbd00fa7381c3160ede06346ce9aff96?  I wonder whether the output of vainfo is right after applying this patch.

Nalorokk, 2017-02-03:
error: can't connect to X server!
libva info: VA-API version 0.39.4
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.39 (libva 1.7.3)
vainfo: Driver version: Intel i965 driver for Intel(R) Ivybridge Mobile - 1.8.0.pre1 (1.8.0.pre1)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD

xhaihao, 2017-02-04:
@Nalorokk 

Thanks for the info, so the patch takes effect. I will submit a pull request

xhaihao, 2017-02-06:
application or middle-ware should not do encoding anymore on 1037U

closed 2017-02-06 by xhaihao