intel/intel-vaapi-driver #5: Bus error (core dumped) 8192x8192 JPEG decode

seanvk, 2017-01-29:
migrated from Bugzilla [#98311](https://bugs.freedesktop.org/show_bug.cgi?id=98311)
status ASSIGNED severity *normal* in component *intel* for *---*
Reported in version *unspecified* on platform *Other*
Assigned to: haihao

Original attachment names and IDs:
- _[file_98311.txt](https://bugs.freedesktop.org//attachment.cgi?id=127383)_ (ID 127383)
- _[gdb-trace-with-drm-bufmgr-debug.txt](https://bugs.freedesktop.org//attachment.cgi?id=127384)_ (ID 127384)
- _[gdb-trace-additional.txt](https://bugs.freedesktop.org//attachment.cgi?id=127388)_ (ID 127388)
- _[attachment-31931-0.html](https://bugs.freedesktop.org//attachment.cgi?id=127549)_ (ID 127549)
- _[attachment-32256-0.html](https://bugs.freedesktop.org//attachment.cgi?id=127550)_ (ID 127550)

On 2016-10-18 18:56:01 +0000, U. Artie Eoff wrote:

> Decoding a 8192x8192 resolution JPEG image sometimes triggers a SIGBUS bus error (core dump).
> 
> This can be reproduced with `./test_i965_drv_video --gtest_filter=Big/JPEGEncodeInputTest.Full/2` (i.e. 8192x8192 UYVY test case)  sometimes.
> 
> Dmesg does not report any errors even with drm.debug=0xe kernel parameter.
> 
> See attachments for some gdb trace details.
> 
> Environment
> -----------
> SKL (Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz)
> Kernel 4.5.0-302.fc24.x86_64
> drm (master) libdrm-2.4.68-0-gfc09c5ab8424 
> mesa (master) heads/master-0-ga1e49be71360 
> libva (master) heads/master-0-g3b7e4999950a 
> intel-driver (master) heads/master-0-ge748bc7f0565

On 2016-10-18 18:57:00 +0000, U. Artie Eoff wrote:

> Comment on attachment 127383
> gdb trace w/drm_intel_bufmgr debug on
> 
> Starting program: /home/uartie/Work/workspace/media/build/intel-driver/test/test_i965_drv_video --gtest_filter=Big/JPEGEncodeInputTest.Full/2
> Missing separate debuginfos, use: dnf debuginfo-install glibc-2.23.1-5.fc24.x86_64
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib64/libthread_db.so.1".
> Note: Google Test filter = Big/JPEGEncodeInputTest.Full/2
> [==========] Running 1 test from 1 test case.
> [----------] Global test environment set-up.
> libva info: VA-API version 0.39.4
> libva info: va_getDriverName() returns 0
> libva info: User requested driver 'i965'
> libva info: Trying to open /home/uartie/Work/workspace/media/build/intel-driver/src/.libs/i965_drv_video.so
> libva info: Found init function __vaDriverInit_0_39
> bo_create: buf 1 (batch buffer) 524288b
> drm_intel_gem_bo_purge_vma_cache: cached=0, open=1, limit=-1
> bo_map: 1 (batch buffer), map_count=1
> bo_map: 1 (batch buffer) -> 0x7ffff7f51000
> bo_create: buf 2 (batch buffer) 524288b
> drm_intel_gem_bo_purge_vma_cache: cached=0, open=2, limit=-1
> bo_map: 2 (batch buffer), map_count=1
> bo_map: 2 (batch buffer) -> 0x7ffff7ed1000
> bo_create: buf 3 (kernel shader) 65328b
> drm_intel_gem_bo_purge_vma_cache: cached=0, open=3, limit=-1
> bo_map: 3 (kernel shader), map_count=1
> bo_map: 3 (kernel shader) -> 0x7ffff7fe6000
> drm_intel_gem_bo_purge_vma_cache: cached=1, open=2, limit=-1
> bo_create: buf 4 (kernel shader) 5696b
> drm_intel_gem_bo_purge_vma_cache: cached=1, open=3, limit=-1
> bo_map: 4 (kernel shader), map_count=1
> bo_map: 4 (kernel shader) -> 0x7ffff7fe4000
> drm_intel_gem_bo_purge_vma_cache: cached=2, open=2, limit=-1
> libva info: va_openDriver() returns 0
> [----------] 1 test from Big/JPEGEncodeInputTest
> [ RUN      ] Big/JPEGEncodeInputTest.Full/2
> bo_create: buf 5 (vaapi surface) 134217728b
[... 374 lines cut ...]
> > I'm trying 4.9.0-rc2+ (i.e. 9fe68cad6e74) right now, and so far I am not seeing
> > the issue with this version after ~100 executions.
> > 
> > You are receiving this mail because:
> > You are the QA Contact for the bug.

On 2016-10-26 08:52:49 +0000, haihao wrote:

> It is hard for us too find the root cause if we don't look into the i915. If we want to know the root cause, I think a better way is find a good commit and a bad commit, then do bisect. 
> 
> In addition, we can make sure it is not a driver issue if the test case works well with a recent kernel.

On 2016-10-26 15:42:43 +0000, U. Artie Eoff wrote:

> I continuously ran the test overnight with kernel.org 4.9.0-rc2+ (i.e. 9fe68cad6e74) and have not encountered this issue.
> 
> Yes, I agree.  We should understand the root-cause even if it's kernel related and not a driver issue.  That way, if this issue shows up again in the future, we can save time by investigating the direct cause.  So far, it appears kernel related.  Or maybe it's a compatibility issue with userspace libdrm and kernel version?
> 
> I am not sure what the chronological relationship is between drm-intel-nightly and kernel.org.  So if you're going to bisect i915, use kernel.org since there is a good and bad there.  Unfortunately, 4.8.4 is the latest mainstream stable release which Sean confirmed is bad.  And other distros are on slightly older versions.  So if we can understand the cause, perhaps we can come up with a driver solution to WA it for now.

On 2016-11-02 17:32:44 +0000, Sean V Kelley wrote:

> Yes, the most important thing is that we need to be able to isolate the root cause as we can hardly promote the framework if we don't know the cause for the errors.  It is good to bisect the kernel and map that to libdrm versions.  I get very wary of using non stable kernels in this testing.  Try to start from the Quarterly release BKC.
> 
> Sean

wangzj0601, 2017-06-29:
This issue can not be duplicated on SKL platform with intel-vaapi-driver 1.8.3.pre1, try 20 times
enc:
platform : SKL
linux kernel: 4.12.0-rc2
Libva :1.8.3.pre1
Libva- intel-driver :1.8.3.pre1
libva-utils : 1.8.3.pre1
Typical invocation:
for i in `seq 1 20`; do ./test_i965_drv_video --gtest_filter=Big/JPEGEncodeInputTest.Full/2; echo "$i PASS" >> zjresult ; done

FocusLuo, 2017-07-12:
Based on wangzj0601 comment, this issue can not be duplicated, so closed it

closed 2017-07-12 by FocusLuo