intel/intel-vaapi-driver #40: Rework VP8 encoder with new VP8 shaders on SKL/APL/KBL

xhaihao, 2017-02-10:
We have the new VP8 shaders for SKL/APL/KBL, need to rework VP8 encoder on SKL/APL/KBL

seanvk, 2017-02-22:
@xhaihao You forgot to add to the commit header Fixes #n  so these bugs are still hanging about.

closed 2017-02-22 by seanvk

xhaihao, 2017-02-23:
@seanvk I'd like to add "Fixes <issue URL>" instead of "Fixes #n" in the commit, github will change the URL to #n automatically in the github environment but keep the commit message unchanged.

E.g. in commit 71f77c6a4b757576edb22642cb9baf1cc9792453
This fixes https://github.com/01org/intel-vaapi-driver/issues/46

The issue of 46 was also closed automatically 

I think a URL is better than a number for a reader and it is easy to reach the issue in a browser.

It was my fault not to provide a whole URL in the commits for vp8-rework

seanvk, 2017-02-23:
>    Reverse link: [6]unknown
>
> References
>
>    Visible links
>    1. https://github.com/seanvk
>    2. https://github.com/01org/intel-vaapi-driver/commit/71f77c6a4b757576edb22642cb9baf1cc9792453
>    3. https://github.com/01org/intel-vaapi-driver/issues/46
>    4. https://github.com/01org/intel-vaapi-driver/issues/40#issuecomment-281855118

On Wednesday 22 Feb 2017 at 16:54:17 (-0800), Xiang, Haihao writes :

>    [1]@seanvk I'd like to add "Fixes " instead of "Fixes #n" in the commit,
>    github will change the URL to #n automatically in the github environment
>    but keep the commit message unchanged.
> 
>    E.g. in commit [2]71f77c6
>    This fixes [3]#46
> 
>    The issue of 46 was also closed automatically
> 
>    I think a URL is better than a number for a reader and it is easy to reach
>    the issue in a browser.
> 
>    It was my fault not to provide a whole URL in the commits for vp8-rework

Haihao,

There is always a record.  Please do both if you want to include the URL.  

Fixes #n is standard practice with Github and if you want to include the URL
you can.  This is no different than Google Chrome.  We all know it is a Github
project and the # can easily be found from the Github project.

Thanks,

Sean

seanvk, 2017-02-23:
And @xhaihao if you use Github, the link will always be in the commit view from github:

For example:
https://github.com/01org/intel-vaapi-driver/commit/03a86fc527febd7a2f2d52d3b46fee2c5ccb641b

So please continue to include Fixes #n  You can add a URL if you want but that is needless duplication of a Github feature.

xhaihao, 2017-02-23:
Yes, it is easy in github. but for a reader who is not using github,  what does #n mean?

seanvk, 2017-02-23:
>    Reverse link: [3]unknown
>
> References
>
>    Visible links
>    1. https://github.com/01org/intel-vaapi-driver/issues/40#issuecomment-281862708

On Wednesday 22 Feb 2017 at 17:35:26 (-0800), Xiang, Haihao writes :

>    Yes, it is easy in github. but for a reader who is not using github, what
>    does #n mean?

The reader has to get the code from github.  Github is where the issue number
is tracked.  They can look up the bug by the number in the commit.  It's
absolutely no different than Chrome or even Bugzilla numbers.  It also allows
us to have a standard flow for issue resolution.

Thanks,

Sean

xhaihao, 2017-02-23:
@seanvk  I agree we should have a standard flow for issue resolution, actually github supports "Fixes <URL>".

My concern is  a number in the commit log is meaningless. We moved from FDO to github, maybe we will move from github to another place in the future.  A reader doesn't know what is the number when he go through the whole git log.

seanvk, 2017-02-23:
This is no different than having a bugzilla number or a chrome bug number.  They can look up where they got the source from.  If you insist its trivial to just say Fixes 01org/intel-vaapi-driver#n or Fixes #n. Either will work.

Thanks,

Sean