intel/intel-vaapi-driver #41: H.265 Main 10 profile encoder aborts when given non-Y-tiled P010 surfaces

fhvwy, 2017-02-11:
On Kaby Lake:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
#1  0x00007ffff5ef940a in __GI_abort () at abort.c:89
#2  0x00007ffff5ef0e47 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x7ffff48e917e "hevc_encoder_surface", file=file@entry=0x7ffff48e9082 "../../vanilla/src/gen9_vme.c", line=line@entry=1433, function=function@entry=0x7ffff48e93d0 <__PRETTY_FUNCTION__.14053> "gen9_vme_hevc_surface_setup")
    at assert.c:92
#3  0x00007ffff5ef0ef2 in __GI___assert_fail (assertion=0x7ffff48e917e "hevc_encoder_surface", file=0x7ffff48e9082 "../../vanilla/src/gen9_vme.c", line=1433, function=0x7ffff48e93d0 <__PRETTY_FUNCTION__.14053> "gen9_vme_hevc_surface_setup") at assert.c:101
#4  0x00007ffff4834733 in gen9_vme_hevc_surface_setup (ctx=0x555556ae75e0, encode_state=0x555556afc298, is_intra=1, encoder_context=0x555557a703a0) at ../../vanilla/src/gen9_vme.c:1433
#5  0x00007ffff4835a07 in gen9_vme_hevc_prepare (ctx=0x555556ae75e0, encode_state=0x555556afc298, encoder_context=0x555557a703a0) at ../../vanilla/src/gen9_vme.c:1914
#6  0x00007ffff4835a94 in gen9_vme_hevc_pipeline (ctx=0x555556ae75e0, profile=VAProfileHEVCMain10, encode_state=0x555556afc298, encoder_context=0x555557a703a0) at ../../vanilla/src/gen9_vme.c:1933
#7  0x00007ffff4880320 in intel_encoder_end_picture (ctx=0x555556ae75e0, profile=VAProfileHEVCMain10, codec_state=0x555556afc298, hw_context=0x555557a703a0) at ../../vanilla/src/i965_encoder.c:1308
#8  0x00007ffff487252d in i965_EndPicture (ctx=0x555556ae75e0, context=33554434) at ../../vanilla/src/i965_drv_video.c:3588
#9  0x00007ffff7bb74ca in vaEndPicture (dpy=0x555556ae7500, context=33554434) at ../../vanilla/va/va.c:1290
```

Other related combinations do work:
* Y-tiled P010 surfaces, encoding to Main 10 profile.
* Non-Y-tiled P010 surfaces, encoding to Main profile (with downsampling).
* Non-Y-tiled NV12 surfaces, encoding to Main profile.

seanvk, 2017-02-12:
Are you testing this with avconv @fhvwy, if so can you share the command line command you are using to reproduce it?

fhvwy, 2017-02-12:
Technically yes, but I'm not sure how helpful it will be because it has incomplete further development on top of current libav.  The P010 surfaces which I am using here are coming from Beignet, and being mapped to VAAPI via clGetMemObjectFdIntel + vaCreateSurfaces with DRM PRIME fds.

If you really want to follow my reproduction steps, then:
* Build and install Beignet with this patch to enable P010 support: https://lists.freedesktop.org/archives/beignet/2017-February/008566.html .
* Get this libav branch: https://github.com/fhvwy/libav/tree/device (in-progress and still has many rough edges).
* Build with --enable-vaapi --enable-opencl.
* Make an OpenCL kernel which writes an output image (e.g. see below).
* Get a 10-bit H.265 input video (e.g. http://jell.yfish.us/ ).
* Run avconv as below for various different cases.

Non-Y-tiled P010 surfaces, encode to Main 10 (aborts):
`./avconv -y -threads 1 -init_hw_device vaapi=/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -vf 'hwmap=derive_device=opencl,opencl_program=test.cl:copy,hwmap=derive_device=vaapi' -c:v hevc_vaapi -profile 2 out.mp4`

Non-Y-tiled P010 surfaces, encode to Main:
`./avconv -y -threads 1 -init_hw_device vaapi=/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -vf 'hwmap=derive_device=opencl,opencl_program=test.cl:copy,hwmap=derive_device=vaapi' -c:v hevc_vaapi -profile 1 out.mp4`

Non-Y-tiled NV12 surfaces, encode to Main:
`./avconv -y -threads 1 -init_hw_device vaapi=/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -vf 'scale_vaapi=1280:720:nv12,hwmap=derive_device=opencl,opencl_program=test.cl:copy,hwmap=derive_device=vaapi' -c:v hevc_vaapi -profile 1 out.mp4`

Y-tiled P010 surfaces, encode to Main 10 (no OpenCL involved):
`./avconv -y -threads 1 -init_hw_device vaapi=/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -vf 'scale_vaapi=1280:720:p010' -c:v hevc_vaapi -profile 2 out.mp4`

A trivial kernel which copies the input:
```
__kernel void copy(__write_only image2d_t dst,
                   __read_only  image2d_t src,
                   unsigned int index)
{
  int2 location = (int2)(get_global_id(0), get_global_id(1));
  write_imagef(dst, location, read_imagef(src, location));
}
```

xhaihao, 2017-02-13:
It should be encode_state->input_yuv_object instead of encode_state->current_render_target  when checking the input YUV surface in the driver.

pengche1, 2017-02-14:
[0001-Fix-the-linear-P010-aborts-bug-for-hevc-main10-encod.txt](https://github.com/01org/intel-vaapi-driver/files/772866/0001-Fix-the-linear-P010-aborts-bug-for-hevc-main10-encod.txt)
@fhvwy, could you help me test this patch? It works for me. but you scenerio is more complicated and please double check. thanks!

xhaihao, 2017-02-14:
@pengche1, could you also take https://github.com/01org/intel-vaapi-driver/issues/42 into account? It seems #42 is also caused by a bug in H265 encoder.

fhvwy, 2017-02-14:
@pengche1  Works for me with the setup above (no longer aborts, output is correct).  Thank you!

yakuizhao, 2017-02-14:
@fhvwy . Currently the driver will help to do the conversion implicitly for some input surface(For example: non-tiled NV12, I420 and so on). But I don't think that it is a good idea.  If one shared non-tiled surface is used as the reconstructed surface, how should we handle it?
It will be better that the explicit conversion is used instead.  In such case it helps the user to choose the better input format for the performance optimization purpose.

fhvwy, 2017-02-14:
@yakuizhao  For reconstructed surfaces I definitely agree - I can't think of any reasonable use case for wanting to supply externally-allocated shared surfaces there.  For input surfaces, that's not true.  The problem, then, is that VAAPI is careful to make this sort of detail (tiling mode) opaque, such that users don't have any outward visibility at all as to what is going on and the driver transparently handles any conversion (with possible performance penalty).

So: I'm not necessarily against forcing the user to do the conversion explicitly, but if such a change is going to be made then there needs to be something added to VAAPI so that the user can distinguish between surfaces which currently appear to be identical (e.g. all NV12 of the same size) but only some of which will actually work (because only some of them are Y-tiled).

Relatedly, is there a table somewhere showing the actual hardware restrictions on what surface formats / tiling modes can be used with each encoder / decoder / other processing element?

seanvk, 2017-02-15:
No we don't explicitly describe the memory organization of the surfaces or the respective color format requirements.  At some point we can add that to the Wiki documentation, otherwise it's in the code and the PRM on 01.org, but not entirely easy to discern.

commit 3aa16e3e (2017-02-16): Fix the linear P010 aborts bug for hevc main10 encoding.

Fixes #41.
closed 2017-02-16 by xhaihao