intel/intel-vaapi-driver #42: H.265 encoder crashes when given surfaces straight from H.264 decoder

fhvwy, 2017-02-12:
With current libav:
`./avconv -y -threads 1 -vaapi_device /dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -c:v hevc_vaapi out.mp4`
crashes with heap corruption.

valgrind says:
```
==9218== Invalid write of size 4
==9218==    at 0x840C7C7: gen9_intel_hevc_input_check (gen9_vme.c:1859)
==9218==    by 0x840C9C8: gen9_vme_hevc_prepare (gen9_vme.c:1909)
==9218==    by 0x840CA83: gen9_vme_hevc_pipeline (gen9_vme.c:1933)
==9218==    by 0x84573D0: intel_encoder_end_picture (i965_encoder.c:1308)
==9218==    by 0x8449562: i965_EndPicture (i965_drv_video.c:3588)
==9218==    by 0x53804C9: vaEndPicture (va.c:1286)
==9218==    by 0x7E9DBF: vaapi_encode_issue (vaapi_encode.c:388)
==9218==    by 0x7EA032: vaapi_encode_step (vaapi_encode.c:609)
==9218==    by 0x7EA331: ff_vaapi_encode2 (vaapi_encode.c:896)
==9218==    by 0x345682: avcodec_encode_video2 (encode.c:231)
==9218==    by 0x345821: do_encode (encode.c:278)
==9218==    by 0x34599B: avcodec_send_frame (encode.c:327)
==9218==  Address 0x9f16448 is 8 bytes after a block of size 32 alloc'd
==9218==    at 0x4C2CBC5: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==9218==    by 0x83FDBAB: gen8_mfd_init_avc_surface (gen8_mfd.c:76)
==9218==    by 0x8400185: gen8_mfd_avc_decode_init (gen8_mfd.c:853)
==9218==    by 0x84004AF: gen8_mfd_avc_decode_picture (gen8_mfd.c:917)
==9218==    by 0x8405AE6: gen8_mfd_decode_picture (gen8_mfd.c:3105)
==9218==    by 0x8449562: i965_EndPicture (i965_drv_video.c:3588)
==9218==    by 0x53804C9: vaEndPicture (va.c:1286)
==9218==    by 0x7E83D9: ff_vaapi_decode_issue (vaapi_decode.c:185)
==9218==    by 0x5ADD60: vaapi_h264_end_frame (vaapi_h264.c:320)
==9218==    by 0x757E70: ff_h264_field_end (h264_picture.c:166)
==9218==    by 0x39D60A: h264_decode_frame (h264dec.c:744)
==9218==    by 0x31C252: decode_simple_internal (decode.c:335)
==9218==    by 0x31C252: decode_simple_receive_frame (decode.c:391)
==9218==    by 0x31C252: decode_receive_frame_internal (decode.c:409)
```

The struct object_surface private_data field seems to be inappropriately shared.  The H.264 decoder allocates it as a GenAvcSurface (32 bytes), but the H.265 encoder is treating it as if it were a GenHevcSurface (48 bytes) and therefore encounters issues when it tries to write the later fields.

(It also happens in some other combinations, e.g. VC-1 decode -> H.265 encode; I haven't studied every case but it looks like the same issue.)

xhaihao, 2017-02-13:
We are aware of this issue and will investigate a solutions.
>  xhaihao 3 days ago Member

It will cause critical issue if releasing the private data which is using in another context. e.g. the dmv buffers in AVC decoding context. Here I don't think releasing private data here is a good idea. We need a rework for private data usage.
@seanvk
seanvk 3 days ago Member

Yes, I agree. Let's file a bug @xhaihao on reworking handling of private data usage for all codecs and consider the contexts. We can leave this as is and rework for separate bug.

xhaihao, 2017-02-14:
I can reproduced this issue with yamitranscode (you need to define DISABLE_SCALE in tests/yamitranscode.cpp) but only when transcoding AVC/VC-1 to H.265, this is because H.265 encoder writes some data in the private data of the input surface.

pengche1, 2017-02-15:
@fhvwy, please help me to check this patch. if any problem, just tell me. thanks!

fhvwy, 2017-02-15:
@pengche1  Tested with avconv, LGTM.  Thank you!

PR #53 (2017-02-16): Move the down-sampled NV12 surface handlers into the reconstructed su…
…rface's private data from the input surface's for hevc encoder

Fixes #42.
commit 7dbd8efe (2017-02-16): Move the down-sampled NV12 surface handlers into the reconstructed surface's private data from the input surface's for hevc encoder

Fixes #42.
closed 2017-02-16 by xhaihao