intel/intel-vaapi-driver #43: AVC encoder quality improvement on SKL/APL

QuPengfei, 2017-02-14:
This will improve the AVC encoder quality on SKL and APL

seanvk, 2017-02-22:
Hi, you just need to use the phrase:  "Fixes #n"  Note the spelling and where n is 43.  You have some weird things like "Fix #n" and 01org#n.

seanvk, 2017-02-22:
@QuPengfei In your recent patch update, where are the actual kernel files?  I don't see them.  All I see is the header with the extern declaration.

QuPengfei, 2017-02-23:
@Sean the kernel binary file in this patch "ENC: add AVC kernel binary on SKL", include .h/c. it looks too large.

here is the info:
12,085 12,085 additions & 0 deletions  src/gen9_avc_encoder_kernels.c
12,085 additions, 0 deletions not shown because the diff is too large. Please use a local Git client to view these changes.

QuPengfei, 2017-02-23:
yes, I will correct the spelling. at the same time 01org#n is the issue link, I will remove it.

seanvk, 2017-02-23:
Ah, I will pull the binaries directly.  I just wanted to make sure the actual kernel files with the binaries were using an Apache 2 header...

commit 07577a73 (2017-03-02): ENC: add AVC kernel binary on SKL

v1:move kernal array into the c file
v2:add file into Makefile.am
v3:use the Apache 2 license header for media kernel binary

Fixes #43

Reviewed-by: Sean V Kelley<seanvk@posteo.de>
commit 35a5575a (2017-03-02): ENC: add kernel related structure and define for AVC

Fixes #43

Reviewed-by: Sean V Kelley<seanvk@posteo.de>
commit 481ec15c (2017-03-02): ENC:support more quality level and switch to new AVC encoder solution on SKL/APL

v1:
add AVC encoder support on APL

Fixes #43

Reviewed-by: Sean V Kelley<seanvk@posteo.de>
commit 4bd03421 (2017-03-02): ENC: add VME pipeline for AVC encoder

VME pipeline:
add resource and surface allocation and free function
add init table for frame mbbrc update
add scaling kernel for AVC encoder
add BRC init reset kernel for AVC RC logic
add BRC frame update-kernel for AVC RC logic
add BRC MB level update kernel for AVC RC logic
add REF frame QA caculation and MB level const data
add MBENC kernel for AVC encoder
add ME kernel for AVC encoder
add WP/SFD kernel for AVC encoder
add kernel init/destroy function for AVC encoder
add kernel related parameter check function for AVC
add VME pipeline init prepare/run  [... cut]
closed 2017-03-02 by xhaihao