intel/intel-vaapi-driver #48: Properly check for X11 backend to avoid compilation break

chivakker, 2017-02-14:
configuring should use pkg-config to know if X11 headers (distributions call this devel packages) are available to the path and error out.

Currently configure won't check for it and as a result in the absence of X11 headers the build breaks when compiling vi_dricommon.h, which is included by va_backend_compat.h.  Compilation relies on a successful configure stage.

yakuizhao, 2017-02-15:
I  don't think that it is a bug. Instead it is related with the system configuration.
In fact if the libva is built without x11 supported, it will disable libva-x11. Then the intel-vaapi-driver can be built with X11 disabled automatically.
But the building system has the libva-x11.pc. In such case it causes that USE_X11 is still configured as 1.
Then the error is triggered.

But it is not the driver bug.

chivakker, 2017-02-15:
The point here is which library needs x11 to compile.  If it is needed to compile then configure should be doing the check.  libva was doing it because the test folder required it, so in that scenario all libraries were covered.  Once libva-utils is introduced, the expectation is that it should be compiled in this order libva, intel-vaapi-driver and the libva-utils.  libva doesn't need to check for x11 stuff anymore, so the driver and libva-utils should do it as they actually need it for compilation.

yakuizhao, 2017-02-16:
The driver has the dependency on libva-x11. If the libva is built without libva-x11,  the x11 is automatically disabled when trying to build the driver.(Although x11 is enabled for libva by default, it will also be disabled if the X11 library doesn't meet with the requirement.)

So if the libva has no libva-x11 library, the mentioned issue in driver is related with the system configuration. 

BTW: libva-x11 is not only required by libva tests,  but also is required by the windows system when running the libva.

xhaihao, 2017-02-16:
@yakuizhao Even though user installs libva & libva-x11, it is possible not to install the develop package for X11, however we need the develop package fox X11 (e.g. libx11-dev on Ubuntu) to build the driver.

yakuizhao, 2017-02-16:
@chivakker @xhaihao Thanks for the more detailed explanation. Sorry that I am wrong. If the libva is installed by the distribution package with libva-x11, the driver building error will be triggered if X11 library doesn't exist.

PR #49 (2017-02-23): configure: check for x11 on path
enable-x11 is set to yes by default and during configuration it has to
check for its header files on the path before breaking the compilation
if those are not present

va_dricommon.h requires x11 on path to compile

Fixes #48
commit 03a86fc5 (2017-02-23): configure: check for x11 on path

enable-x11 is set to yes by default and during configuration it has to
check for X11 header files on the path then the build system is prepared
to avoid compiling X11 related code if headers are not present.

Fixes #48

TEST="autogen.sh --prefix=<some prefix>, make sure X11 headers are not installed"
TEST="x11 should not print out as supported on the configure summary"
closed 2017-02-23 by seanvk