intel/intel-vaapi-driver #56: VDENC: incorrect buffer domain for Downscaled reconstruct buffer

yakuizhao, 2017-02-17:
It should be configured as write. But now it is marked as "READ_DOMAIN".
In fact as it is still treated as R/W,  there is no issue.
But it will be better to fix it.

seanvk, 2017-02-18:
Note that this was already merged but the commit header was wrong so the webhook did not pick up the change and so this bug is still open.  SO I will close it now.

closed 2017-02-18 by seanvk

xhaihao, 2017-02-20:
Sorry it was my fault. I didn't notice the issue number #45 doesn't match the issue URL in the message. 

    Fix the issue #45
    https://github.com/01org/intel-vaapi-driver/issues/44

The bug would be closed automatically if the message was

    Fix the issue #45
    https://github.com/01org/intel-vaapi-driver/issues/45

BTW I think an issue URL in the commit message is more meaningful than an issue number.