intel/intel-vaapi-driver #58: Convince me you are using Android.mk

seanvk, 2017-02-18:
Because we are not testing Android build with it and yet the makefile is still there and we are rarely adding files to it...I'd honestly prefer to see that maintained at the packaging level of an android project.  Anyway, why can't Android just cooperate and not require their own Makefile?

Begin

xhaihao, 2017-02-20:
Yes Android Makefiles are out of date now. In the past, someone used this driver on Android and it worked fine. 

I think Android is a little different from other Linux OS distributions. Other projects such as mesa, libdrm also include android Makefiles.

seanvk, 2017-02-22:
I think you are right @xhaihao .  However, since we are not actively building it, then it is always in a state of uncertainty.  In the future @xuguangxin, will you be doing Android test builds say for YunOS?  Just trying to determine if we are actually going to be maintaining it.

evelikov, 2017-02-26:
@seanvk afaict it's mostly ok - barring a few small cleanups.

Personally, I suggest reaching out to the Android-ia/Android-x86 people - most Android (and Linux as well) distros are under the impression that upstream "does not give a sh*t" , so they keep quiet. Even when they have good bugfixes/other.

By doing so you might even get some extra contributors to help with the mundane cleanups, that the rest of the team tends to lack time for ;-)

cwhuang, 2017-04-07:
I've sent a pull request to fix Android building issue.
Please review: https://github.com/01org/intel-vaapi-driver/pull/126

xhaihao, 2017-05-11:
Closed as the building issue was fixed by @cwhuang

closed 2017-05-11 by xhaihao