intel/intel-vaapi-driver #72: Unittest Regression: AVCEncode/AVCEContextTest.QualityRange test cases

uartie, 2017-03-02:
PR #33 caused the `AVCEncode/AVCEContextTest.QualityRange` unittests to regress in the `test_i965_drv_video` test suite.  Based on the nature of the change, this only appears to be an issue with the test cases being out of date now.  That is, the default quality ranges for AVCE were changed in PR #33 but the test cases were not updated to reflect the new defaults.  The good thing is that the test cases reveal that the new changes in the driver are working... just that the expected values are no longer correct for the tests.  Please update these tests to fix this.

```
$ ./test/test_i965_drv_video --gtest_filter=AVCEncode/AVCEContextTest.QualityRange*

Note: Google Test filter = AVCEncode/AVCEContextTest.QualityRange*
[==========] Running 8 tests from 1 test case.
[----------] Global test environment set-up.
Seeded std::rand() with 1488480252.
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /opt/media/build/intel-vaapi-driver/src/.libs/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
[----------] 8 tests from AVCEncode/AVCEContextTest
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/0
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 2
To be equal to: hw_context->quality_range
      Which is: 8
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/0, where GetParam() = (VAProfileH264ConstrainedBaseline, VAEntrypointEncSlice) (0 ms)
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/1
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 8
To be equal to: hw_context->quality_range
      Which is: 2
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/1, where GetParam() = (VAProfileH264ConstrainedBaseline, VAEntrypointEncSliceLP) (0 ms)
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/2
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 2
To be equal to: hw_context->quality_range
      Which is: 8
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/2, where GetParam() = (VAProfileH264Main, VAEntrypointEncSlice) (1 ms)
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/3
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 8
To be equal to: hw_context->quality_range
      Which is: 2
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/3, where GetParam() = (VAProfileH264Main, VAEntrypointEncSliceLP) (0 ms)
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/4
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 2
To be equal to: hw_context->quality_range
      Which is: 8
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/4, where GetParam() = (VAProfileH264High, VAEntrypointEncSlice) (0 ms)
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/5
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 8
To be equal to: hw_context->quality_range
      Which is: 2
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/5, where GetParam() = (VAProfileH264High, VAEntrypointEncSliceLP) (0 ms)
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/6
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 1
To be equal to: hw_context->quality_range
      Which is: 8
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/6, where GetParam() = (VAProfileH264MultiviewHigh, VAEntrypointEncSlice) (1 ms)
[ RUN      ] AVCEncode/AVCEContextTest.QualityRange/7
i965_avce_context_test.cpp:240: Failure
      Expected: qranges.at(profile)
      Which is: 1
To be equal to: hw_context->quality_range
      Which is: 8
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/7, where GetParam() = (VAProfileH264StereoHigh, VAEntrypointEncSlice) (0 ms)
[----------] 8 tests from AVCEncode/AVCEContextTest (2 ms total)

[----------] Global test environment tear-down
[==========] 8 tests from 1 test case ran. (19 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 8 tests, listed below:
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/0, where GetParam() = (VAProfileH264ConstrainedBaseline, VAEntrypointEncSlice)
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/1, where GetParam() = (VAProfileH264ConstrainedBaseline, VAEntrypointEncSliceLP)
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/2, where GetParam() = (VAProfileH264Main, VAEntrypointEncSlice)
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/3, where GetParam() = (VAProfileH264Main, VAEntrypointEncSliceLP)
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/4, where GetParam() = (VAProfileH264High, VAEntrypointEncSlice)
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/5, where GetParam() = (VAProfileH264High, VAEntrypointEncSliceLP)
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/6, where GetParam() = (VAProfileH264MultiviewHigh, VAEntrypointEncSlice)
[  FAILED  ] AVCEncode/AVCEContextTest.QualityRange/7, where GetParam() = (VAProfileH264StereoHigh, VAEntrypointEncSlice)
```

QuPengfei, 2017-03-03:
Pull request is on going

QuPengfei, 2017-03-03:
with the each platform adding for AVC encoder， here still need the changes.

PR #75 (2017-03-06): Fix the quality range regression issue
Fixed #72

Fix the quality range regression issue
commit a7517133 (2017-03-06): Fix the quality range regression issue

v1:
use bool to define the member
define the one map data for quality range according to the platform type

Fixed #72

For low quali# Please enter the commit message for your changes. Lines starting
closed 2017-03-06 by uartie