intel/intel-vaapi-driver #76: Add support for NetBSD (and maybe other OSes without <alloca.h>)

iamleot, 2017-03-06:
intel-vaapi-driver does not build on NetBSD due to non-existent
<alloca.h> (alloca(3) is in <stdlib.h>).
Using `AC_FUNC_ALLOCA` macro in `configure.ac` and adjusting
`src/i965_decoder_utils.c` accordingly fixes that.

A possible patch is attached in this issue (please let me know if I have missed
something!).

Thank you very much for the attention!

[0001-configure-check-for-AC_FUNC_ALLOCA.txt](https://github.com/01org/intel-vaapi-driver/files/822437/0001-configure-check-for-AC_FUNC_ALLOCA.txt)

chivakker, 2017-03-06:
Please submit your patch as a pull request to get it reviewed and approved.

iamleot, 2017-03-06:
Hello Daniel,

Daniel Charles writes:
> Please submit your patch as a pull request to get it reviewed and approved. 

Whoops, sorry for the noise. I have just submitted it as pull
request #77.

Thank you!

closed 2017-03-09 by xhaihao

iamleot, 2017-03-09:
Hello Xiang,

"Xiang, Haihao" writes:
> Closed #76.
> 

Thank you very much!