intel/intel-vaapi-driver #87: VP9 encode rate control hangs GPU if framerate is 127 or higher

fhvwy, 2017-03-13:
```
Thread 1 "avconv" hit Breakpoint 1, gen9_vp9_set_curbe_brc (ctx=0x555556aee540, encode_state=0x555556b02bb0, gpe_context=0x555556eb4fd0, encoder_context=0x555556eb1850, param=0x7fffffffd040) at ../../vanilla/src/gen9_vp9_encoder.c:1208
1208        i965_gpe_context_unmap_curbe(gpe_context);
(gdb) p *cmd
$1 = {dw0 = {frame_width = 1920, frame_height = 1080}, dw1 = {frame_type = 0, segmentation_enable = 0, ref_frame_flags = 0, num_tlevels = 1}, dw2 = {reserved = 0, intra_mode_disable = 0, loop_filter_type = 0}, dw3 = {max_level_ratiot0 = 0, max_level_ratiot1 = 0, max_level_ratiot2 = 0, max_level_ratiot3 = 0}, dw4 = {
    profile_level_max_frame = 2073600}, dw5 = {init_buf_fullness = 750000}, dw6 = {buf_size = 1000000}, dw7 = {target_bit_rate = 1000000}, dw8 = {max_bit_rate = 1000000}, dw9 = {min_bit_rate = 0}, dw10 = {frame_ratem = 126}, dw11 = {frame_rated = 1}, dw12 = {brc_flag = 16, gopp = 249}, dw13 = {
    init_frame_width = 1920, init_frame_height = 1080}, dw14 = {avbr_accuracy = 30, avbr_convergence = 150}, dw15 = {min_qp = 0, max_qp = 255}, dw16 = {cq_level = 30, reserved = 0}, dw17 = {enable_dynamic_scaling = 0, brc_overshoot_cbr_pct = 150}, dw18 = {pframe_deviation_threshold0 = 208, 
    pframe_deviation_threshold1 = 211, pframe_deviation_threshold2 = 215, pframe_deviation_threshold3 = 219}, dw19 = {pframe_deviation_threshold4 = 37, pframe_deviation_threshold5 = 41, pframe_deviation_threshold6 = 45, pframe_deviation_threshold7 = 48}, dw20 = {vbr_deviation_threshold0 = 208, 
    vbr_deviation_threshold1 = 211, vbr_deviation_threshold2 = 214, vbr_deviation_threshold3 = 219}, dw21 = {vbr_deviation_threshold4 = 80, vbr_deviation_threshold5 = 84, vbr_deviation_threshold6 = 93, vbr_deviation_threshold7 = 97}, dw22 = {kframe_deviation_threshold0 = 209, kframe_deviation_threshold1 = 212, 
    kframe_deviation_threshold2 = 218, kframe_deviation_threshold3 = 222}, dw23 = {kframe_deviation_threshold4 = 34, kframe_deviation_threshold5 = 40, kframe_deviation_threshold6 = 45, kframe_deviation_threshold7 = 48}, dw24 = {target_size = 0}, dw25 = {frame_number = 0}, dw26 = {reserved = 0}, dw27 = {
    hrd_buffer_fullness_upper_limit = 0}, dw28 = {hrd_buffer_fullness_lower_limit = 0}, dw29 = {reserved = 0}, dw30 = {reserved = 0}, dw31 = {reserved = 0}, dw32 = {seg_delta_qp0 = 0, seg_delta_qp1 = 0, seg_delta_qp2 = 0, seg_delta_qp3 = 0}, dw33 = {seg_delta_qp4 = 0, seg_delta_qp5 = 0, seg_delta_qp6 = 0, 
    seg_delta_qp7 = 0}, dw34 = {temporal_id = 0, multi_ref_qp_check = 0, reserved = 0}, dw35 = {max_num_pak_passes = 0, sync_async = 0, overflow = 0, mbrc = 0}, dw36 = {reserved = 0, segmentation = 0, reserved1 = 0}, dw37 = {cur_qpindex = 0, lastref_qpindex = 0, goldref_qpindex = 0, altref_qpindex = 0}, dw38 = {
    qdelta_ydc = 0, qdelta_uvac = 0, qdelta_uvdc = 0, reserved = 0}, dw39 = {reserved = 0}, dw40 = {reserved = 0}, dw41 = {reserved = 0}, dw42 = {reserved = 0}, dw43 = {reserved = 0}, dw44 = {reserved = 0}, dw45 = {reserved = 0}, dw46 = {reserved = 0}, dw47 = {reserved = 0}, dw48 = {brc_y4x_input_bti = 0}, dw49 = {
    brc_vme_coarse_intra_input_bti = 1}, dw50 = {brc_history_buffer_bti = 2}, dw51 = {brc_const_data_input_bti = 3}, dw52 = {brc_distortion_bti = 4}, dw53 = {brc_mmdk_pak_output_bti = 5}, dw54 = {brc_enccurbe_input_bti = 6}, dw55 = {brc_enccurbe_output_bti = 7}, dw56 = {brc_pic_state_input_bti = 8}, dw57 = {
    brc_pic_state_output_bti = 9}, dw58 = {brc_seg_state_input_bti = 10}, dw59 = {brc_seg_state_output_bti = 11}, dw60 = {brc_bitstream_size_data_bti = 12}, dw61 = {brc_hfw_data_output_bti = 13}, dw62 = {reserved = 0}, dw63 = {reserved = 0}}
```
-> works.

```
Thread 1 "avconv" hit Breakpoint 1, gen9_vp9_set_curbe_brc (ctx=0x555556aee540, encode_state=0x555556b02bb0, gpe_context=0x555556eb51b0, encoder_context=0x555556eb1a30, param=0x7fffffffd040) at ../../vanilla/src/gen9_vp9_encoder.c:1208
1208        i965_gpe_context_unmap_curbe(gpe_context);
(gdb) p *cmd
$1 = {dw0 = {frame_width = 1920, frame_height = 1080}, dw1 = {frame_type = 0, segmentation_enable = 0, ref_frame_flags = 0, num_tlevels = 1}, dw2 = {reserved = 0, intra_mode_disable = 0, loop_filter_type = 0}, dw3 = {max_level_ratiot0 = 0, max_level_ratiot1 = 0, max_level_ratiot2 = 0, max_level_ratiot3 = 0}, dw4 = {
    profile_level_max_frame = 2073600}, dw5 = {init_buf_fullness = 750000}, dw6 = {buf_size = 1000000}, dw7 = {target_bit_rate = 1000000}, dw8 = {max_bit_rate = 1000000}, dw9 = {min_bit_rate = 0}, dw10 = {frame_ratem = 127}, dw11 = {frame_rated = 1}, dw12 = {brc_flag = 16, gopp = 249}, dw13 = {
    init_frame_width = 1920, init_frame_height = 1080}, dw14 = {avbr_accuracy = 30, avbr_convergence = 150}, dw15 = {min_qp = 0, max_qp = 255}, dw16 = {cq_level = 30, reserved = 0}, dw17 = {enable_dynamic_scaling = 0, brc_overshoot_cbr_pct = 150}, dw18 = {pframe_deviation_threshold0 = 208, 
    pframe_deviation_threshold1 = 211, pframe_deviation_threshold2 = 215, pframe_deviation_threshold3 = 219}, dw19 = {pframe_deviation_threshold4 = 37, pframe_deviation_threshold5 = 41, pframe_deviation_threshold6 = 45, pframe_deviation_threshold7 = 48}, dw20 = {vbr_deviation_threshold0 = 208, 
    vbr_deviation_threshold1 = 211, vbr_deviation_threshold2 = 214, vbr_deviation_threshold3 = 219}, dw21 = {vbr_deviation_threshold4 = 80, vbr_deviation_threshold5 = 84, vbr_deviation_threshold6 = 93, vbr_deviation_threshold7 = 97}, dw22 = {kframe_deviation_threshold0 = 209, kframe_deviation_threshold1 = 212, 
    kframe_deviation_threshold2 = 218, kframe_deviation_threshold3 = 222}, dw23 = {kframe_deviation_threshold4 = 34, kframe_deviation_threshold5 = 40, kframe_deviation_threshold6 = 45, kframe_deviation_threshold7 = 48}, dw24 = {target_size = 0}, dw25 = {frame_number = 0}, dw26 = {reserved = 0}, dw27 = {
    hrd_buffer_fullness_upper_limit = 0}, dw28 = {hrd_buffer_fullness_lower_limit = 0}, dw29 = {reserved = 0}, dw30 = {reserved = 0}, dw31 = {reserved = 0}, dw32 = {seg_delta_qp0 = 0, seg_delta_qp1 = 0, seg_delta_qp2 = 0, seg_delta_qp3 = 0}, dw33 = {seg_delta_qp4 = 0, seg_delta_qp5 = 0, seg_delta_qp6 = 0, 
    seg_delta_qp7 = 0}, dw34 = {temporal_id = 0, multi_ref_qp_check = 0, reserved = 0}, dw35 = {max_num_pak_passes = 0, sync_async = 0, overflow = 0, mbrc = 0}, dw36 = {reserved = 0, segmentation = 0, reserved1 = 0}, dw37 = {cur_qpindex = 0, lastref_qpindex = 0, goldref_qpindex = 0, altref_qpindex = 0}, dw38 = {
    qdelta_ydc = 0, qdelta_uvac = 0, qdelta_uvdc = 0, reserved = 0}, dw39 = {reserved = 0}, dw40 = {reserved = 0}, dw41 = {reserved = 0}, dw42 = {reserved = 0}, dw43 = {reserved = 0}, dw44 = {reserved = 0}, dw45 = {reserved = 0}, dw46 = {reserved = 0}, dw47 = {reserved = 0}, dw48 = {brc_y4x_input_bti = 0}, dw49 = {
    brc_vme_coarse_intra_input_bti = 1}, dw50 = {brc_history_buffer_bti = 2}, dw51 = {brc_const_data_input_bti = 3}, dw52 = {brc_distortion_bti = 4}, dw53 = {brc_mmdk_pak_output_bti = 5}, dw54 = {brc_enccurbe_input_bti = 6}, dw55 = {brc_enccurbe_output_bti = 7}, dw56 = {brc_pic_state_input_bti = 8}, dw57 = {
    brc_pic_state_output_bti = 9}, dw58 = {brc_seg_state_input_bti = 10}, dw59 = {brc_seg_state_output_bti = 11}, dw60 = {brc_bitstream_size_data_bti = 12}, dw61 = {brc_hfw_data_output_bti = 13}, dw62 = {reserved = 0}, dw63 = {reserved = 0}}
```
-> GPU hang.

The only difference here is dw10 (frame_ratem) being 126 or 127.

Also reproducible with fractions (frame_ratem / frame_rated): 1269/10 works but 1270/10 hangs.

(Kaby Lake GT2.)

yakuizhao, 2017-03-14:
Thanks for reporting the issue and detailed debug.
This is really an issue. In theory no hang should be triggered. 
Based on your debug the used frame rate is too high while the bit-rate is so low. I don't think that it is a common usage scenario. So IMO it is not worth taking too much effort on the corner case.

fhvwy, 2017-03-14:
The bitrate is irrelevant, as are all of the frame parameters (resolution, etc) as far as I can tell.

For example:
```
1208        i965_gpe_context_unmap_curbe(gpe_context);
(gdb) p *cmd
$1 = {dw0 = {frame_width = 128, frame_height = 128}, dw1 = {frame_type = 0, segmentation_enable = 0, ref_frame_flags = 0, num_tlevels = 1}, dw2 = {reserved = 0, intra_mode_disable = 0, loop_filter_type = 0}, dw3 = {max_level_ratiot0 = 0, max_level_ratiot1 = 0, max_level_ratiot2 = 0, max_level_ratiot3 = 0}, dw4 = {
    profile_level_max_frame = 16384}, dw5 = {init_buf_fullness = 7500000}, dw6 = {buf_size = 10000000}, dw7 = {target_bit_rate = 10000000}, dw8 = {max_bit_rate = 10000000}, dw9 = {min_bit_rate = 10000000}, dw10 = {frame_ratem = 127}, dw11 = {frame_rated = 1}, dw12 = {brc_flag = 32, gopp = 249}, dw13 = {
    init_frame_width = 128, init_frame_height = 128}, dw14 = {avbr_accuracy = 30, avbr_convergence = 150}, dw15 = {min_qp = 0, max_qp = 255}, dw16 = {cq_level = 30, reserved = 0}, dw17 = {enable_dynamic_scaling = 0, brc_overshoot_cbr_pct = 150}, dw18 = {pframe_deviation_threshold0 = 208, 
    pframe_deviation_threshold1 = 211, pframe_deviation_threshold2 = 215, pframe_deviation_threshold3 = 219}, dw19 = {pframe_deviation_threshold4 = 37, pframe_deviation_threshold5 = 41, pframe_deviation_threshold6 = 45, pframe_deviation_threshold7 = 48}, dw20 = {vbr_deviation_threshold0 = 208, 
    vbr_deviation_threshold1 = 211, vbr_deviation_threshold2 = 214, vbr_deviation_threshold3 = 219}, dw21 = {vbr_deviation_threshold4 = 80, vbr_deviation_threshold5 = 84, vbr_deviation_threshold6 = 93, vbr_deviation_threshold7 = 97}, dw22 = {kframe_deviation_threshold0 = 209, kframe_deviation_threshold1 = 212, 
    kframe_deviation_threshold2 = 218, kframe_deviation_threshold3 = 222}, dw23 = {kframe_deviation_threshold4 = 34, kframe_deviation_threshold5 = 40, kframe_deviation_threshold6 = 45, kframe_deviation_threshold7 = 48}, dw24 = {target_size = 0}, dw25 = {frame_number = 0}, dw26 = {reserved = 0}, dw27 = {
    hrd_buffer_fullness_upper_limit = 0}, dw28 = {hrd_buffer_fullness_lower_limit = 0}, dw29 = {reserved = 0}, dw30 = {reserved = 0}, dw31 = {reserved = 0}, dw32 = {seg_delta_qp0 = 0, seg_delta_qp1 = 0, seg_delta_qp2 = 0, seg_delta_qp3 = 0}, dw33 = {seg_delta_qp4 = 0, seg_delta_qp5 = 0, seg_delta_qp6 = 0, 
    seg_delta_qp7 = 0}, dw34 = {temporal_id = 0, multi_ref_qp_check = 0, reserved = 0}, dw35 = {max_num_pak_passes = 0, sync_async = 0, overflow = 0, mbrc = 0}, dw36 = {reserved = 0, segmentation = 0, reserved1 = 0}, dw37 = {cur_qpindex = 0, lastref_qpindex = 0, goldref_qpindex = 0, altref_qpindex = 0}, dw38 = {
    qdelta_ydc = 0, qdelta_uvac = 0, qdelta_uvdc = 0, reserved = 0}, dw39 = {reserved = 0}, dw40 = {reserved = 0}, dw41 = {reserved = 0}, dw42 = {reserved = 0}, dw43 = {reserved = 0}, dw44 = {reserved = 0}, dw45 = {reserved = 0}, dw46 = {reserved = 0}, dw47 = {reserved = 0}, dw48 = {brc_y4x_input_bti = 0}, dw49 = {
    brc_vme_coarse_intra_input_bti = 1}, dw50 = {brc_history_buffer_bti = 2}, dw51 = {brc_const_data_input_bti = 3}, dw52 = {brc_distortion_bti = 4}, dw53 = {brc_mmdk_pak_output_bti = 5}, dw54 = {brc_enccurbe_input_bti = 6}, dw55 = {brc_enccurbe_output_bti = 7}, dw56 = {brc_pic_state_input_bti = 8}, dw57 = {
    brc_pic_state_output_bti = 9}, dw58 = {brc_seg_state_input_bti = 10}, dw59 = {brc_seg_state_output_bti = 11}, dw60 = {brc_bitstream_size_data_bti = 12}, dw61 = {brc_hfw_data_output_bti = 13}, dw62 = {reserved = 0}, dw63 = {reserved = 0}}
```
also hangs (128x128 at 10Mbps).

yakuizhao, 2017-03-14:
It is really an issue if the hang is triggered regardless of bit-rate.
And the driver needs to handle it.

chivakker, 2017-04-25:
@fhvwy can you provide some explanation for the need of higher framerate?  This can help me provide a proper fix.  I need to understand your need to go above 60fps.

fhvwy, 2017-04-27:
I don't have any specific use-case in mind.  There shouldn't be anything special about any particular framerate, though, so I would regard it as surprising if any number in allowed range of the VAAPI parameters (1/65535 to 65535/1) was not supported.

YuJiankang, 2017-06-15:
I set the frame rate to 127 and it seems I can't reproduce this issue with latest driver code.
@fhvwy

fhvwy, 2017-06-16:
Still happens completely consistently for me with current git (7633739615b39762a10ccc85c3e52aa281e98592).

Encoding at just under 126.9fps with:
`
 ./avconv -y -init_hw_device vaapi:/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -c:v vp9_vaapi -r 1269/10 -b:v 5M -frames:v 1000 out.webm
`
works perfectly but encoding at 127fps with
`
 ./avconv -y -init_hw_device vaapi:/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i in.mp4 -an -c:v vp9_vaapi -r 127 -b:v 5M -frames:v 1000 out.webm
`
hangs immediately.

Machine is 7500U (Kaby Lake GT2, 24 EUs).

fhvwy, 2017-06-16:
Hang log in kernel:
```
[2545026.067337] [drm] GPU HANG: ecode 9:0:0x96d9ccee, in avconv [3630], reason: Hang on render ring, action: reset
[2545026.067347] [drm] GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace.
[2545026.067350] [drm] Please file a _new_ bug report on bugs.freedesktop.org against DRI -> DRM/Intel
[2545026.067352] [drm] drm/i915 developers can then reassign to the right component if it's not a kernel issue.
[2545026.067355] [drm] The gpu crash dump is required to analyze gpu hangs, so please always attach it.
[2545026.067359] [drm] GPU crash dump saved to /sys/class/drm/card0/error
[2545026.067405] drm/i915: Resetting chip after gpu hang
[2545026.067468] [drm] RC6 on
[2545026.083886] [drm] GuC firmware load skipped
```

GPU hang dump: http://ixia.jkqxz.net/~mrt/libva/gpu_hang_dump

(X is running on the same GPU, though it won't be doing anything particularly interesting.)

YuJiankang, 2017-06-19:
@fhvwy 
I use the yami for the test with current intel-driver git (7633739).
I set the frame rate to 127 and can't reproduce the gpu hang issue.

I used the following commands:
yamiencode -i ./500F_1920x1080.I420 -W 1920 -H 1080 --rcmode CBR -f 127 -b 4096 -c VP9 -o test.vp9

Is it possible that other encoder parameters(not only the frame rate 127) passed down by the application led to this problem? or would you please help have a try with yami? thanks a lot

fhvwy, 2017-06-19:
Also happens with current git libyami and intel driver 7633739:
`
tests/yamiencode -i /dev/zero -W 1920 -H 1080 --rcmode CBR -f 127 -b 4096 -c VP9 -o test.vp9
`
hangs the GPU, while
`
tests/yamiencode -i /dev/zero -W 1920 -H 1080 --rcmode CBR -f 126 -b 4096 -c VP9 -o test.vp9
`
does not.

YuJiankang, 2017-06-20:
@fhvwy
we use the same command, Can't reproduce this problem in two kbl platforms.
we suspect the Linux kernel version may be related to this issue. 
our two platform using linux kernel version  4.12.0-rc2 and 4.8.0-22-generic, and they works well with frame rate 127.
so would you please help try update your Linux kernel? thanks a lot.

fhvwy, 2017-06-20:
Currently it's stock 4.9.0-3 from Debian (derived from 4.9.25).  I'll try a 4.12 RC later.

Given that one of your versions is earlier than the one I'm using, are you aware of any specific commits which might be in it which could be relevant?

I'm also curious what GPUs you have tried on - was that a 7500U, or another U-type GT2 which should be similar, or another version entirely?

YuJiankang, 2017-06-20:
@fhvwy 
thank you for your detailed information, I can reproduce this issue on my platform and got the similar GPU hang log. After the GPU reset, It seems encoded bit stream is still ok with the correct frame rate 127 ,  and I neglected to check the GPU hang log.
we will check more and at least, no hang should be triggered. thanks a lot.

fhvwy, 2017-07-15:
I had another look at this.  I don't think the suggested workaround to reject framerates of 127 or higher is sufficient, because there is also corruption in some other output.

With current git ffmpeg and i965, and Linux 4.11.6 from Debian testing, on Kaby Lake 7500U:
```
for fr in $(seq 60 10 150) ; do ./ffmpeg_g -y -init_hw_device vaapi:/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i bbb_1080_264.mp4 -an -c:v vp9_vaapi -r $fr -b 5M -frames:v 100 out_$fr.webm ; done
```
Gives the output files here: <http://ixia.jkqxz.net/~mrt/i965/vp9/>.

The 60, 70 and 80 fps files look sensible.  The 90 and 100 fps files have something strange in the first few frames (first frame is mid-grey rather than black, with some corruption, but it corrects on frame 4).  The 110 and 120 fps files are entirely broken (no good frames at all).  The 130, 140 and 150 fps files all triggered GPU hangs when creating them, but the output files seem sensible after that.

chivakker, 2017-07-28:
After checking the design of this, it should only work up to 60 fps.  The driver should return an error when a higher framerate is attempted.

fhvwy, 2017-07-29:
What range of framerates are allowed by the proprietary shader binaries, then?  Since the VP9 bitstream doesn't include any metadata for the bitrate/framerate, we should be able to transparently scale to the supported range without the user needing to know.

xhaihao, 2018-01-18:
The maximum framerate is limited to 60fps by https://github.com/01org/intel-vaapi-driver/pull/209.

closed 2018-01-18 by xhaihao