intel/intel-vaapi-driver #97: B-frame encoding on SKL+ is broken in current master

xhaihao, 2017-03-23:
sreerenjb commented 2 days ago

I think B-frame encoding is broken in current master, because my gstreamer-vaapi based test cases showing clear artifacts and macroblock decoding errors. And this is reproducible only for streams encoded with B frames.

sreerenjb commented 9 hours ago

B frame encode while enabling High profile is the usage scenario which can reproduce the issue. Till now it was only reproducible in SKL. But with today's git master(c0ea9ac), the issue is reproducible in KBL also, because we started using the same codepath(with new media kernel) of SKL in KBL too.

chivakker, 2017-03-27:
A work around is not sufficient to close a bug.  Re-opening for the root cause analysis to continue.

tmjohns1, 2017-03-29:
Patch to enable direct_spatial_mv_pred_flag seems to fix r2r issues.
Typical test invocation:
avcenc 176 144 176x144-25_1882_seq4.yuv 176x144-25_1882_seq4-1.264 --fb=500 --mode=2

With the encoded streams showing:
transform_8x8_mode_flag = 0
direct_spatial_mv_pred_flag = 0

sha1sum output:
60995b82fb7d4694caa2b9b98847b8647c936363  176x144-25_1882_seq4-1.264
e0299cad0d36780b16f44c46aafd8439d35c2e6d  176x144-25_1882_seq4-2.264
22fc7f350654164d8a4141017ee80bd96db3f9a8  176x144-25_1882_seq4-3.264
e0299cad0d36780b16f44c46aafd8439d35c2e6d  176x144-25_1882_seq4-4.264
885adb12206d23b8121aef1649e57839dd603bf0  176x144-25_1882_seq4-5.264
0fcf36609b6c16a3db21295460dfdd061cd70029  176x144-25_1882_seq4-6.264
c062960fdee66820a146eebe16253a286a06bf6e  176x144-25_1882_seq4-7.264
30368262965c69af552176047786de598c7afba5  176x144-25_1882_seq4-8.264
baed4eeeac2f2ab410045b152688a25a0f0a8e24  176x144-25_1882_seq4-9.264

After patching with Pengfei’s commit 215c04.

The encoded stream shows:
transform_8x8_mode_flag = 0
direct_spatial_mv_pred_flag = 1
Code changes (in encode/avcenc.c:avcenc_update_slice_parameter):
slice_param->direct_spatial_mv_pred_flag = 1;
slice_param->cabac_init_idc = 1;

sha1sum output:
d387b985d84eefabbdbf160e00d35d12bac17525  176x144-25_1882_seq4-1.264
d387b985d84eefabbdbf160e00d35d12bac17525  176x144-25_1882_seq4-2.264
d387b985d84eefabbdbf160e00d35d12bac17525  176x144-25_1882_seq4-3.264
d387b985d84eefabbdbf160e00d35d12bac17525  176x144-25_1882_seq4-4.264
d387b985d84eefabbdbf160e00d35d12bac17525  176x144-25_1882_seq4-5.264

commit 228e4fc1 (2017-03-27): Fix: WA the AVC B frame issue with high profile

this is WA to disable the transform_8x8_mode_flag in the driver.

Fixes #97
commit 371ba259 (2017-03-27): Fix: WA the AVC B frame issue with high profile

this is WA to disable the transform_8x8_mode_flag in the driver.

Fixes #97

(cherry picked from commit 228e4fc197e2a180dcbfcaba4df2a0717f6ec853)
closed 2017-04-19 by xhaihao