intel/intel-vaapi-driver #99: Use a unified code style in driver source code

xhaihao, 2017-03-24:
Currently we don't have a unified code style, which make it is harder to read the code and add maintenance cost

seanvk, 2017-04-24:
Why was this merged without some review?

If you are going to have a new coding style, it is not enough to say "use this command to format".  People need to see instructions on the style and best practices.  I don't want blindly running some hook to format.

This needs:

1) Documentation
2) Examples/Best practices

As a part of the commit.

xhaihao, 2017-04-25:
@seanvk  please check the corresponding PR (https://github.com/01org/intel-vaapi-driver/pull/100), 4 developers reviewed the PR.  

The commit updated the coding style part in CONTRIBUTING.md.  Actually most code in the driver follows the Linux coding style with a few changes before the merging. The Linux coding style is well known and people can download the doc from the Internet, so I don't give the URL in CONTRIBUTING.md

I'd like to close this issue. Please file a new issue for updating CONTRIBUTING.md if you think it is necessary.

seanvk, 2017-04-25:
Okay. I see PR#100 has the necessary reviews. The issue itself was lacking in the details. In the future for a change touching so many files your issue should have more details.

closed 2017-04-25 by seanvk