intel/intel-vaapi-driver #108: ROI(Region of Interest) of avc encding is broken on the master.

zzoon, 2017-03-29:
Since the commit e9a0038 landed, it seems that ROI is broken.
In addition, a size of produced h264 data with roi is much more (over ten times) than h264 data's size without roi. Is this expected?

I tested with libva-utils/encode/avcenc and KBL.

zzoon, 2017-03-29:
I'm sorry for confusing.
I tested on SKL not KBL.

QuPengfei, 2017-03-29:
@zzoon yes. new improved AVC encoder on SKL/KBL broken the ROI feature. we have plan to enable ROI under CQP in the next step.

zzoon, 2017-03-29:
Thanks for reply, then I should wait for it. Feel free to close this issue.

ceyusa, 2017-03-29:
> Thanks for reply, then I should wait for it. Feel free to close this issue.

Perhaps it should remain open, as a remainder of a pending task.

PR #120 (2017-04-06): ENC: Disable the ROI feature of AVC encoder on SKL/APL/KBL
this is WA to disable the feature as the new improved AVC encoder does not support it by now.

Fixes #108
commit c883fa35 (2017-04-06): ENC: Disable the ROI feature of AVC encoder on SKL/APL/KBL

this is WA to disable the feature as the new improved AVC encoder does not support it by now.

Fixes #108

(cherry picked from commit eb2b7f3e9f40f32506136be4f2cec3f1b8d1699f)
closed 2017-04-06 by xhaihao