intel/intel-vaapi-driver #114: gcc 4.8.5 reports possibly uninitialized variable in the driver

dvrogozh, 2017-04-04:
gcc 4.8.5 reports possibly uninitialized variable in the driver:
```
gen9_avc_encoder.c: In function 'gen9_avc_set_curbe_mbenc':
gen9_avc_encoder.c:3469:8: warning: 'cmd.g9' may be used uninitialized in this function [-Wmaybe-uninitialized]
     if(cmd.g9->dw4.use_actual_ref_qp_value)
```
Looking into the code, this happens in the gen9_avc_set_curbe_mbenc() function which is designed to be workable only on the SKL, BXT and KBL platforms. Please, make sure that the function has a default path for other platforms throwing assertion and error rather than trigger undefined behavior.

I was on the following commit:
```
commit 437cbe03eaeb08e6f4ab70cdf8757bccb8de15dd
Author: Víctor Manuel Jáquez Leal <vjaquez@igalia.com>
Date:   Mon Mar 27 09:05:40 2017 -0700

    gen8: accept P010 as valid format
    
    The functions gen8_gpe_set_surface2_state() and
    gen8_gpe_media_chroma_surface_setup() asserts for the surface color
    format, admiting only NV12.
    
    Nonetheless, gen8 also can encoder surfaces with P010 color format.
    
    This patch adds P010 color format in the assertion condition.
    
    Fixes #106
```

xhaihao, 2017-04-05:
This issue is gcc related. the issue disappears if you use a newer version of gcc. Yes, gen9_avc_set_curbe_mbenc was designed to be workable on SKL, BXT and KBL and presently gen9_avc_set_curbe_mbenc is only called on SKL, BXT and KBL. A newer version of gcc can handle this case.

yakuizhao, 2017-04-05:
Although the issue can be fixed on the newer GCC, it is still reasonable request that the code can be compiled/built correctly on the older GCC version.  The distribution system will use the specific GCC version. We can't ask them to update the GCC version  in order to fix one issue. 
So the reasonable fix is to make them be compatible with the old gcc version.

QuPengfei, 2017-04-05:
@dvrogozh this file gen9_avc_encoder.c is designed only for Gen9(SKL/KBL/APL). for other platform gen8/7 etc, the other path will be called.

xhaihao, 2017-04-05:
@yakuizhao the code can be built correctly with the older GCC

yakuizhao, 2017-04-05:
Sometimes the warning is also treated as serious issue on some distribution systems. 
So it will be better that the warning is also fixed.

QuPengfei, 2017-04-05:
the initialization always happened in the Gen9(gen9_avc_encoder.c), APL/SKL or KBL. so this warning is incorrect.

closed 2017-05-12 by xhaihao