intel/intel-vaapi-driver #127: VAAPI using a lot of GPU resources

bengji, 2017-04-09:
When transcoding with Gstreamer-VAAPI a lot more of GPU resources are being used then Gstreamer-MSDK.

Live Transcode VAAPI: GPU 27%
Live Transcode  MSDK: GPU 6%

VAAPI seems to be using 4 times more GPU then MSDK.

Please see the attached image I took screenshot of both servers. 

Environment: 

2x identical servers with Xeon(R) CPU E3-1245 v5 SKYLAKE

Server 1 with VAAPI: Ubuntu 16.04 
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 1.8.2.pre1 (1.7.3-372-g2f0a844)

Gstreamer-VAAPI Pipeline: 
gst-launch-1.0 souphttpsrc location="http://localhost:80/oranews_HD/mpegts" is-live=true ! tsdemux name=demux  ! queue max-size-buffers=1200 max-size-buffers=0 max-size-time=0 ! \
h264parse ! vaapih264dec ! vaapipostproc width=1280 height=720 ! vaapih264enc rate-control=2 bitrate=1700 ! h264parse ! \
flvmux streamable=true name=mux ! rtmpsink location="rtmp://localhost:1935/pushrtmp/vappi_outs live=1" demux. ! queue max-size-buffers=1200 max-size-buffers=0 max-size-time=0 ! \
mpegaudioparse ! queue ! avdec_mp2float plc=true ! audioconvert ! queue ! voaacenc bitrate=128000 ! mux.

Server 2 with MSDK: Centos 7.2
Media Server Studio 2017
vainfo: VA-API version: 0.99 (libva 1.67.0.pre1)

Gstreamer-MSDK Pipeline: 
gst-launch-1.0 souphttpsrc location="http://localhost:80/oranews_HD/mpegts" is-live=true ! tsdemux name=demux  ! queue max-size-buffers=1200 max-size-buffers=0 max-size-time=0 \
! h264parse ! mfxh264dec ! mfxvpp width=1280 height=720 ! mfxh264enc rate-control=1 bitrate=1700 ! flvmux streamable=true name=mux ! rtmpsink location="rtmp://localhost:1935/pushrtmp/mfx_out live=1" demux. \
! queue max-size-buffers=1200 max-size-buffers=0 max-size-time=0 ! mpegaudioparse ! queue ! avdec_mp2float plc=true ! audioconvert ! queue ! voaacenc bitrate=128000 ! mux.

[image]

QuPengfei, 2017-04-11:
@bengji what is the resolution of input? it looks there is scaling in the pipeline. You can remove the VPP(scaling) and try it again. let us narrow down to decoder/encoder only firstly.

you can try GPU analysis with vtune. it will more accurate.

bengji, 2017-04-11:
resolution is 1920x1080, we tried removing vappipostproc and that seems to be the problem, the load went down to 10% from 30%

scott-d-phillips, 2017-04-11:
@bengji could you run both pipelines with `-v` and paste the output here? It's possible that the two pipelines are doing different things (something like, one is encoding interlaced constrained baseline profile, and the other is encoding progressive high profile for example).

bengji, 2017-04-12:
@scott-d-phillips I'm using the same source for both VAAPI and MSDK it's interlaced 25 FPS,

I noticed that deinterlace is being inserted automatically and it's making it 50 fps after encoding it.

I tried disabling deinterlace in vappipostproc and then the load seems to be normal similar to what im getting with MSDK.

The bug seems to be when source is interlaced it doubles the framerate automatically. 

Please find the pipeline logs' for both MSDK and VAAPI so you can compare.

[VAAPI_pipeline_verbose.txt](https://github.com/01org/intel-vaapi-driver/files/918109/VAAPI_pipeline_verbose.txt)
[MSDK_pipline_verbose.txt](https://github.com/01org/intel-vaapi-driver/files/918110/MSDK_pipline_verbose.txt)

xhaihao, 2017-04-13:
@bengji so it is gst-vaapi related issue. Per https://gstreamer.freedesktop.org/documentation/contribute/,
could you file a gstreamer bug at http://bugzilla.gnome.org/. I will close this issue as 'not our bug'

closed 2017-04-13 by xhaihao