intel/intel-vaapi-driver #146: HEVC encoder produces jumpy video

sammys, 2017-04-27:
HEVC encodes in my setup appear to have encoded frame drops in the same part of every second so it is jumpy in a repetitive way. H264 encodes are smooth.

QP=19
FFMPEG 3.2.4
intel-vaapi-driver and libva HEAD as of 2017-04-20
Ubuntu 16.04

fhvwy, 2017-04-27:
Might be <https://trac.ffmpeg.org/ticket/5821>?  Try with a more current ffmpeg version or with B-frames disabled.

sammys, 2017-04-28:
Disabling B-frames did the trick! Thank you! It seems odd though that not supplying `-bf` also produces jerky video yet ffprobe (version as above) reports there are no B-frames in the stream. Thanks again!

pengche1, 2017-04-28:
@sammys , it seems B frames can't be encoded with FFMPEG vaapi accelerator?  have you try this https://trac.ffmpeg.org/ticket/5821?

sammys, 2017-04-28:
@pengche1  That ticket explains setting `-bf` to zero and, as I reported, that trick works. What I meant by my comment is that not using `-bf` on the command line produces a stream that does not contain B-frames yet it is jerky. I find that odd because ffmpeg appears to disable B-frames internally (or vaapi doesn't support HEVC B-frames) when `-bf` is not used yet one gets jerky video unless `-bf 0` is forced on the command line.

pengche1, 2017-04-28:
Thanks,@sammy, I have gotten your point, -_-. Vaapi do support HEVC B encoding. You could have a try with gstreamer-vaapi/libyami for HEVC B frame encoding. And none B frame may be a ffmpeg issue. You can talk with them. Thanks!

sammys, 2017-04-28:
Thanks, @pengche1. Will look into gstreamer!

fhvwy, 2017-04-28:
The error isn't in the B-frame encode, it's in the metadata used by the decoder to decide the amount of buffering to apply to the stream to fix reordering.  That wasn't sufficient, so you get the jerky effect when some B-frames end up being presented after other frames which they were meant to be before.  (Hence the observed behaviour that disabling B-frames removes the issue.)

In any case, as noted in the ffmpeg bug, this was fixed months ago - it works correctly in any current version (3.3 or git).

sammys, 2017-04-29:
@fhvwy Makes total sense. Thanks! I'll look into ffmeg 3.3+ too

sammys, 2017-04-29:
Maybe I ought to report this to ffmpeg but since I started things here I'll continue for now and we can work towards a solution. I appreciate the direct guidance I've received in here and the efforts everyone has put in to get VA-API going.

I've given ffmpeg 3.3 a whirl and it has improved things. I've tried `-bf 4` and `-bf 2` on the same video and the setting of 4 is less jumpy than I had when using ffmpeg 3.2.4 but in things like clouds there is still some jump. Setting of 2 results in less jump in things like clouds but it's still there... just.

I will see if I can figure out how to increase the buffer size and then whether it improves stuff.

xhaihao, 2017-05-11:
@sammys Is your issue fixed? I'd like to close this issue as it is not a driver issue.

sammys, 2017-05-11:
Yes, thank you to everyone for their help!

closed 2017-05-11 by sammys