intel/intel-vaapi-driver #151: Regression in H.265 encode performance

fhvwy, 2017-04-30:
Skylake GT2, transcoding 1080p H.264 input to H.265, CQP = 25, 2 B-frames:

aa77cb71b3a052cffc085aedcdf437a320fb7377 (no quality options):  ~140fps
5471b938d580e0e35227d305d05baa4b74db1402 (quality = 1):  ~45fps
5471b938d580e0e35227d305d05baa4b74db1402 (quality = 0 / default):  ~85fps
5471b938d580e0e35227d305d05baa4b74db1402 (quality = 8):  ~70fps

(Also somewhat odd that the "default" option is faster than the supposedly fastest level, possibly this is related to #137.)

pengche1, 2017-05-02:
yes, it should be the same issue as #137. Could you try it again with new pull #154?

fhvwy, 2017-05-02:
c2574ec653dd486dd61759b8f2b800fdf9a1de93 (quality = 8): ~165fps

4K:
aa77cb71b3a052cffc085aedcdf437a320fb7377 (default): ~49fps
c2574ec653dd486dd61759b8f2b800fdf9a1de93 (quality = 1): ~18fps
c2574ec653dd486dd61759b8f2b800fdf9a1de93 (quality = 0 / default): ~25fps
c2574ec653dd486dd61759b8f2b800fdf9a1de93 (quality = 8): ~57fps

So yeah, looks good - close this when that gets merged :)

Related:  Was it a deliberate decision to choose a slower mode as the default?  Given the usual use-case of hardware encoders (maximise speed / minimise power consumption, size less important), having the fastest mode as the default would seem more appropriate to me, especially when the difference in speed between them is this large.  (Or does the fastest mode actually use more power somehow?)

pengche1, 2017-05-03:
thanks for your testing. You can choose any level you want. In my opinion, there should not have too much difference of the power consumption between them.

xhaihao, 2017-05-11:
Since pengche1's patch got merged in mater, I closed this issue.

closed 2017-05-11 by xhaihao