intel/intel-vaapi-driver #157: Does SKL support 10bit HEVC decode?

codeathon, 2017-05-04:
Wanted to find out if HEVC 10bit profile is supported for decode on SKL?
The README does not mention it, so wanted to confirm.

uartie, 2017-05-05:
The [README](https://github.com/01org/intel-vaapi-driver/blob/master/README) says:

HEVC 10bit     D       BXT+

And in [src/i965_device_info.c](https://github.com/01org/intel-vaapi-driver/blob/master/src/i965_device_info.c#L327), the `.has_hevc10_decoding` is not set for SKL.  So there is no support for HEVC 10bit decode on SKL.

closed 2017-05-05 by uartie

ngeorgis, 2017-05-05:
Weird.. I remember I was able to decode HEVC 10bit in hybrid mode (not fixed function like on my Kabylake) using VAAPI.  CPU load was 30% to 40% for 10-bit QFHD HEVC on a Skylake Core i7 6770.

QuPengfei, 2017-05-05:
@ngeorgis i think you are using intel media sdk.

ngeorgis, 2017-05-05:
No media sdk on this system as I have not patched the kernel yet.

uartie, 2017-05-05:
If you're using intel media sdk, it is not compatible with this driver.  See media sdk [README](https://github.com/Intel-Media-SDK/MediaSDK/blob/master/README.md) about that.