intel/intel-vaapi-driver #161: [SKL] AVC encoder MB rate control can't work in CBR/VBR mode

mypopydev, 2017-05-08:
When enable MB rate control (mb_rate_control == 1) in AVC encoder, it's can't work in CBR/VBR mode.

reproduce step:

1. Apply the patch as follow
 in FFmpeg master branch, and build FFmpeg with config "./configure --enable-vaapi", after build finish, the MB rate control can be check with the command:
./ffmpeg -h encoder=h264_vaapi

media@media-desktop:~/Sources/FFmpeg$ cat enable_mbbrc.patch 
diff --git a/libavcodec/vaapi_encode_h264.c b/libavcodec/vaapi_encode_h264.c
index 92e29554ed..23e8cdc7e2 100644
--- a/libavcodec/vaapi_encode_h264.c
+++ b/libavcodec/vaapi_encode_h264.c
@@ -168,6 +168,7 @@ typedef struct VAAPIEncodeH264Options {
     int qp;
     int quality;
     int low_power;
+    int mbbrc;
 } VAAPIEncodeH264Options;
 
 
@@ -1157,6 +1158,12 @@ static av_cold int vaapi_encode_h264_configure(AVCodecContext *avctx)
 #endif
     }
 
+    if (ctx->va_rc_mode == VA_RC_CQP && opt->mbbrc != 0)
+        av_log(avctx, AV_LOG_WARNING, "The MB level bitrate control is not "
+               "compatible with Constant QP, it's will ignored it.\n");
+    else
+        ctx->rc_params.rc.rc_flags.bits.mb_rate_control = opt->mbbrc;
+
     return 0;
 }
 
@@ -1283,6 +1290,11 @@ static const AVOption vaapi_encode_h264_options[] = {
     { "low_power", "Use low-power encoding mode (experimental: only supported "
       "on some platforms, does not support all features)",
       OFFSET(low_power), AV_OPT_TYPE_INT, { .i64 = 0 }, 0, 1, FLAGS },
+    { "mbbrc", "MB level bitrate control",
+      OFFSET(mbbrc), AV_OPT_TYPE_FLAGS, { .i64 = 0 }, 0, 2, FLAGS, "mbbrc" },
+        { "auto", NULL, 0,  AV_OPT_TYPE_CONST, { .i64 = 0 }, 0, 0, FLAGS, "mbbrc"},
+        { "on", NULL, 0,  AV_OPT_TYPE_CONST, { .i64 = 1 }, 0, 0, FLAGS, "mbbrc"},
+        { "off", NULL, 0,  AV_OPT_TYPE_CONST, { .i64 = 2 }, 0, 0, FLAGS, "mbbrc"},
     { NULL },
 };

2. Encoder with CBR mode or VBR mode like this:

a). CBR mode 
./ffmpeg -y -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -f rawvideo -s 1280x720 -pix_fmt nv12 -i ../../clip/YUV/panasonicf1_1280x720_300frame_NV12.yuv  -an -vf 'format=nv12,hwupload' -c:v h264_vaapi -b 3M -maxrate 3M -mbbrc 1 mbbrc_1_cbr.h264

b). VBR mode
./ffmpeg -y -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -f rawvideo -s 1280x720 -pix_fmt nv12 -i ../../clip/YUV/panasonicf1_1280x720_300frame_NV12.yuv  -an -vf 'format=nv12,hwupload' -c:v h264_vaapi -b 3M -maxrate 7M -mbbrc 1 mbbrc_1_vbr.h264

than you can use ffplay check the mbbrc_1_cbr.h264/mbbrc_1_vbr.h264

QuPengfei, 2017-05-08:
please try this patch
https://github.com/01org/intel-vaapi-driver/pull/159

mypopydev, 2017-05-09:
@QuPengfei This fix is working, verified in SKL, with the master branch.

mypopydev, 2017-05-10:
@QuPengfei @xhaihao This fix is working, verified in the SKL

QuPengfei, 2017-05-10:
@mypopydev thanks. close this issue.

closed 2017-05-10 by QuPengfei

Brainiarc7, 2019-05-19:
Hello there,

Is the patch for enabling mbbrc for VAAPI upstream on FFmpeg?
And if not, are there plans of porting it upstream?