intel/intel-vaapi-driver #163: [APL] Running AVC encode with lowpower using yamitranscode causing GPU hung issue

SiewHoon, 2017-05-11:
Without GuC and HuC firmware loaded in for CQP mode, running yamitranscode with lowpower immediately causing GPU hung.

Intel VA Driver:  00fdcd3b9395e6355bdfc793e6f8d793c06b0dd2
Libva:  d829a36cf71ee3bcbeff670fde4287070bb6000a
Libyami and Libyami-utils: 1.2.0 version

Command:
./yamitranscode -i 1920x1080.i420 -W 1920 -H 1080 -c AVC -s I420 --rcmode CQP -b 2000 --lowpower -o test264.264 &

dmesg:
[  609.270567] ata1.00: Enabling discard_zeroes_data
[  612.726330] [drm] GPU HANG: ecode 9:2:0xa8ffff7c, in yamitranscode [640], reason: No progress on bsd ring, action: reset
[  612.726334] [drm] GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace.
[  612.726335] [drm] Please file a _new_ bug report on bugs.freedesktop.org against DRI -> DRM/Intel
[  612.726336] [drm] drm/i915 developers can then reassign to the right component if it's not a kernel issue.
[  612.726337] [drm] The gpu crash dump is required to analyze gpu hangs, so please always attach it.
[  612.726338] [drm] GPU crash dump saved to /sys/class/drm/card0/error
[  612.726510] drm/i915: Resetting chip after gpu hang
[  612.728732] [drm] RC6 on
[  612.729090] [drm] GuC firmware load skipped

This issue able to reproduce in linux-4.11 and 4.10.14 version get from kernel.org (https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.11.tar.xz).

xhaihao, 2017-05-11:
@FocusLuo could you give a try with AVC VDEnc on APL?

FocusLuo, 2017-05-11:
Tried on my side, no GPU hang issue found.
Can you help to update your cmd line as below?
./yamitranscode -i 1920x1080.i420 -W 1920 -H 1080 -c AVC -s I420 --rcmode CQP -qp 26 --lowpower -o test264.264 &

SiewHoon, 2017-05-11:
FYI. I tested with linux-4.11 with GuC and HuC firmware loaded in APL, the GPU hung issue can't reproduce in CQP mode using yamitranscode.

SiewHoon, 2017-05-11:
This is a script that I used: 
#!/bin/sh
for i in {1..8}
do 
   echo "play $i"
   ./yamitranscode -i 1920x1080.i420 -W 1920 -H 1080 -c AVC -s I420 --rcmode CQP -b 2000 --lowpower -o test264_$i.264 &
done

SiewHoon, 2017-05-11:
@FocusLuo , you really can't reproduce the issue without HuC firmware loaded? May I know what is the boot kernel parameter that you are using?

xhaihao, 2017-05-11:
@siewhoon VDEnc+CQP doesn't use HuC no matter huc is loaded or not.

The root cause of this issue is that yamitranscode still select CBR mode when both CQP and bitrate are specified in the command line.  VDEnc+CBR doesn't work without HuC firmware loaded.

Although the driver reports CQP only through vaGetConfigAttributes() when HuC firmware isn't loaded, and expect that the middleware uses the right bitrate control mode, the driver should check CBR is supported or not when creating VA config in case the middleware doesn't follow VA-API requirement. We are using https://github.com/01org/intel-vaapi-driver/issues/35 to track this issue.

FocusLuo, 2017-05-11:
@siewhoon Can you help to update your cmd line as below?
./yamitranscode -i 1920x1080.i420 -W 1920 -H 1080 -c AVC -s I420 --rcmode CQP -qp 26 --lowpower -o test264.264 &

SiewHoon, 2017-05-11:
@FocusLuo , tried with your command. Yes, it can't reproduce the issue anymore. But according to yamitranscode --help, -qp is optional.  :( 

Ok noted, @xhaihao

xhaihao, 2017-05-12:
Closed as the issue disappears if using the right command. We use https://github.com/01org/intel-vaapi-driver/issues/35 to track the issue caused by the incorrect config settings.

closed 2017-05-12 by xhaihao