intel/intel-vaapi-driver #183: sigserv/segfaultt observed on /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so when the VAAPI encoder for VP8 is called up on Skylake

Brainiarc7, 2017-06-01:
**Attempting to encode VP8 via the vp8_vaapi encoder on FFmpeg (and liabv) fails on Intel Skylake with a sigserv/segfault:**

First, I built and configured FFmpeg with debugging enabled so as to run a gdb trace and find out what triggers the sigserv:

**FFmpeg build configuration:**

This generates a local FFmpeg build for use with VAAPI only.

Includes extra components as needed:

**Build and deploy Yasm:**
[Yasm](http://yasm.tortall.net/) is an assembler for x86 optimizations used by x264 and FFmpeg. Highly recommended or your resulting build may be very slow.

    cd ~/ffmpeg_sources
    wget http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz
    tar xzvf yasm-1.3.0.tar.gz
    cd yasm-1.3.0
    ./configure --prefix="/apps/ffmpeg/dyn" --bindir="/apps/ffmpeg/dyn/bin"
    make -j$(nproc) VERBOSE=1
    make -j$(nproc) install
    make -j$(nproc) distclean

**Build and deploy libx264 statically:**
This library provides a H.264 video encoder. See the [H.264 Encoding Guide](https://trac.ffmpeg.org/wiki/Encode/H.264) for more information and usage examples.
This requires ffmpeg to be configured with *--enable-gpl* *--enable-libx264*.

    cd ~/ffmpeg_sources
    wget http://download.videolan.org/pub/x264/snapshots/last_x264.tar.bz2
    tar xjvf last_x264.tar.bz2
    cd x264-snapshot*
    PATH="/apps/ffmpeg/dyn/bin:$PATH" ./configure --prefix="/apps/ffmpeg/dyn" --bindir="/apps/ffmpeg/dyn/bin" --enable-static --disable-opencl
    PATH="/apps/ffmpeg/dyn/bin:$PATH" make -j$(nproc) VERBOSE=1
    make -j$(nproc) install VERBOSE=1
    make -j$(nproc) distclean

**Build and configure libx265:**
This library provides a H.265/HEVC video encoder. See the [H.265 Encoding Guide](https://trac.ffmpeg.org/wiki/Encode/H.265) for more information and usage examples.

    sudo apt-get install cmake mercurial
    cd ~/ffmpeg_sources
    hg clone https://bitbucket.org/multicoreware/x265
    cd ~/ffmpeg_sources/x265/build/linux
    PATH="$/apps/ffmpeg/dyn/bin:$PATH" cmake -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX="/apps/ffmpeg/dyn" -DENABLE_SHARED:bool=off ../../source
    make -j$(nproc) VERBOSE=1
    make -j$(nproc) install VERBOSE=1
    make -j$(nproc) clean VERBOSE=1

**Build and deploy the libfdk-aac library:**
This provides an AAC audio encoder. See the [AAC Audio Encoding Guide](https://trac.ffmpeg.org/wiki/Encode/AAC) for more information and usage examples.
This requires ffmpeg to be configured with *--enable-libfdk-aac* (and *--enable-nonfree* if you also included *--enable-gpl*).

    cd ~/ffmpeg_sources
    wget -O fdk-aac.tar.gz https://github.com/mstorsjo/fdk-aac/tarball/master
    tar xzvf fdk-aac.tar.gz
    cd mstorsjo-fdk-aac*
    autoreconf -fiv
    ./configure --prefix="/apps/ffmpeg/dyn" --disable-shared
    make -j$(nproc)
    make -j$(nproc) install
    make -j$(nproc) distclean

**Build and configure libvpx:**

  

     cd ~/ffmpeg_sources
       git clone https://github.com/webmproject/libvpx/
       cd libvpx
       ./configure --prefix="/apps/ffmpeg/dyn" --enable-runtime-cpu-detect --enable-vp9 --enable-vp8 \
       --enable-postproc --enable-vp9-postproc --enable-multi-res-encoding --enable-webm-io --enable-vp9-highbitdepth --enable-onthefly-bitpacking --enable-realtime-only \
[... 376 lines cut ...]
        04:00.0 PCI bridge: Intel Corporation Device 1576
        04:01.0 PCI bridge: Intel Corporation Device 1576
        04:02.0 PCI bridge: Intel Corporation Device 1576
        05:00.0 System peripheral: Intel Corporation Device 1575
        07:00.0 USB controller: Intel Corporation Device 15b5
        3c:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a804
        3d:00.0 Ethernet controller: Qualcomm Atheros Killer E2400 Gigabit Ethernet Controller (rev 10)
        3e:00.0 Network controller: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32)

3. OS information:

        uname -r

        4.10.12

        lsb_release -a

        $ lsb_release -a
        LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:printing-9.20160110ubuntu0.2-amd64:printing-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
        Distributor ID:	Ubuntu
        Description:	Ubuntu 16.04.2 LTS
        Release:	16.04
        Codename:	xenial

So far, it seems that this crash originates from the i915 driver's interactions with VAAPI, and I was also able to reproduce the exact crash with libav. Please help.

QuPengfei, 2017-06-01:
Do you reproduce it  without "scale_vaapi=w=1920:h=1080"?

Brainiarc7, 2017-06-01:
Hello @QuPengfei ,

Yes, I was able to reproduce the exact crash without using the vaapi scale filter, as shown below:

```
lin@phantom:~/Desktop/encodes/src$ gdb ~/ffmpeg_sources/FFmpeg/ffmpeg_g 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/lin/ffmpeg_sources/FFmpeg/ffmpeg_g...done.
(gdb) run -loglevel debug -threads 4 -vaapi_device /dev/dri/renderD129 -i "Demon.Hunter.One.Step.Behind.mkv" -c:a aac \
-filter:v 'format=nv12,hwupload' \
-c:v vp8_vaapi -loop_filter_level:v 32 -b:v 2500k \
-f matroska -y "Demon.Hunter.One.Step.Behind-VAAPIEXT.mkv"
Starting program: /home/lin/ffmpeg_sources/FFmpeg/ffmpeg_g -loglevel debug -threads 4 -vaapi_device /dev/dri/renderD129 -i "Demon.Hunter.One.Step.Behind.mkv" -c:a aac -filter:v 'format=nv12,hwupload' -c:v vp8_vaapi -loop_filter_level:v 32 -b:v 2500k -f matroska -y "Demon.Hunter.One.Step.Behind-VAAPIEXT.mkv"
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
ffmpeg version N-86329-ga47273c Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 20160609
  configuration: --pkg-config-flags=--static --prefix=/apps/ffmpeg/dyn --extra-cflags=-I/apps/ffmpeg/dyn/include --extra-ldflags=-L/apps/ffmpeg/dyn/lib --bindir=/apps/ffmpeg/dyn/bin --enable-debug=3 --enable-vaapi --enable-libvorbis --enable-libvpx --enable-gpl --cpu=native --enable-opengl --enable-libfdk-aac --enable-libx264 --enable-libx265 --enable-nonfree
  libavutil      55. 63.100 / 55. 63.100
  libavcodec     57. 96.101 / 57. 96.101
  libavformat    57. 72.101 / 57. 72.101
  libavdevice    57.  7.100 / 57.  7.100
  libavfilter     6. 90.100 /  6. 90.100
  libswscale      4.  7.101 /  4.  7.101
  libswresample   2.  8.100 /  2.  8.100
  libpostproc    54.  6.100 / 54.  6.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-threads' ... matched as AVOption 'threads' with argument '4'.
Reading option '-vaapi_device' ... matched as option 'vaapi_device' (set VAAPI hardware device (DRM path or X11 display name)) with argument '/dev/dri/renderD129'.
Reading option '-i' ... matched as input url with argument 'Demon.Hunter.One.Step.Behind.mkv'.
Reading option '-c:a' ... matched as option 'c' (codec name) with argument 'aac'.
Reading option '-filter:v' ... matched as option 'filter' (set stream filtergraph) with argument 'format=nv12,hwupload'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'vp8_vaapi'.
Reading option '-loop_filter_level:v' ... matched as AVOption 'loop_filter_level:v' with argument '32'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '2500k'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'matroska'.
Reading option '-y' ... matched as option 'y' (overwrite output files) with argument '1'.
Reading option 'Demon.Hunter.One.Step.Behind-VAAPIEXT.mkv' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Applying option vaapi_device (set VAAPI hardware device (DRM path or X11 display name)) with argument /dev/dri/renderD129.
[AVHWDeviceContext @ 0x29e34c0] Opened VA display via DRM device /dev/dri/renderD129.
libva info: VA-API version 0.39.2
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[AVHWDeviceContext @ 0x29e34c0] Initialised VAAPI connection: version 0.39
[AVHWDeviceContext @ 0x29e34c0] Format 0x32315659 -> yuv420p.
[AVHWDeviceContext @ 0x29e34c0] Format 0x30323449 -> unknown.
[AVHWDeviceContext @ 0x29e34c0] Format 0x3231564e -> nv12.
[AVHWDeviceContext @ 0x29e34c0] Format 0x32595559 -> yuyv422.
[AVHWDeviceContext @ 0x29e34c0] Format 0x59565955 -> uyvy422.
[AVHWDeviceContext @ 0x29e34c0] Format 0x48323234 -> yuv422p.
[AVHWDeviceContext @ 0x29e34c0] Format 0x58424752 -> rgb0.
[... 157 lines cut ...]
      encoder         : Lavc57.96.101 aac
[vp8_vaapi @ 0x2a0c840] Encode frame: 1920x1080 (0).
[vp8_vaapi @ 0x2a0c840] Pictures: IDR (0/0)
[vp8_vaapi @ 0x2a0c840] Issuing encode for pic 0/0 as type IDR.
[vp8_vaapi @ 0x2a0c840] No reference pictures.
[vp8_vaapi @ 0x2a0c840] Input surface is 0x4000000.
[vp8_vaapi @ 0x2a0c840] Recon surface is 0x4000003.
[vp8_vaapi @ 0x2a0c840] Allocated output buffer 0x8000000
[vp8_vaapi @ 0x2a0c840] Output buffer is 0x8000000.
[vp8_vaapi @ 0x2a0c840] Param buffer (27) is 0x8000001.
[vp8_vaapi @ 0x2a0c840] Param buffer (27) is 0x8000002.
[vp8_vaapi @ 0x2a0c840] Param buffer (22) is 0x8000003.
[vp8_vaapi @ 0x2a0c840] Param buffer (23) is 0x8000004.
[vp8_vaapi @ 0x2a0c840] Param buffer (11) is 0x8000005.
[hevc @ 0x2a4a040] Decoded frame with POC 2.
[hevc @ 0x2a0a1a0] Decoded frame with POC 6.

Thread 1 "ffmpeg_g" received signal SIGSEGV, Segmentation fault.
0x00007fffedf009ad in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
(gdb) 
```

Regards,

Brainiarc7

QuPengfei, 2017-06-01:
would you try the latest VAAPI version?

the following command worked well in my side.

ffmpeg -vaapi_device /dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -i ~/workspace/stream/avc-improvement/otc_CBR_1920x1080_blue_sky_2M_ipb_01.h264 -an -vf 'format=nv12|vaapi,hwupload' :v 3000 -c:v vp8_vaapi output_vp8.ivf

Detail info:
ffmpeg version N-85704-gcbfd44a Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 6.2.0 (Ubuntu 6.2.0-5ubuntu12) 20161005
  configuration:
  libavutil      55. 61.100 / 55. 61.100
  libavcodec     57. 93.100 / 57. 93.100
  libavformat    57. 72.101 / 57. 72.101
  libavdevice    57.  7.100 / 57.  7.100
  libavfilter     6. 87.100 /  6. 87.100
  libswscale      4.  7.101 /  4.  7.101
  libswresample   2.  8.100 /  2.  8.100
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /opt/X11R7/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
Input #0, h264, from '/home/bqu/workspace/stream/avc-improvement/otc_CBR_1920x1080_blue_sky_2M_ipb_01.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuv420p(progressive), 1920x1080, 25 fps, 25 tbr, 1200k tbn, 50 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> vp8 (vp8_vaapi))
Press [q] to stop, [?] for help
Output #0, ivf, to 'output_vp8.ivf':
  Metadata:
    encoder         : Lavf57.72.101
    Stream #0:0: Video: vp8 (vp8_vaapi) (VP80 / 0x30385056), vaapi_vld, 1920x1080, q=2-31, 3 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc57.93.100 vp8_vaapi
frame=  217 fps=132 q=-0.0 Lsize=    1003kB time=00:00:08.68 bitrate= 946.5kbits/s speed=5.28x
video:1000kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.257334%

Brainiarc7, 2017-06-01:
Alright, let me try again with vaapi 0.40.0.

Brainiarc7, 2017-06-01:
I'll be testing against this PPA as it has the latest VAAPI builds for Ubuntu 16.04LTS at the moment.

QuPengfei, 2017-06-01:
Also VAAPI driver version is 1.8.0+

here is vainfo
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /opt/X11R7/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.40 (libva )
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 1.8.3.pre1 (glk-alpha-51-g5f479ac)
vainfo: Supported profile and entrypoints

Brainiarc7, 2017-06-01:
Hello @QuPengfei,

It worked!

Upgrading to the latest libva and intel-vaapi-driver package fixed the problem.

I can now comfortably close this issue, and thanks for your time :-)

closed 2017-06-01 by Brainiarc7