intel/intel-vaapi-driver #186: Core Dumping Whenever VAAPI is Called

SentToDevNull, 2017-06-02:
On my desktop system, I use a dedicated NVIDIA GPU for graphics such as games and would like to use the graphics acceleration on my Intel CPU with VAAPI to accelerate my ffmpeg encodes. I'm using Debian with non-free packages enabled and all dependencies installed. It's not an issue with the Debian packages because when I compile and install both libva and i965-va-driver myself, the core dump gives the same message.

The i915 kernel module is loaded.

Whenever the VAAPI driver is called, be it from vainfo or ffmpeg, I just get core dumps. Here's an example with output of what I run:

```
user@system$ VDPAU_DRIVER=va_gl LD_PRELOAD=/usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so LIBVA_DRIVER_NAME=i965 vainfo
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/local/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
vainfo: intel_driver.c:106: intel_driver_init: Assertion `VA_CHECK_DRM_AUTH_TYPE(ctx, VA_DRM_AUTH_DRI1) || VA_CHECK_DRM_AUTH_TYPE(ctx, VA_DRM_AUTH_DRI2) || VA_CHECK_DRM_AUTH_TYPE(ctx, VA_DRM_AUTH_CUSTOM)' failed.
```

Here are some commands and output showing that I have the necessary dependencies:

```
# List installed libva packages
user@system$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep libva
libva-dev: 1.7.3-2
libva-drm1: 1.7.3-2
libva-egl1: 1.7.3-2
libva-glx1: 1.7.3-2
libva-tpi1: 1.7.3-2
libva-wayland1: 1.7.3-2
libva-x11-1: 1.7.3-2
libva1: 1.7.3-2
```

```
# The vainfo installed
user@system$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep vainfo
vainfo: 1.7.3-2
```

```
# List installed i915 Intel va driver
user@system$ dpkg-query --showformat='${Package}: ${Version}\n' --show | grep i965
i965-va-driver: 1.7.3-1
```

xhaihao, 2017-06-02:
vainfo is trying to access /dev/dri/renderD128 or /dev/dri/card0. but on your machine, /dev/dri/card0 and /dev/dri/renderD128 should be NVIDIA GPU device.

xhaihao, 2017-06-02:
You can use '-vaapi_device' to specify the device when using ffmpeg.

SentToDevNull, 2017-06-02:
@xhaihao Thanks for getting back to me. I currently am using "-vaapi_device" with ffmpeg when I run the following:
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so LIBVA_DRIVER_NAME=i965 ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -i f.mp4 -vf 'format=nv12|vaapi,hwupload' -threads 8 -c:v hevc_vaapi -preset slower -c:a libopus "f-new.mkv"`

It produces the same VA_CHECK_DRM_AUTH error when calling the Intel vaapi driver, even with those 2 new commits just added.

xhaihao, 2017-06-03:
Could you double check whether /dev/dri/renderD128 is a intel device? 

$> sudo cat  /sys/kernel/debug/dri/128/name | grep i915

xhaihao, 2017-06-03:
Do you mean you still see the same assertion fault after applying https://github.com/01org/intel-vaapi-driver/pull/188?

SentToDevNull, 2017-06-04:
Both card0 and renderD128 seem to be the nvidia card. (I don't think that should be happening. I have both the nvidia and intel DRM modules loaded, so one would expect to see both DRI devices.)

```
user@system$ cat /sys/kernel/debug/dri/128/name
nvidia-drm dev=0000:02:00.0 unique=0000:02:00.0

user@system$ cat /sys/kernel/debug/dri/0/name 
nvidia-drm dev=0000:02:00.0 unique=0000:02:00.0

```

After applying #188, the assertion fault is no longer present. (I hadn't applied #188 before my last comment, sorry.)

Running the same command, I now get the following error though whenever I try to use VAAPI with card0 or renderD128:

```
user@system$ LD_PRELOAD=/usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so LIBVA_DRIVER_NAME=i965 ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -i f.mp4 -vf 'format=nv12|vaapi,hwupload' -threads 8 -c:v hevc_vaapi -preset slower -c:a libopus "f-new.mkv"
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns -1
libva info: User requested driver 'i965'
libva info: Trying to open /usr/local/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva error: /usr/local/lib/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
[AVHWDeviceContext @ 0x56293ddda500] Failed to initialise VAAPI connection: -1 (unknown libva error).
[vaapi @ 0x56293ccabbd0] Failed to create a VAAPI device
```

Without defining the libva driver or device in that command, for both card0 and renderD128, I get the following error:
```
user@system$ ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -i f.mp4 -vf 'format=nv12|vaapi,hwupload' -threads 8 -c:v hevc_vaapi -preset slower -c:a libopus "f-new.mkv"
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[AVHWDeviceContext @ 0x55714635f500] Failed to initialise VAAPI connection: -1 (unknown libva error).
[vaapi @ 0x557143da7bd0] Failed to create a VAAPI device
```

xhaihao, 2017-06-04:
Besides card0 and renderD128, is there any other device file in /dev/dri?

SentToDevNull, 2017-06-04:
Nope. Just card0 and renderD128.

xhaihao, 2017-06-05:
It seems the i915 kernel modules isn't loaded.  Is there a similar message below in your dmesg output?
 
`[    7.059481] [drm] Initialized i915 1.6.0 20170515 for 0000:00:02.0 on minor 0`

SentToDevNull, 2017-06-05:
It shows that the kernel module is loaded:
```
root@system$ lsmod | grep i915
i915                       1232896  0
i2c_algo_bit           16384  1 i915
video                      40960  2 asus_wmi,i915
button                    16384  1 i915
drm_kms_helper   155648  2 i915,nvidia_drm
drm                        360448  5 i915,nvidia_drm,drm_kms_helper
```

When I search dmesg output however, the only drm-related messages are related to the nvidia driver:
```
root@system$ dmesg | grep drm
[   29.017157] [drm] Initialized
[   29.026735] [drm] [nvidia-drm] [GPU ID 0x00000200] Loading driver
```

xhaihao, 2017-06-06:
Maybe your Intel gpu was disabled. could you attach the output of 'lspci -nn' ?

SentToDevNull, 2017-06-07:
Here's the Intel-related output on my system:
```
root@system$ lspci -nn | grep Intel
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:591f] (rev 05)
00:01.0 PCI bridge [0604]: Intel Corporation Skylake PCIe Controller (x16) [8086:1901] (rev 05)
00:01.1 PCI bridge [0604]: Intel Corporation Skylake PCIe Controller (x8) [8086:1905] (rev 05)
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:a2af]
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:a2ba]
00:17.0 RAID bus controller [0104]: Intel Corporation SATA Controller [RAID mode] [8086:2822]
00:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:a2e7] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:a290] (rev f0)
00:1c.4 PCI bridge [0604]: Intel Corporation Device [8086:a294] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:a298] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2c5]
00:1f.2 Memory controller [0580]: Intel Corporation Device [8086:a2a1]
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:a2f0]
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:a2a3]
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (2) I219-V [8086:15b8]
```

xhaihao, 2017-06-08:
Your Intel GPU isn't enabled, you should enable your Intel GPU in the BIOS firstly.

dbermond, 2017-06-22:
I'm having this same issue on a similiar setup: a desktop computer with Intel Haswell i7-4790K and a dedicated nvidia GTX 750 card. The nvidia card is being used as the primary display with proprietary nvidia drivers. The monitor is attached to the nvidia card (single monitor setup). OS: Arch Linux x86_64.

Intel GPU is enabled in bios:
```
$ lspci -nn | grep 'VGA'
00:02.0 VGA compatible controller [0300]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller [8086:0412] (rev 06)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM107 [GeForce GTX 750] [10de:1381] (rev a2)
```

Contents of `/dev/dri`:
```
$ ls -al /dev/dri/
total 0
drwxr-xr-x   2 root root        80 jun 21 19:40 .
drwxr-xr-x  20 root root      3340 jun 21 19:40 ..
crw-rw----+  1 root video 226,   0 jun 21 12:14 card0
crw-rw----+  1 root video 226, 128 jun 21 12:14 renderD128
```

renderD128 is a nvidia device:
```
$ sudo cat /sys/kernel/debug/dri/128/name
nvidia-drm dev=0000:01:00.0 unique=0000:01:00.0
```

The `i915` kernel module is loaded:
```
$ lsmod | grep 'i915'
i915                 1433600  0
intel_gtt              20480  1 i915
i2c_algo_bit           16384  1 i915
video                  36864  1 i915
button                 16384  1 i915
drm_kms_helper        126976  2 i915,nvidia_drm
drm                   299008  5 i915,nvidia_drm,drm_kms_helper
```

Relevant dmesg output:
```
$ dmesg | grep 'i915'
[    5.159272] i915 0000:00:02.0: enabling device (0006 -> 0007)
[    5.169994] [drm] Initialized i915 1.6.0 20170123 for 0000:00:02.0 on minor 1
[    5.171481] snd_hda_intel 0000:00:03.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.199188] i915 0000:00:02.0: fb1: inteldrmfb frame buffer device
[26820.852340] vboxpci: detected device: 8086:0412 at 00:02.0, driver i915
[26821.766440] Modules linked in: bnep ppdev parport snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm iTCO_wdt irqbypass iTCO_vendor_support crct10dif_pclmul crc32_pclmul nls_iso8859_1 ghash_clmulni_intel fuse nls_cp437 mousedev vfat evdev fat pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate intel_rapl_perf pcspkr i915 snd_hda_intel e1000e snd_usb_audio snd_hda_codec lpc_ich i2c_i801 mei_me input_leds snd_usbmidi_lib snd_soc_rt5640 snd_hda_core snd_soc_rl6231 ptp snd_rawmidi snd_soc_ssm4567 intel_gtt mac_hid snd_seq_device mei i2c_algo_bit snd_hwdep shpchp pps_core fan thermal battery snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd snd_soc_sst_acpi soundcore elan_i2c i2c_hid
```

Necessary packages are installed:
```
$ pacman -Qs libva
local/lib32-libva 1.8.1-1
    Video Acceleration (VA) API for Linux (32-bit)
local/lib32-libva-vdpau-driver 0.7.4-5
    VDPAU backend for VA API (32-bit)
local/libva 1.8.2-1
    Video Acceleration (VA) API for Linux
local/libva-intel-driver 1.8.2-1
    VA-API implementation for Intel G45 and HD Graphics family
local/libva-mesa-driver 17.1.3-1
    VA-API implementation for gallium
local/libva-utils 1.8.2-1
    Intel VA-API Media Applications and Scripts for libva
local/libva-vdpau-driver 0.7.4-2
    VDPAU backend for VA API

$ pacman -Qs va-gl                                                                                                     
local/libvdpau-va-gl 0.4.2-1
    VDPAU driver with OpenGL/VAAPI backend
```
[... 75 lines cut ...]

$ LIBVA_DRIVER_NAME='i965' vainfo
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva error: /usr/lib/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

$ LIBVA_DRIVER_NAME='i965' vainfo --display drm
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns -1
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva error: /usr/lib/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```

`ffmpeg` git master gives `Failed to initialise VAAPI connection: -1 (unknown libva error)`. Used `-vaapi_device /dev/dri/renderD128` and the latest git master versions of `libva`, `libva-intel-driver` and `libva-utils`.

Sorry for the long post.

xhaihao, 2017-06-22:
@dbermond You should find out why the Intel device isn't created under /dev/dri firstly.  BTW what is in /sys/bus/pci/devices/0000:00:02.0/drm ?

dbermond, 2017-06-22:
@xhaihao Thank you for the reply.

> You should find out why the Intel device isn't created under /dev/dri firstly.

Somehow, the proper devices mysteriously appeared on `/dev/dri/`. Can't tell what I did or what happened. Maybe a temporary boot problem? Who knows...
The fact is that now I'm getting this:

```
$ lspci | grep 'VGA'
00:02.0 VGA compatible controller: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller (rev 06)
01:00.0 VGA compatible controller: NVIDIA Corporation GM107 [GeForce GTX 750] (rev a2)

$ ls -al /dev/dri/
total 0
drwxr-xr-x   2 root root       120 jun 22 13:29 .
drwxr-xr-x  20 root root      3320 jun 22 13:29 ..
crw-rw----+  1 root video 226,   0 jun 22 13:29 card0
crw-rw----+  1 root video 226,   1 jun 22 13:29 card1
crw-rw----+  1 root video 226, 128 jun 22 13:29 renderD128
crw-rw----+  1 root video 226, 129 jun 22 13:29 renderD129

$ sudo cat /sys/kernel/debug/dri/128/name
nvidia-drm dev=0000:01:00.0 unique=0000:01:00.0

$ sudo cat /sys/kernel/debug/dri/129/name
i915 dev=0000:00:02.0 unique=0000:00:02.0
```

> BTW what is in /sys/bus/pci/devices/0000:00:02.0/drm ?

```
$ ls -al /sys/bus/pci/devices/0000:00:02.0/drm
total 0
drwxr-xr-x  4 root root 0 jun 22 13:29 .
drwxr-xr-x 13 root root 0 jun 22 13:29 ..
drwxr-xr-x  7 root root 0 jun 22 13:29 card1
lrwxrwxrwx  1 root root 0 jun 22 13:35 controlD65 -> card1
drwxr-xr-x  3 root root 0 jun 22 13:29 renderD129

$ ls -al /sys/bus/pci/devices/0000:01:00.0/drm
total 0
drwxr-xr-x 4 root root 0 jun 22 13:29 .
drwxr-xr-x 9 root root 0 jun 22 13:29 ..
drwxr-xr-x 3 root root 0 jun 22 13:29 card0
drwxr-xr-x 3 root root 0 jun 22 13:29 renderD128
```

As you can see, everything seems to be in place now. But `vainfo` and `ffmpeg` are still showing the same errors. Hardware accelerated encoding with intel vaapi is not working.

```
$ LIBVA_DRIVER_NAME='i965' vainfo
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva error: /usr/lib/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

$ LIBVA_DRIVER_NAME='i965' ffmpeg -v error -vaapi_device /dev/dri/renderD129 -i input.mp4 -vf 'format=nv12,hwupload' -c:v h264_vaapi -c:a copy -y out.vaapi.mp4
libva info: VA-API version 0.99.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib64/dri/i965_drv_video.so
libva error: /usr/lib64/dri/i965_drv_video.so has no function __vaDriverInit_0_32
libva info: va_openDriver() returns -1
[AVHWDeviceContext @ 0x556439810f40] Failed to initialise VAAPI connection: -1 (unknown libva error).
Segmentation fault (core dumped)
```

xhaihao, 2017-06-23:
@dspmeng  vainfo only check /dev/dri/card0 and /dev/dri/renderD128, so it is still showing the errors. I added a new option to vainfo, could you please try https://github.com/01org/libva-utils/pull/46?  

$> vainfo --display drm --device /dev/dri/renderD129

xhaihao, 2017-06-23:
> $ LIBVA_DRIVER_NAME='i965' ffmpeg -v error -vaapi_device /dev/dri/renderD129 -i input.mp4 -vf 'format=nv12,hwupload' -c:v h264_vaapi -c:a copy -y out.vaapi.mp4
> libva info: VA-API version 0.99.0
> libva info: va_getDriverName() returns 0

The VA API version was 0.99, not 0.40.0 in this case,  /usr/lib64/dri/i965_drv_video.so doesn't work with VA-API 0.99

dbermond, 2017-06-23:
`vainfo` is working nice after applying [01org/libva-utils#46](https://github.com/01org/libva-utils/pull/46). It works even without specifying `LIBVA_DRIVER_NAME='i965'`. It's a pleasure to see it working now:

```
$ vainfo --display drm --device /dev/dri/renderD129
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.40 (libva )
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 1.8.4.pre1 (glk-alpha-71-gc3110dc)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
```

As a suggestion, I think that it would be nice to see these `--display` and `--device` options to be documented. They seem to be somewhat hidden by now. Something like a `--help` switch to show the available options and/or a manpage would be good.

Regarding `ffmpeg`, that's true. My custom `ffmpeg` build was wrongly linking against an old version of libva and I wasn't even aware of it. Intel vaapi hardware accelerated video encoding worked fine after adjusting my `ffmpeg` build.

For me it's everything solved, as long as [01org/libva-utils#46](https://github.com/01org/libva-utils/pull/46) get merged. Thank you @xhaihao for your attention and support.

xhaihao, 2017-07-18:
https://github.com/01org/libva-utils/pull/46 was merged, now user can use '--device <device>' to specify the dri device when drm is used. Per the comments from @dbermond, the VAAPI driver works /dev/dri/renderD129

closed 2017-07-18 by xhaihao

aenertia, 2018-06-15:
Is there a way to force use of that backend in the environment like the LIBVA_DRIVER_NAME= variable? for the --device and --display drm options?

I am trying to get libfreenect2 to use the vaapi via intel path (it uses the nvidia/vavdpau unsupport jpeg offload profile for the rgb stream). There are other programs which I would like to inherit the --device and --display options when calling vaapi