intel/intel-vaapi-driver #204: howto get g45-h264 source branch from new release

andysendo, 2017-06-18:
g45-h264 source banch is very useful, cant find from this repo, SourceForge.net cant download anymore.

xhaihao, 2017-06-18:
@andysendo, we don't maintain g45-h264 branch any more. you can find g45-h264 branch on people.freedesktop.org/~haihao/libva if you want to use g45-h264.

andysendo, 2017-06-18:
Browse people.freedesktop.org/~haihao/libva get 403 Forbidden error.
Howto get the source?

xhaihao, 2017-06-19:
Please navigate your browser to https://cgit.freedesktop.org/~haihao/libva/, you will find the branch names in the top of the page and git repo URL in the bottom of the page. You should contact your IT administrator if you are behind a firewall.

xhaihao, 2017-06-25:
@andysendo did you get the source code?

andysendo, 2017-06-25:
yes, i get it. 
i think its too old so there are some compile error.
i get source from https://aur.archlinux.org/packages/libva-intel-driver-g45-h264/ thats newer and runs ok.

Thanks for help.

closed 2017-06-26 by xhaihao

mirh, 2017-09-13:
I was wondering.. Why g45-h264 has never been 'mainlined' in 6 years?

xhaihao, 2017-09-14:
It doesn't play some 1080p videos smoothly.

mirh, 2017-09-14:
I have read that can happily be addressed by increasing igp dedicated memory
...
But even if that wasn't the case, why couldn't one just limit acceleration to 720p or lower?