intel/intel-vaapi-driver #217: High Frame Rate video - loosing/skiping frames

piotrasd, 2017-07-02:
Hi
im using latest intel-vaapi-driver 1.8.3 
(kernel 4.12rc7 and 4.11)
mesa 17.1.4
xf86-video-intel - latest git

and tested with kodi, and latest chromium
video playback 

Im noticed - movies HFR - with 60fps in 4k dropping skipping frames

here sample:
https://www.sendspace.com/file/8cryl0

rest i think you can test on Youtube
some videos was 4k VP9 HFR - but not possible in HFR watch, was fine
when i changed resultion to 1440 then picture was smooth 

i can delivery some logs but i never before debug vaapi co please give me some tip thanks.

QuPengfei, 2017-07-04:
@piotrasd would you try it with yamidecode, which is a locale tool ? 
https://github.com/01org/libyami
https://github.com/01org/libyami-utils

QuPengfei, 2017-07-04:
@piotrasd i can not open the link. https://www.sendspace.com/file/8cryl0
it show "This site can’t be reached"

xhaihao, 2017-07-13:
@piotrasd could you provide some device info, e.g. the graphics PCI id? You may run the command below 
$> lspci -nn | grep VGA

BTW I also failed to access https://www.sendspace.com/file/8cryl0