intel/intel-vaapi-driver #218: error: 'struct VADriverVTableWayland' has no member named 'has_prime_sharing'

mausspb, 2017-07-04:
I try install vaapi driver in docker ubuntu, command - 
`git clone https://github.com/01org/intel-vaapi-driver &&\
cd intel-vaapi-driver &&\
./autogen.sh &&\
./configure --enable-hybrid-codec &&\
make -j$(nproc) VERBOSE=1 &&\
sudo make -j$(nproc) install &&\
sudo ldconfig -vvvv`
And i get next error message :
`i965_output_wayland.c: In function 'va_GetSurfaceBufferWl':
i965_output_wayland.c:267:16: error: 'struct VADriverVTableWayland' has no member named 'has_prime_sharing'
     if (!vtable->has_prime_sharing || (drm_intel_bo_gem_export_to_prime(obj_surface->bo, &fd) != 0)) {
                ^
Makefile:1325: recipe for target 'libi965_drv_video_la-i965_output_wayland.lo' failed
make[3]: *** [libi965_drv_video_la-i965_output_wayland.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/root/ffmpeg_sources/intel-vaapi-driver/src'
Makefile:1344: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/root/ffmpeg_sources/intel-vaapi-driver/src'
Makefile:657: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/root/ffmpeg_sources/intel-vaapi-driver/src'
Makefile:421: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1`

What is the problem?

xhaihao, 2017-07-10:
@mausspb, you should update your libva firstly.

mausspb, 2017-07-10:
ok, i try it

xhaihao, 2017-07-13:
Does the issue disappear?

mausspb, 2017-07-13:
Hello.
I have paused this task so far, I will return to this task next week.

xhaihao, 2017-08-04:
@kvpramod91 could you help to check whether struct VADriverVTableWayland in va_backend_wayland.h has 'unsigned int has_prime_sharing' ?

uartie, 2017-08-04:
@xhaihao that's odd... shouldn't configure fail if libva is not at least 1.7.3 (which is when has_prime_sharing was added)?  That is, users shouldn't even get as far as "make".

uartie, 2017-08-04:
@mausspb what is the current commit id of intel-vaapi-driver after you "git clone" when this happens?

xhaihao, 2017-08-04:
@uartie according to @kvpramod91's description on https://github.com/01org/intel-vaapi-driver/issues/243, the latest libva was installed.

uartie, 2017-08-04:
Ah yes, libva from source was probably installed in /usr/local since that is default for autogen/configure prefix but driver does not look there during autogen/configure by default on ubuntu.  Thus it finds system installed libva in /usr/.../x86_64-linux-gnu.

Still seems like there is a configure.ac issue since it should reject system libva.  Was libva API version bumped when has_prime_sharing was introduced?  I see package version was bumped, but driver does not consider the package version during configure.

uartie, 2017-08-04:
nonetheless, i can't seem to reproduce.

what version of ubuntu container is this?

xhaihao, 2018-01-18:
Closed as no response from reporter. Please feel free to reopen this issue if you still experience the bug.

closed 2018-01-18 by xhaihao