intel/intel-vaapi-driver #220: Missing source for shaders

sebastinas, 2017-07-05:
It appears as if the released tarball and the git repository do not contain sources for all (compiled) shaders. For the 1.8.3 tarballs the sources are missing for

```
src/shaders/brc/bsw/hme_downscale_genx_0.g8b
src/shaders/brc/bsw/hme_genx_0.g8b
src/shaders/brc/bsw/vp8_brc_init_genx_0.g8b
src/shaders/brc/bsw/vp8_brc_reset_genx_0.g8b
src/shaders/brc/bsw/vp8_brc_update_genx_0.g8b
src/shaders/brc/bsw/vp8_enc_genx_0.g8b
src/shaders/brc/bsw/vp8_enc_genx_1.g8b
src/shaders/brc/bsw/vp8_enc_genx_2.g8b
src/shaders/brc/bsw/vp8_intra_distortion_genx_0.g8b
src/shaders/brc/bsw/vp8_mpu_genx_0.g8b
src/shaders/brc/bsw/vp8_tpu_genx_0.g8b
src/shaders/brc/skl/hme_downscale_genx_0.g9b
src/shaders/brc/skl/hme_genx_0.g9b
src/shaders/brc/skl/vp8_brc_init_genx_0.g9b
src/shaders/brc/skl/vp8_brc_reset_genx_0.g9b
src/shaders/brc/skl/vp8_brc_update_genx_0.g9b
src/shaders/brc/skl/vp8_enc_genx_0.g9b
src/shaders/brc/skl/vp8_enc_genx_1.g9b
src/shaders/brc/skl/vp8_enc_genx_2.g9b
src/shaders/brc/skl/vp8_intra_distortion_genx_0.g9b
src/shaders/brc/skl/vp8_mpu_genx_0.g9b
src/shaders/brc/skl/vp8_tpu_genx_0.g9b
src/shaders/h264/mc/avc_mc.g4b
src/shaders/post_processing/gen75/sharpening_h_blur.g75b
src/shaders/post_processing/gen75/sharpening_unmask.g75b
src/shaders/post_processing/gen75/sharpening_v_blur.g75b
src/shaders/post_processing/gen8/conv_nv12.g8b
src/shaders/post_processing/gen8/sharpening_h_blur.g8b
src/shaders/post_processing/gen8/sharpening_unmask.g8b
src/shaders/post_processing/gen8/sharpening_v_blur.g8b
src/shaders/post_processing/gen9/conv_nv12.g9b
src/shaders/post_processing/gen9/conv_p010.g9b
src/shaders/post_processing/gen9/pa_to_pa.g9b
src/shaders/post_processing/gen9/pa_to_pl2.g9b
src/shaders/post_processing/gen9/pa_to_pl3.g9b
src/shaders/post_processing/gen9/pl2_to_pa.g9b
src/shaders/post_processing/gen9/pl2_to_pl2.g9b
src/shaders/post_processing/gen9/pl2_to_pl3.g9b
src/shaders/post_processing/gen9/pl2_to_rgbx.g9b
src/shaders/post_processing/gen9/pl3_to_pa.g9b
src/shaders/post_processing/gen9/pl3_to_pl2.g9b
src/shaders/post_processing/gen9/pl3_to_pl3.g9b
src/shaders/post_processing/gen9/rgbx_to_nv12.g9b
```
(List taken from https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=867340)

Also, when looking at the current master branch, some sources are missing. If you one deletes all compiled shaders contained in the repository, at least some are not re-generated:

```
src/shaders/brc/bsw/vp8_brc_init_genx_0.g8b
src/shaders/brc/skl/vp8_brc_init_genx_0.g9b
src/shaders/post_processing/gen75/sharpening_h_blur.g75b
src/shaders/post_processing/gen9/conv_p010.g9b
src/shaders/h264/mc/avc_mc.g4b
src/shaders/post_processing/gen5_6/null.g6b
src/shaders/post_processing/gen8/conv_nv12.g8b
```

(Listen taken from `find -name "*.g*b" -delete; autoreconf -vfi; ./configure; make -i`).

Please ensure that all sources are contained in the git repository and the released tarballs.

QuPengfei, 2017-07-06:
@sebastinas That is true there are no source codes for some shaders. it have been long time. Actually, for some reason, some source of shader are private and not allowed to upstream. Maybe you can understand this situation.

sebastinas, 2017-07-06:
This is unfortunate and a very serious issue for Debian (see [DFSG](https://www.debian.org/social_contract#guidelines)) and we need to resolve this issue somehow. The easiest solution for us - of course - would be the inclusion of the respective sources. But if this is impossible, as you say, Debian has some possible ways forward:

1. We strip shaders without source and disable the code paths that depend on those shaders. So the question is, which features we would loose by simply removing them. But assuming that this would only affect the encoders, we could probably live with that and direct any users that need hardware accelerated encoding to you.
2. We strip shaders without source, but move it to a non-free package similar to what we do for the kernel and proprietary firmware.
3. We remove the driver from Debian.

The last option is the least desirable one as it's a disservice to our users. 1. is probably the easiest one to implement for us provided that the shaders are not essential for decoding. 2. would be doable but would take some time.

So let's start with the easy questions: if we'd remove the affected shaders, what would break?

In any case, it'd be great if we could fix this issue somehow without too much inconvenience to our users. I'll also talk to the co-maintainers of the package in Debian. Maybe they have some other ideas on possible ways forward.

sebastinas, 2017-07-06:
Looks like we also need to add `src/gen9_avc_encoder_kernels.c`, `src/gen9_hevc_enc_kernels_binary.c` and `src/gen9_vpc9_encoder_kernels.c` to the list.

QuPengfei, 2017-07-07:
@sebastinas those shaders mainly affect the encoder,such as VP8/AVC/HEVC.

For solution 2, 
- is there guide how to move the shaders without source into a non-free package? what is the package name and where to put？
-  what does this solution affect the users or customers?

sebastinas, 2017-07-08:
I don't think there is a guide I can point you to. But what we basically need would be something like:

* The shader package installs all the shaders to same fixed path, say `$datadir/i965-va-driver-shaders` (if the shaders are architecture independent) or `$libdir/i965-va-driver-shaders` (if the the shaders are architecture dependent). The choice probably also depends on the format of the data.
  * If you just want to move the symbols defining the shaders to a different so, I'd put them somewhere in `$libdir/i964-va-driver-shaders`, `dlopen` the file during initialization of the driver and get the shaders with `dlsym` if available. Given the current representation of the shaders, this looks like a relatively easy solution to implement.
  * Another option would be to store the shaders in some file and then `fopen` and `fread` them during startup.
* In case loading of any of the  shaders fail, the driver could then simply not offer VP8/AVC/HEVC encoding.

Users then need to install one extra package to get the encoders. The existing package can suggest the new package and we will document in Debian's release notes, that users how need support for hardware accelerated encoding need to install this extra package.

sebastinas, 2017-10-03:
Is there any progress on this issue? Do you need help with implementing a solution?

We should get this fixed before the release of 2.0.0. This could become a major blocker otherwise.

xhaihao, 2017-10-05:
You will be appreciated if you could help to implement it.  BTW this is not an API/ABI compatibility issue, so I'd like to fix it in 2.x.x, not in 2.0.0.

sebastinas, 2017-10-16:
You can follow the progress at https://github.com/sebastinas/intel-vaapi-driver/tree/external-kernels

sebastinas, 2017-10-30:
FWIW, a build with that patch set is not available in the Debian archive. The extra shared library is still awaiting review by our FTP masters.

sebastinas, 2018-02-15:
@xhaihao  So what's the status of this issue? I see new shaders merged without source instead of this issue getting fixed.

sebastinas, 2018-09-29:
This is my current branch: https://github.com/sebastinas/intel-vaapi-driver/tree/2.2.0-remove-non-free-shaders. I'd love some feedback - especially with respect to the removed `.has_XYZ` flags.

uis246, 2020-12-14:
What wrong with vp8?