intel/intel-vaapi-driver #222: vaCreateSurfaces asserts when using a external buffer

chivakker, 2017-07-08:
assert is on i965_drv_video.c:1443.

There's an assumption few lines before where height is calculated that data_size is not zero.  Client may not populate it but instead pitches and offsets.  When it is one plane that'll result in height being zero for obj_surface and the assert will be triggered

Please use pitches and offsets to calculate obj_surface->height properly

This is a major regression as no external buffer can be used.

xhaihao, 2017-07-11:
There isn't an assert() on line 1443 in i965_drv_video.c, which version of intel-vaapi-driver are you using?

sreerenjb, 2017-07-11:
Ideally, the middleware should submit the data_size if it trying to create a buffer from  external memory(dmabuf or gem). But nevertheless, it would be good if the driver can calculate the data_size based on user provided VASurfaceAttribExternalBuffers width/height/pitches in case user forgot to populate the data_size, or at least driver should error out.

chivakker, 2017-08-29:
There are still problems with the asserts when input information comes in more that 1 plane.  For 1080p resolution, height is not 16 aligned i.e. 1080 is not aligned while 1088 is.  

Can this be revisited?, It looks like the asserts are not really helping here.

chivakker, 2017-08-30:
I have the following information coming from the application:  obj_surface height is 360 which is not 16 aligned. tiling is 0 and num_planes are 3.

xhaihao, 2018-01-18:
The issue has been fixed.

closed 2018-01-18 by xhaihao