intel/intel-vaapi-driver #233: Regression of  quality in case of AVC encoded(CQP)

zzoon, 2017-07-21:
Hi, I'm from https://bugzilla.gnome.org/show_bug.cgi?id=784318

As per my investigation, the quality of AVC encoded got lower than before even if I set 1 to queality level.
(Currently default is 4 in gst-vaapi, it results in very bad quality in cases in the issue above, and if I set 1 to quality level, it gets better but still not able to meet the pass standard)

If I go back before the commit 481ec15, the quality is good enough to pass.
I'm wondering if this is regression in the driver, or there's something more to do in gst-vaapi.
Could you clarify this?

zzoon, 2017-07-21:
In addition, I have tested with libyami and seems I got same result.

xhaihao, 2017-07-21:
@FocusLuo  did you see this issue on SKL+?

FocusLuo, 2017-07-21:
I guess it is not a regression issue.
You need to use the bitreate-psnr curve to figure out the video quality
You will notify that when you use the commit 481ec15, the encoded bitstream file will consume more less bitrate than pre-commit 481ec15
[image]

zzoon, 2017-07-24:
@FocusLuo Sorry I don't get it.
Do you mean it has worse quality with the commit 481ec15 than pre-commit 481ec15?
If so, that's exactly what I want to say in this thread.

FocusLuo, 2017-07-26:
@zzoon , Oh, no, my mean is that the quality is almost the same quality as before from our test result. For quality compare, you need to draft the bitrate-psnr curve between the commit 481ec15 than pre-commit 481ec1. 
You can't just focus on only the psnr value. The new commit got the lower psnr, but it also cost more little bitrate than before.

zzoon, 2017-07-26:
@FocusLuo Thanks for clarification. I believe you're right.
The reason that I mentioned "quality" is that I can tell the difference of quality of encoded data just by my eyes as human being.

FocusLuo, 2017-07-26:
@zzoon Ok, let me to close this issue:). If you find any other issue in future, please help to open a new one.

closed 2017-07-26 by FocusLuo