intel/intel-vaapi-driver #238: Intel iGPU VAAPI Support for VC-1 1080i

lqvnguyen, 2017-07-28:
Under Windows there is hardware acceleration for VC-1 1080i. Can we look into adding VAAPI support for VC-1 1080i on Linux?

lqvnguyen, 2017-08-03:
Similar request previously opened and closed:
https://github.com/01org/intel-vaapi-driver/issues/12

This works under Windows on the same hardware. You really need to consider this as Linux users are not small. They are simply more resourceful and using CPU decoding as a workaround.

Kodi is very big on normal Linux, Kodi for LibreELEC and embedded PMP. These are popular platforms and are only increasing in popularity. Nvidia has this working on a $200 Nvidia Shield. Many users are abandoning Intel's counterpart, the NUC, because of all the issues like this.

You guys have a resolution to the VC-1 macro blocking artifacts now. Please consider working on closing the loop with the deficiencies of VC-1 on Intel GPUs by supporting VC-1 1080i.

carpalis, 2017-10-27:
Good news: I have been working on implementing VC-1 interlaced hardware decoding. You can find the code in my GitHub repository: carpalis/intel-vaapi-driver in branch vc1interlace or vc1interlace-1.8

You will need to build a new libva and ffmpeg which are also in my repository. I have got to the point that I can play the Kodi 1080i sample file VC-1_29.970_sample.mkv without issues on Haswell platform. Multi-slice should work, but is not yet implemented in my ffmpeg branch. I am also not sure if intensity compensation works in all cases.

I have no other hardware to test on, so please test the code before I will try to get this stuff upstream. Anything that takes the gen7 code path should work. Sandy Bridge is not implemented due to the lack of publicly available hardware documentation.

Regards,
Jerome Borsboom

lqvnguyen, 2017-11-09:
What else do you need for testing?

carpalis, 2017-11-10:
I have been using the code for the last to weeks to watch my VC-1 1080i content through Kodi. All the content showed no glitches or other problems when hardware decoding on my gen75 hardware. The best thing people could do is just compile and use the code and report back if anything is not decoded cleanly in hardware.

I will submit a pull request in the coming days to get this stuff into mainline as that seems the best way to get more exposure and, hence, testing of the code.

lqvnguyen, 2018-02-06:
Hi Jerome,

Did you submit your ffmpeg commits to the ffmpeg mailing lists?

carpalis, 2018-02-07:
The patches for ffmpeg are not yet submitted. I wait for the release of a version of the driver and libva library that contains the interlaced decoding code. The changes in ffmpeg need to be made conditional on this version. Otherwise ffmpeg will not compile with an older version of libva.

xhaihao, 2018-02-12:
libva / intel-vaapi-driver 2.1.0 supports decoding interlaced vc1, the VA-API version is 1.1.0.

carpalis, 2018-02-25:
I have submitted the changes to the ffmpeg-devel mailinglist a few minutes ago.

lqvnguyen, 2018-04-22:
Did they accept your changes?