intel/intel-vaapi-driver #240: gpu hang occurs when run hevc cbr encode quality test on kbl platform

wangzj0601, 2017-07-31:
System Environment
linux kernel:  4.12.0-rc2
system: Ubuntu 14.04
libyami: 1.2.0
libva: commit b65a3d9deacb72bd9f96d3e023d4f8ed58bd8d0a
intel driver: commit 7f50a13b4eb93ddd2a52b2ebb1c2f7554c23ccb8

HW:
KBL-U
CPU: Dual core; Genuine Intel(R) CPU 0000 @ 2.60GHz
GPU: PCI ID 8086:5916, GT2
Memory: 6G

Reproduce Steps
1. configure the environment and install the required driver as above
2. execute encode_quality_hevc_TUdef_main script, gpu hang occurs during run hevc cbr encode quality test

dmesg info when gpu hang
root@media_libva_kbl:/home/focus/media/media_encode_quality/encode_hevc/otc_yami# dmesg |grep hang
[    7.723563] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[ 1655.812153] [drm] GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace.
[ 1655.812154] [drm] The gpu crash dump is required to analyze gpu hangs, so please always attach it.
[ 1655.812251] drm/i915: Resetting chip after gpu hang
[ 1663.806056] drm/i915: Resetting chip after gpu hang
[ 1671.805898] drm/i915: Resetting chip after gpu hang
[ 8509.789999] drm/i915: Resetting chip after gpu hang
[ 8517.822032] drm/i915: Resetting chip after gpu hang
[ 8525.822020] drm/i915: Resetting chip after gpu hang
[ 8533.822131] drm/i915: Resetting chip after gpu hang
[11817.790046] drm/i915: Resetting chip after gpu hang

process when gpu hang
root      1525  0.0  0.0  16648  2216 pts/3    S+   10:27   0:00 /bin/bash ./encode_quality_hevc_TUdef_main.sh
root      1529  0.0  0.0  16900  2576 pts/3    S+   10:27   0:00 /bin/bash ./hevc_encoding_cbr_main.sh GOP30_TU4
root      9969  0.3  0.0      0     0 ?        S    13:05   0:09 [kworker/u8:4]
root     10067  0.0  0.1 109800  6500 ?        Ss   13:13   0:00 sshd: root@pts/5
root     10145  0.0  0.1  27292  6084 pts/5    Ss   13:13   0:00 -bash
root     10341  0.0  0.0      0     0 ?        S    13:23   0:00 [kworker/2:1]
root     10380  0.0  0.0      0     0 ?        S    13:30   0:00 [kworker/0:2]
root     10428  0.0  0.0      0     0 ?        S    13:30   0:00 [kworker/1:1]
root     10431  0.0  0.0  48296  5012 pts/5    S+   13:31   0:00 ssh x-build32
root     10471  0.0  0.0      0     0 ?        S    13:36   0:00 [kworker/2:0]
root     10472  0.0  0.0      0     0 ?        S    13:36   0:00 [kworker/0:1]
root     10621  0.0  0.0      0     0 ?        S    13:37   0:00 [kworker/1:0]
root     10630  0.0  0.0      0     0 ?        S    13:37   0:00 [kworker/3:2]
root     11042  0.0  0.0      0     0 ?        S    13:37   0:00 [kworker/u8:2]
root     12985  0.0  0.0  16724  3336 pts/3    S+   13:41   0:00 /bin/bash ./hevc_encoding_cbr.sh 480P_webRTC_rc1_gop30_fp
root     13031 98.4  0.7 1444524 42724 pts/3   R+   13:41   8:24 yamiencode -c HEVC -s I420 -i /root/media_streams/encoder
root     13033  0.0  0.0      0     0 ?        S    13:43   0:00 [kworker/3:0]
root     13035  0.0  0.0      0     0 ?        S    13:46   0:00 [kworker/u8:0]
root     13036  0.0  0.0  22656  2764 pts/1    R+   13:49   0:00 ps -aux

wangzj0601, 2017-08-01:
this issue also can be duplicate on kbl nuc platform.

pengche1, 2017-08-01:
@wangzj0601 ,please tell me this issue is caused by which command or in random? thanks!

wangzj0601, 2017-08-01:
@pengche1, I have modified the repro step in bug. FYI.
Reproduce Steps

configure the environment and install the required driver as above
run command yamiencode -c HEVC -s I420 -i /root/media_streams/encoderbitstreams/encoderbitstreams_quality/480P_webRTC_video_conference_640x480_822frames.yuv -W 640 -H 480 --rcmode CBR -b 5000 --ipperiod 2 --intraperiod 30 -f 30 --quality-level 4 -o ./480P_webRTC_rc1_gop30_fps30_bps5000_ipb_tu4.h265
use dmesg |grep hang

closed 2017-08-08 by xhaihao

pengche1, 2017-08-08:
@wangzj0601 , this patch has been merged. please check this issue again.

wangzj0601, 2017-08-10:
This issue can not be duplicated with then new patch merged on master branch commit 6b26f3fb7367836291db40dd6e36cf6762ab32c8 on kbl platform.

PR #242 (2017-08-08): Update new media kernels for HEVC encoder
Fixes #240
commit 8b8a0f6d (2017-08-08): Update new media kernels for HEVC encoder

Fixes #240