intel/intel-vaapi-driver #248: Error while compiling...  i965_drv_video.c: In function ‘i965_GetConfigAttributes’

jkepp, 2017-08-12:
i receive an error while compiling:

`i965_drv_video.c: In function ‘i965_GetConfigAttributes’:
i965_drv_video.c:1105:45: error: ‘struct <anonymous>’ has no member named ‘roi_rc_qp_delat_support’; did you mean ‘roi_rc_priority_support’?
                             roi_config->bits.roi_rc_qp_delat_support = 0;
                                             ^
i965_drv_video.c:1110:45: error: ‘struct <anonymous>’ has no member named ‘roi_rc_qp_delat_support’; did you mean ‘roi_rc_priority_support’?
                             roi_config->bits.roi_rc_qp_delat_support = 1;
                                             ^
i965_drv_video.c:1118:37: error: ‘struct <anonymous>’ has no member named ‘roi_rc_qp_delat_support’; did you mean ‘roi_rc_priority_support’?
                     roi_config->bits.roi_rc_qp_delat_support = 1;
                                     ^
i965_drv_video.c:1136:37: error: ‘struct <anonymous>’ has no member named ‘max_num_temporal_layers_minus1’; did you mean ‘num_temporal_layers_minus1’?
                     val_config->bits.max_num_temporal_layers_minus1 = MAX_TEMPORAL_LAYERS - 1;
                                     ^
i965_drv_video.c:1137:37: error: ‘struct <anonymous>’ has no member named ‘temporal_layer_bitrate_control_flag’
                     val_config->bits.temporal_layer_bitrate_control_flag = 1;
`

xhaihao, 2017-08-14:
Could you please update libva (https://github.com/01org/libva)?

xhaihao, 2017-08-23:
@jkepp Does the issue disappear?

xhaihao, 2017-09-28:
I closed this issue because no response from the reporter. Please feel free to reopen or file a new issue if you still experience this issue.

closed 2017-09-28 by xhaihao