intel/intel-vaapi-driver #262: VP9 Decoding Looks Awful

Queuecumber, 2017-08-31:
No idea whats causing this or if it's something fixable but I am trying to use VAAPI to decode VP9 video and it looks really bad. I grabbed the test video from [here](https://base-n.de/webm/out9.webm) and did two tests in Chromium and in `mpv` both with VAAPI enabled. 

In either case I get an image similar to

[image]

This particular image was produced with `mpv --hwdec=vaapi Downloads/out9.webm`

For reference image should look something like 

[image]

Is there any way to fix this either through configuration or code? Apologies in advance it this was already reported. 

My processor model is listed as 

> Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz

in /proc/cpuinfo  and I am running Ubuntu 17.04.

pengche1, 2017-09-01:
@Queuecumber , it seems it is the similar issue I am working on. could you tell me what version is your Linux kernel? I also have this issue on 4.13-rc7, but it just disappears on 4.12-X.
Also, please lspci | grep VGA and add it in the comment.

pengche1, 2017-09-04:
Could you help me to try this patch #263? thanks!

Queuecumber, 2017-09-10:
Yes that seems to be the problem, I was on 4.12-rc(something) when I noticed the issue and assumed I just hadn't played a VP9 video until then. I just booted into 4.12.7 and the same video plays fine with vaapi. So I guess we're calling it a 4.13 regression then. I'd love to test the patch, is this a kernel patch or something else?

Queuecumber, 2017-09-10:
Oh and for completeness sake, `lspci | grep VGA` gives

```
00:02.0 VGA compatible controller: Intel Corporation Device 591b (rev 04)
11:00.0 VGA compatible controller: NVIDIA Corporation GP104 [GeForce GTX 1080] (rev a1)
```

but ignore the nvidia card I don't use it for the desktop environment

pengche1, 2017-09-11:
#263 is a intel-vaapi-driver patch. it is a potential vaapi driver issue. And it just appears when migrate to 4-13-xx Linux kernel.

xhaihao, 2017-09-18:
I closed this issue because https://github.com/01org/intel-vaapi-driver/pull/263 has been merged, please feel free to reopen this issue or file a new issue if you still experience this issue with master branch.

closed 2017-09-18 by xhaihao

MatMaul, 2017-10-16:
Any chance to see 1.8-branch updated with this kind of bugfixes integrated ? If I understand correctly we need libva v2 for master, which could take a while to reach various downstreams.

xhaihao, 2017-10-16:
I cherry-picked some patches including the patch to fix this VP9 issue from master to v1.8-branch.

MatMaul, 2017-10-16:
Awesome thanks :) Do you plan to tag a new 1.8.x bugfix release at some point ?

chivakker, 2017-11-11:
@MatMaul there's no plan to cut another release on the 1.8.x family.  Can you carry this patch as an exception on your setup?

MatMaul, 2017-11-28:
@chivakker I can just use the branch for my purpose (flatpak packages) however I think it would be nice to cut a release otherwise those fixes will probably never reach other downstream packages before they switch to 2.x, which could take a while. Your call :)

MatMaul, 2017-12-26:
Can I ask a backport of 9d66570032fb02b1e79a883af7697b035d700a8e on 1.8-branch, which seems to be a follow up of this one ?

@chivakker it would actually be really nice to cut a release, I didn't experienced the bug before but it seems most of (all?) the WebM VP9 files coming from YouTube exhibit the problem. Which represents I guess 90%+ of all VP9 files available :)
Same reason as before, if you don't cut a release those files will look like garbage on all current distributions forever.

xhaihao, 2018-01-08:
fcf9041b29cee938a228dc7950955ccebb43c6bf was merged into v1.8-branch as well.

vali20, 2018-08-21:
@MatMaul Thank you very much for the request, thanks to it being back ported, I was able to compile libva 1.7.0 (0.39.4), and intel-vaapi-driver 1.8.3 from source, and now VP9 decoding works fine in Chromium VAAPI on Ubuntu 16.04.5. A million thanks for addressing this issue, not everyone is able to upgrade to the latest LSB, especially in such short time, and having that broken was pretty sad since YouTube is basically VP9 these days.