intel/intel-vaapi-driver #264: GPU hang in decoding hevc video by using ffmpeg

hsyrja, 2017-09-07:
GPU hang occurred when hw accelerated decoding is issued by ffmpeg and video codec is Hevc encoded. 

ffmpeg command (decode only)
ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -I hang.vid -f null -

Tested in Skylake as wel Kabyake i7. 

Here is the input video for reproduce the hang.
[hang.zip](https://github.com/01org/intel-vaapi-driver/files/1284329/hang.zip)

Behaviour is random hang won't happen every time when it's happen it is always in ffmpeg start.

I don't know if this is intel-vaapi-driver issue or FFmpeg issue so I will file bug to both projects.

xhaihao, 2017-09-08:
1.  I saw lots of error messages when using a tool to parse hang.vid
2.  I can't reproduce this issue with gstreamer-vaapi, it seems ffmpeg and gstreamer handle the element errors in different way so they pass different picture parameters and slice parameters to the driver.  You may enable VA tracer by setting LIBVA_TRACE in your environment then check the tracer log.  
 
e.g.  in gstreamer
 
[48642.979908][ctx 0x02000000]  pcm_sample_bit_depth_luma_minus1 = 0
[48642.979909][ctx 0x02000000]  pcm_sample_bit_depth_chroma_minus1 = 0
...
[48642.979952][ctx 0x02000000]  slice_parsing_fields = 15246
 
however in FFmpeg
 
[48794.782034][ctx 0x02000000]  pcm_sample_bit_depth_luma_minus1 = 255
[48794.782034][ctx 0x02000000]  pcm_sample_bit_depth_chroma_minus1 = 255
...
[48794.782054][ctx 0x02000000]  slice_parsing_fields = 910
 
According to the tracer log, I think this is a FFmpeg specific issue. 
 
3.  driver(HW) provides error handling, but it can't cover all of errors.

fhvwy, 2017-09-19:
See notes on <https://trac.ffmpeg.org/ticket/6642>.  Maybe something is going wrong around the broken link handling for the initial part of the stream?  (The non-failure with gstreamer seems to come from throwing away everything up to the first IDR frame.)

hsyrja, 2017-09-19:
This is how I’ve seen it behaving, always when the stream starts, when the source is live (could start any point of the stream) with a good luck it may start successfully and then it works fine as long as you want.