intel/intel-vaapi-driver #274: mapping vaapi buffer slow

Frank6201141, 2017-09-21:
If the number of decoded videos increase, the time-span of gst_buffer_map() increases dramatically.

With 16 h264 full HD videos each mapping needs more than 16 ms.

We map the buffer to copy the video data. The copy itself needs only 1 ms.

We assume the increase of the mapping time is a bug, maybe a concurrency issue. Maybe inside intel vaapi driver, not the gstreamer code.

This issue was created at gstreamer-bugzilla, bug 786054

Gstreamer-bugzilla confirmed the issue and recommended to open a ticket here.

Please see https://bugzilla.gnome.org/show_bug.cgi?id=786054

Thanks.

fhvwy, 2017-09-21:
This is not a bug, the behaviour is exactly as expected.

The decode operation does not contain a synchronisation point on its output - it issues the decode to the hardware, but we only wait for its completion when the output is actually used.  In your case, this first use is at your mapping of the output surface.  A normal 1080p decode takes maybe 2-4ms in isolation, which will give your best case.  If you have 16 streams decoding at the same time then they overlap somewhat, but each map operation will still be blocked by 16 decode operations on average and therefore take much longer.

If you want to check this in your test setup, add the time spent in the decode operation (vaBegingPicture() .. vaEndPicture()) - you will see that it is approximately zero regardless of how many streams there are.

Frank6201141, 2017-10-04:
Hi Mark,

thank you for your reply.

We have done some further investigations for this issue.
If we got it right the mapping operation is the first blocking operation in your case. So it does make sense that this call took longer if we increase the amount of videos.
But in our case we did the measurements in a gstreamer pipeline. In this case there is a delay between the start of the decoding (vaBeginPicture()) in the decoder element and the point in time when we get the buffer in the appsink. We expect, that if this delay is big enough, the mapping call does not block or return soon.
In gstreamer it is not trivial to measure the time between two points in time in different pipeline elements.
In order to measure the time, we modified the gstreamer-vaapi code to print some timestamps as soon as vaBeginPicture() is called. We also adapted the reproducer example [1] to print additional timestamps just before and after gst_map_buffer() is called. To relate the timestamps we also print the PTS timestamp, which should be unique for every frame.
We have written small c++ program which helps us to parse the output. We also have written a bash script to start the benchmark with 1...n videos and to automatically parse the output and generate some overview files.
All the files mentioned above as well as the benchmark results can be found here [2].

In the archive mentioned above you can find a patch file for gstreamer-vaapi, the code for the reproducer and the parser, the bash script and our results for one benchmark. 
We also generate a chart for the results (benchmark results/results.png). 
In this chart :
•	the red line is the average of the timespan between starting decoding and the point in time after the mapping call 
•	the blue line the average of the timespan between starting decoding and the point in time before the mapping call 
•	the yellow line is the timespan if decoding one video takes 4 ms (under linearity assumption) 
•	the green line, if decoding one video takes 2 ms

For one video, the results show, that it took more than 4 ms after starting the decoding until we call gst_map_buffer(). If decoding of one video took 2-4ms as you wrote, we expect that the decoding is already done before we get the buffer in the appsink and we expect that there is no delay due to the gst_map_buffer call. 
For the chart, this means that we expect, that if the blue line is above the yellow line, then the mappings should not block or wait and therefore the red line should be near the blue line - or decoding is much slower then 2-4 ms (per video).

[1] https://bugzilla.gnome.org/attachment.cgi?id=360184&action=edit
[2] https://barcozone-my.sharepoint.com/personal/frank_huber_barco_com/_layouts/15/guestaccess.aspx?docid=19ca9fa2feab944a3b34c73b2103c3d5f&authkey=AVGDyF9y8fYGoMxPrq5MO58

[image]

fhvwy, 2017-10-04:
What is your hardware setup?  The estimated numbers I suggested above for a single decode would reflect a normal desktop or mobile GPU between Gen 7.5 and Gen 9.5 with both memory channels active.  Given that your total throughput based on the graph is only on the order of 200fps, I think something else about your system is making the results worse - I get over 600fps total throughput at 1080p (including downloading the result, but not doing anything with it) on a desktop Gen 7.5 with ffmpeg.

Frank6201141, 2017-10-04:
Our hardware setup is:

ComExpress Module
4th Gen Core
i7-4700EQ
Haswell-H
2.40 GHz
CPUs:  4/8
GPU: HD-4600
Memory Performance (Int GB/s) 15

2 Memory Banks used
Assuming it’s Gen 7.5

If I understand it right, your setup decodes only one video. 

In our setup 16 HD videos are running in parallel, with gstreamer.  
The pipeline writes/reads several times every stream, means there is a high memory load.
In another setup we run up to 50 SD videos, also a different setup.
Our setups are total different use cases.

Maybe you can run some tests with gstreamer and up to 16 HD videos.

fhvwy, 2017-10-04:
That's decoding and downloading 16 1920x1080 H.264 streams simultaneously as fast as possible.  Each stream gets about 40fps, for total throughput of around 640fps.

I don't know anything about gstreamer, so I can't comment on what to do there.

xhaihao, 2017-10-05:
@Frank6201141 AFAIK, vaGetImage() is used in gst-vaapi, which means there are some other operations besides decoding.

Frank6201141, 2017-10-11:
OK, you achieve a throughput of 640 fps in your setup with ffmpeg.
We achieve less than 480 fps throughput in our setup with gstreamer.

So we are back to the first question, why the mapping call gst_buffer_map() needs up to 16 ms?

You say, you know nothing about gstreamer, but the gstreamer people confirmed our measurement and told us to fill an issue in intel-vaapi-driver.
Please see again https://bugzilla.gnome.org/show_bug.cgi?id=786054

Anyway there should be a reason why there is this difference in throughput.

xhaihao, 2017-10-11:
@sreerenjb In order to get the output data, is vaGetImage() or veDeriveImage() used in gstreamer-vaapi?

sreerenjb, 2017-10-11:
@xhaihao gstreamer-vaapi has bit complicated logic based on different use cases. The internal GstVaapiVideoMemory uses both vaGetImage and vaDeriveImage based on circumstances, mainly depends on capabilities of downstream/upstream elements (for eg: where there is a need of implicit CSC, whether there is a software vpp element in the middle which requires write permission etc).
Usually "..vaapiXXXdec !  filesink " do a GetImage and "..vaapiXXXdec ! software_renderer_which_supports_videometa..." do a vaDeriveImage.

xhaihao, 2017-10-12:
I think vaGetImage() is used in Frank6201141's setup, otherwise copy is slow because reading data from tiled graphics memory is slow. 

@Frank6201141 vaGetImage() is not used in fhvwy's setup. vaGetImage() is an expensive operation, we did some optimization for vaGetImage() on master, you may give a try. But even with the optimization, I don't think you can get a similar result with ffmpeg setup.

sreerenjb, 2017-11-03:
We fixed the issue in middleware. Please close the bug.

closed 2017-11-04 by uartie