intel/intel-vaapi-driver #279: What is the different between this repo, and github.com/01org/iotg-lin-gfx-libva?

pauldotknopf, 2017-09-29:
Sorry if it seems like a dumb question, but I am having trouble making sense of all these repositories and figuring out which one I should use.

xhaihao, 2017-09-30:
https://github.com/01org/libva is the official git repo for libva.

pauldotknopf, 2017-09-30:
What is the difference between that repo and this intel-vaapi-driver?

Also, what is this one for? https://github.com/01org/iotg-lin-gfx-libva

xhaihao, 2017-09-30:
> What is the difference between that repo and this intel-vaapi-driver?

iotg-lin-gfx-libva is a fork of libva, not intel-vaapi-driver.  libva is an implementation for VA-API and intel-vaapi-driver is an Intel GEN GPU specific acceleration backend driver for libva. 

> Also, what is this one for? https://github.com/01org/iotg-lin-gfx-libva

You should ask the owner of https://github.com/01org/iotg-lin-gfx-libva

closed 2017-10-18 by xhaihao