intel/intel-vaapi-driver #284: VP9 CBR video corruption

asquared, 2017-10-17:
Garbled output is produced near a dissolve from a fullscreen graphic when encoding to VP9 on Kabylake at constant bitrate using FFmpeg. I attempted to reproduce this using vp9enc from vaapi-utils, but was unable to do so. Also I'm not able to reproduce using any other vaapi codec.

Here's the input (h.264) file, resulting VP9 output (garbled around 8 seconds in) and the ffmpeg log:
 [vp9bug.tar.gz](https://github.com/01org/intel-vaapi-driver/files/1392780/vp9bug.tar.gz)

Possibly related, when encoding the same file in VBR, the output is at approximately half the bitrate requested.

xhaihao, 2017-10-17:
@lizhong1008 It is most likely an issue in FFmpeg,  could you help to look into this issue?

asquared, 2017-10-18:
I had already opened https://trac.ffmpeg.org/ticket/6755 on the ffmpeg side.

lizhong1008, 2017-10-18:
Looks like a driver issue.  I tested with different bit rate(8800k, CBR)
./ffmpeg -vaapi_device /dev/dri/renderD128 -i /root/vp9bug/test.mp4 -vf "format=nv12,hwupload,deinterlace_vaapi,scale_vaapi=w=1920:h=1080" -c:v vp9_vaapi -b:v 8800k -bufsize 17600k -maxrate 8800k -g 150 -keyint_min 150 -bf 1 -bsf:v vp9_raw_reorder,vp9_superframe -y -an -f webm -dash 1 -r 30 vp9_v8800_CBR.webm

 and tested with another clips with same bitrate (4400k, CBR):
./ffmpeg -vaapi_device /dev/dri/renderD128 -i /root/Videos/skyfall2-trailer.mp4 -vf "format=nv12,hwupload,deinterlace_vaapi,scale_vaapi=w=1920:h=1080" -c:v vp9_vaapi -b:v 4400k -bufsize 17600k -maxrate 4400k -g 150 -keyint_min 150 -bf 1 -bsf:v vp9_raw_reorder,vp9_superframe -an -f webm -dash 1 -r 29.97 sky__v4400_CBR.webm

There are no garbled output.

asquared, 2017-10-18:
I can reproduce the garbled output at 8800k CBR if I remove "-r 30". I can produce correct output at 4400k CBR if I change "-r 29.97" to "-r 30". I wonder if this is related to #87.

It's not clear to me why the framerate should have any effect at all on the VP9 bitstream produced... shouldn't this be just a piece of metadata added by the muxer in ffmpeg?

asquared, 2017-10-18:
Also, even with "-r 30", the bitrate is lower than expected in VBR mode.

asquared, 2017-10-18:
I can also produce garbled output at 30fps under some conditions. Will post a reproduction soon.

asquared, 2017-10-18:
I think the frame rate was a red herring. I get similar output with or without -r 30. Reproduction is more dependent on CBR bit rate. Input, outputs, and ffmpeg logs: [test_20171018.tar.gz](https://github.com/01org/intel-vaapi-driver/files/1395035/test_20171018.tar.gz)

I tried both of these with the -r 30 option as well. No difference in the output; playback freezes at 8 seconds in Chrome and VLC shows a few seconds of garbage 8 seconds in.

xhaihao, 2017-10-18:
please double-check the code below in vaapi_encode.c in FFmpeg 

>  1196      if (avctx->framerate.num > 0 && avctx->framerate.den > 0)        
>  1197          av_reduce(&fr_num, &fr_den,                                  
>  1198                    avctx->framerate.num, avctx->framerate.den, 65535);
>  1199      else                                                             
>  1200          av_reduce(&fr_num, &fr_den,                                  
>  1201                    avctx->time_base.den, avctx->time_base.num, 65535);
> 

either line 1197-1198 or line 1200-1201 is wrong

xhaihao, 2017-10-18:
> int av_reduce(int *dst_num, int *dst_den,           
>               int64_t num, int64_t den, int64_t max)

so line 1200-1201 is wrong. it should be 

> av_reduce(&fr_num, &fr_den, avctx->time_base.num, avctx->time_base.den, 65535);

asquared, 2017-10-18:
I think that the flipped den/num is deliberate there. time_base has units of seconds, so this is taking 1/time_base to get the frame rate in fps in the case where framerate is not set.

xhaihao, 2017-10-18:
@asquared  yes, you are right. I should check the code more carefully.

lizhong1008, 2017-10-19:
> It's not clear to me why the framerate should have any effect at all on the VP9 bitstream produced... shouldn't this be just a piece of metadata added by the muxer in ffmpeg?
Bitrate (bits/second) is passed to driver, and driver need to calculate how many bits should be assigned to each frame, then the framerate should be known. Refer:
https://github.com/01org/intel-vaapi-driver/blob/master/src/gen9_vp9_encoder.c#L1070

lizhong1008, 2017-10-19:
> It's not clear to me why the framerate should have any effect at all on the VP9 bitstream produced... shouldn't this be just a piece of metadata added by the muxer in ffmpeg?

It's not clear to me why the framerate should have any effect at all on the VP9 bitstream produced... shouldn't this be just a piece of metadata added by the muxer in ffmpeg?
Bitrate (bits/second) is passed to driver, and driver need to calculate how many bits should be assigned to each frame, then the framerate should be known. Refer:
https://github.com/01org/intel-vaapi-driver/blob/master/src/gen9_vp9_encoder.c#L1070

asquared, 2017-10-19:
of course, sorry for the stupid question :)

lizhong1008, 2017-11-01:
After analysis,  it should be a driver issue, which only happens with some special bitrate and clips (At least the CBR case it is). So assign it to @xhaihao.

lizhong1008, 2017-11-09:
The garbed picture has an error TX_MODE though driver set it to be TX_MODE_SELECT. This issue only happens on key frame of some special clips with specific bitrate.  So it is a driver issue,  @xhaihao could you help to check it?

lizhong1008, 2017-11-09:
@asquared before this issue is fixed, maybe you can try a larger key frame period (e.g set -g 200 or 250) as workaround.

asquared, 2017-11-10:
-g 250 allows me to produce CBR files with correct bitrate (with -g 150 I got about half the expected bitrate). But in some cases I still see corrupted output with -g 250.

Is there a way to detect this corruption in the bitstream? Even if there's no fix yet, it'd be helpful to identify this problem without having to watch the video.

lizhong1008, 2017-11-13:
> Is there a way to detect this corruption in the bitstream? Even if there's no fix yet, it'd be helpful to identify this problem without having to watch the video.
One suggestion may be help: 1. calculate the average psnr of this video.  2. compare the psnr value between the the average psnr value and psnr of each frame.  3. if the delta_psnr is larger than some  threshold value  (8 db, for example), then return a fail message and report this frame number.

lizhong1008, 2017-11-13:
> Is there a way to detect this corruption in the bitstream? Even if there's no fix yet, it'd be helpful to identify this problem without having to watch the video.

Is there a way to detect this corruption in the bitstream? Even if there's no fix yet, it'd be helpful to identify this problem without having to watch the video.
One suggestion may be help: 1. calculate the average psnr of this video. 2. compare the psnr value between the the average psnr value and psnr of each frame. 3. if the delta_psnr is larger than some threshold value (8 db, for example), then return a fail message and report this frame number.

lizhong1008, 2017-11-16:
> +    vpic->luma_dc_qindex_delta   = 1;
> +    vpic->chroma_ac_qindex_delta = 1;
> +    vpic->chroma_dc_qindex_delta = 1;

That's for reporting this. It is so helpful to find out the root cause.  I've sent a patch to fix it.

xhaihao, 2017-12-01:
The corresponding PR has been merged

closed 2017-12-01 by xhaihao