intel/intel-vaapi-driver #288: multiple definition of 'gen9_hevc_encoder_kernels'

Ristovski, 2017-10-22:
When compiling the latest master (also happens on 2.0.0 release), the compile fails with the following: 
```
/bin/sh ../libtool  --tag=CC   --mode=link gcc -Wall -fvisibility=hidden    -module -avoid-version -no-undefined -Wl,--no-undefined   -o i965_drv_video.la -rpath /usr/lib64/va/drivers  libi965_drv_video.la -lpthread -lm -ldl -ldrm  -ldrm_intel  -lm 
libtool: link: gcc -shared  -fPIC -DPIC  -Wl,--whole-archive ./.libs/libi965_drv_video.a -Wl,--no-whole-archive  -lpthread -ldl -ldrm -ldrm_intel -lm  -Wl,--no-undefined   -Wl,-soname -Wl,i965_drv_video.so -o .libs/i965_drv_video.so
/usr/lib/gcc/x86_64-pc-linux-gnu/8.0.0-alpha20171015/../../../../x86_64-pc-linux-gnu/bin/ld: error: ./.libs/libi965_drv_video.a(libi965_drv_video_la-gen9_hevc_encoder.o): multiple definition of 'gen9_hevc_encoder_kernels'
/usr/lib/gcc/x86_64-pc-linux-gnu/8.0.0-alpha20171015/../../../../x86_64-pc-linux-gnu/bin/ld: ./.libs/libi965_drv_video.a(libi965_drv_video_la-gen9_hevc_enc_kernels_binary.o): previous definition here
collect2: error: ld returned 1 exit status
```

Using libva from git.
GCC 8.0.0

stripes416, 2017-10-23:
Is GCC 8.0 distributed by any distro at this point?

xhaihao, 2017-11-28:
@Ristovski It seems the PR you submitted for this issue was closed because the corresponding branch doesn't exist any more,  could you help to resubmit your PR?