intel/intel-vaapi-driver #290: segfault when rendering black frame (YV12) buffer.

pauldotknopf, 2017-11-06:
My application crashes when rendering a YV12 buffer where all Y's have a ```0x00``` value.

```
segfault at 5649cf278ac8 ip 00007f657cfbdf7a sp 00007ffedc2a9e60 error 4 in i965_dri.so[7f657cb99000+7ef000]
```

This is what I do to reproduce the issue.

```cpp
int index = 0;
for(int width=1;width<currentStatus->width;width++) {
    for(int height=1;height<currentStatus->height;height++) {
        data[index] = 0x00;
        index++;
    }
}
```

This version works.

```cpp
int index = 0;
for(int width=1;width<currentStatus->width;width++) {
    for(int height=1;height<currentStatus->height;height++) {
        data[index] = 0x01;
        index++;
    }
}
```

Notice the value I set to ```data[index]```.

Here is my driver info from ```vainfo```.

```
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.40 (libva )
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 1.8.3
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
```

xhaihao, 2017-11-27:
sorry for slow response, could you provide a simple test case?

pauldotknopf, 2017-11-27:
I could provide a boot-able image which reproduces this (yocto). Would this work for you?

pauldotknopf, 2017-11-27:
I haven't generated backtrace for kernel modues before, let me look into it.

xhaihao, 2017-11-27:
Do you mean the crash occurs in kernel module? I thought you are talking about user space driver issue.

pauldotknopf, 2017-11-27:
Actually, you might be right. It is the i965_dri.so library.

I am working right now getting you a backtrace.

This is very important to our company, and i appreciate you looking into this.

Stay tuned for the backtrace.

xhaihao, 2017-11-27:
i965_dri.so is the 3D driver for Intel Gen Graphics, I think you should talk to Mesa guys.

xhaihao, 2017-11-27:
I close this issue because this is not a intel-vaapi-driver bug.  please file your 3D driver related issue on https://bugs.freedesktop.org.

closed 2017-11-27 by xhaihao

pauldotknopf, 2017-11-28:
I will start an issue on the other forum. For the same of being thorough, I found the line causing the issue.

https://cgit.freedesktop.org/mesa/mesa/tree/src/mesa/drivers/dri/i965/intel_tiled_memcpy.c?id=mesa-17.0.2#n509

Here was the output.

```
Thread 1 "mxencoder-test-" received signal SIGSEGV, Segmentation fault.
0x00007fffe970a980 in linear_to_ytiled_faster (x0=<optimized out>, x1=<optimized out>, x2=<optimized out>, x3=<optimized out>, y0=y0@entry=0,
    y1=y1@entry=32, dst=0x7fff82c59000 "", src=0x25a5448 "", src_pitch=3840, swizzle_bit=0, mem_copy=0x33e6727670 <__memmove_avx_unaligned_erms>)
    at /usr/src/debug/mesa/2_17.0.2-r0/mesa-17.0.2/src/mesa/drivers/dri/i965/intel_tiled_memcpy.c:509
```

pauldotknopf, 2017-11-28:
Also, here is the bug I reported at freedesktop.

https://bugs.freedesktop.org/show_bug.cgi?id=103948