intel/intel-vaapi-driver #293: Fails with "DRM_IOCTL_I915_GEM_APERTURE failed: No such file or directory" when i915=/dev/dri/card1 & radeon=/dev/dri/card0

Hagbard-Celin, 2017-11-08:
I get this error when running test_va_api or vlc playback with LIBVA_DRIVER_NAME=i965 when i915.ko is /dev/dri/card1 and radeon.ko is /dev/dri/card0:
"DRM_IOCTL_I915_GEM_APERTURE failed: No such file or directory"
Before I set LIBVA_DRIVER_NAME vlc complained:
"vaapi_drm error: Failed to initialize the VAAPI device"
The errors go away if I jump trough a lot of hoops to force the kernel to load i915.ko before radeon.ko but that leads to random X crashes with no error messages whatsoever (even tried enabling extra debugging of i915 in kernel config). It also breaks my early-prink setup.
I run 1.19.5 on i7-4790 integrated-GPU with SNA detected as gen7.5, gt2 with latest git version of xf86-video-intel, and mesa 17.2.4.
The radeon card is a R5 230 OEM PCIe 1x card used for fbcon.
Also X  is always started with  -sharevts -novtswitch (The setup equals mult-seat with all keyboard and mouse inputs routed trough a switch to one set.)

xhaihao, 2017-11-27:
Sorry for slow response, does 'vainfo --display drm --device /dev/dri/renderD129' work for you?

Hagbard-Celin, 2017-11-28:
No, this is the result I'm getting:
```
% vainfo --display drm --device /dev/dri/renderD129
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib64/va/drivers/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
DRM_IOCTL_I915_GEM_APERTURE failed: No such file or directory
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [13]
param: 4, val: 0
vainfo: /pt/portage/x11-libs/libva-intel-driver-1.8.3/work/intel-vaapi-driver-1.8.3/src/intel_memman.c:38: intel_memman_init: Assertion `intel->bufmgr' failed.
zsh: abort      vainfo --display drm --device /dev/dri/renderD129
```
It works if I replace every occurrence of "card0" and "renderD128" with "card1" and "renderD129" respectively in libva, libva-utils and libva-intel-driver sources, and recompile. In VLC I still have to change to va-via-X11 for it to work, va-via-drm gives the same error even with the adjusted sources. 
It seems to me that when selecting the Intel driver with LIBVA_DRIVER_NAME env-variable, libva should select the right card, or we need a LIBVA_DEVICE_NAME (or/and LIBVA_RENDER_DEVICE_NAME) variable too.

xhaihao, 2017-11-28:
> libva info: User requested driver 'i965'
You set the driver  via LIBVA_DRIVER_NAME,  could you give a try without LIBVA_DRIVER_NAME when running 'vainfo --display drm --device /dev/dri/renderD129'

Hagbard-Celin, 2017-11-28:
I have already set driver with LIBVA_DRIVER_NAME=i965 as stated in my first post. What I am saying is the setting driver should also select the right "/dev/dri/cardx" and "/dev/dri/renderDxxx", as it does not do at the moment. Or, we need additional variable(s) to set those. As it is now, it seems card0 and renderD128 gets hard-coded (at-least in the software I use) when using va via drm, that works badly for people with multiple GPU's.

Hagbard-Celin, 2017-11-28:
Sorry I misread your comment, here is result:
```
% unset LIBVA_DRIVER_NAME 
% vainfo --display drm --device /dev/dri/renderD129    
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/va/drivers/r600_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

```

xhaihao, 2017-11-28:
1. I thought your i915 is using card1 and render129 as the title says 'i915=/dev/dri/card1'. Actually your i915 is using card0 and render128 because you changed the loading order for i915.ko and radeon.ko.  
so you should run 'vainfo --display drm --device /dev/dri/renderD128'. You needn't set LIBVA_DRIVER_NAME because the device has been set via option '--device'

2. Even for Intel GPU, there are two different DRI devices, so it is hard for driver to select the device. 

3. VA-API provides the interface to accept the file descriptor from app, so it would be better the software can provide an option to select different devices, like as what we did in vainfo. 
 
4. dri2 prime support is needed for the current implementation If you want to select a different driver upon X11.

Hagbard-Celin, 2017-11-28:
My Intel card is /dev/dri/card1 and /dev/dri/renderD129 as shown below.
I tried switching load order but that lead to all kind of problems as explained in my first post, so I switched back.
```
% ls -l /dev/dri/by-path 
total 0
lrwxrwxrwx 1 root root  8 Nov 27 22:25 pci-0000:00:02.0-card -> ../card1
lrwxrwxrwx 1 root root 13 Nov 27 22:25 pci-0000:00:02.0-render -> ../renderD129
lrwxrwxrwx 1 root root  8 Nov 27 22:25 pci-0000:12:00.0-card -> ../card0
lrwxrwxrwx 1 root root 13 Nov 27 22:25 pci-0000:12:00.0-render -> ../renderD128
```
```
% sudo lspci
Password: 
00:00.0 Host bridge: Intel Corporation 4th Gen Core Processor DRAM Controller (rev 06)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller (rev 06)
00:02.0 Display controller: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller (rev 06)
00:03.0 Audio device: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller (rev 06)
00:14.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI (rev 05)
00:16.0 Communication controller: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 (rev 04)
00:1a.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 (rev 05)
00:1b.0 Audio device: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller (rev 05)
00:1c.0 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 (rev d5)
00:1c.1 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 (rev d5)
00:1c.2 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 (rev d5)
00:1c.3 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 (rev d5)
00:1c.4 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 (rev d5)
00:1d.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 (rev 05)
00:1f.0 ISA bridge: Intel Corporation C226 Series Chipset Family Server Advanced SKU LPC Controller (rev 05)
00:1f.2 SATA controller: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] (rev 05)
00:1f.3 SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 05)
01:00.0 VGA compatible controller: NVIDIA Corporation GK110B [GeForce GTX 780 Ti] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GK110 HDMI Audio (rev a1)
02:00.0 SATA controller: Marvell Technology Group Ltd. 88SE9172 SATA 6Gb/s Controller (rev 11)
03:00.0 SATA controller: Marvell Technology Group Ltd. 88SE9172 SATA 6Gb/s Controller (rev 11)
04:00.0 Ethernet controller: Intel Corporation I210 Gigabit Network Connection (rev 03)
05:00.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
06:01.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
06:04.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
06:05.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
06:06.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
06:07.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
06:08.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
06:09.0 PCI bridge: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch (rev ba)
07:00.0 FireWire (IEEE 1394): VIA Technologies, Inc. VT6315 Series Firewire Controller (rev 01)
08:00.0 Ethernet controller: Intel Corporation I210 Gigabit Network Connection (rev 03)
09:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 01)
0a:00.0 USB controller: ASMedia Technology Inc. ASM1042 SuperSpeed USB Host Controller
0c:00.0 PCI bridge: ASMedia Technology Inc. ASM1083/1085 PCIe to PCI Bridge (rev 03)
0e:00.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
0f:01.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
0f:02.0 PCI bridge: Pericom Semiconductor Device 2304 (rev 05)
10:00.0 USB controller: Fresco Logic FL1100 USB 3.0 Host Controller (rev 01)
11:00.0 FireWire (IEEE 1394): LSI Corporation FW643 [TrueFire] PCIe 1394b Controller (rev 08)
12:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Caicos [Radeon HD 6450/7450/8450 / R5 230 OEM]
12:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Caicos HDMI Audio [Radeon HD 6450 / 7450/8450/8490 OEM / R5 230/235/235X OEM]
```
If the driver can not select the devise automaticity there should be a env-variable, like the LIBVA_DRIVER_NAME does for selecting driver. Implementing a card selection in applications would make sense if the applications where already where the driver is selected. But it is not the driver is selected by libva via LIBVA_DRIVER_NAME, not by the application.
Also, this is a problem in every application using libva via drm, fixing every application seems like lots of duplicate work.

xhaihao, 2017-11-28:
> My Intel card is /dev/dri/card1 and /dev/dri/renderD129 as shown below

The output of vainfo shows the driver for your renderD129 is r600_drv_video.so, so I think your Intel card should be card0 and renderD128. It would be better to check the files in /sys/devices/pci0000:00/0000:00:02.0/drm 

> 00:02.0 Display controller: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller (rev 06)

It is not a VGA compatible controller. 

> Also, this is a problem in every application using libva via drm, fixing every application seems like lots of duplicate work.

vaGetDisplayDRM(int fd) requires that application provides a right fd.

Hagbard-Celin, 2017-11-28:
This is not a laptop. This is a Asrock C266 WS. My Xorg.log as proof:
```

% cat /var/log/Xorg.0.log
[ 95123.045] 
X.Org X Server 1.19.5
Release Date: 2017-10-12
[ 95123.045] X Protocol Version 11, Revision 0
[ 95123.045] Build Operating System: Linux 4.13.11-gentoovaapi x86_64 Gentoo
[ 95123.045] Current Operating System: Linux 40o2 4.14.2-gentoo #2 SMP PREEMPT Mon Nov 27 22:20:12 CET 2017 x86_64
[ 95123.045] Kernel command line: acpi_osi=! acpi_osi="Windows 2013" intremap=no_x2apic_optout \vmlinuz-4.14.2-gentoo.efi iommu=pt root=PARTUUID=56ed4fcd-3fa0-4c55-8567-a8287a2241c6 rw rootfstype=f2fs rootflags=disable_ext_identify,discard,inline_xattr,inline_dentry video=VGA-1:1440x900@75 add_efi_memmap pci-stub.ids=1b21:0612 vfio-pci.ids=10de:100a,10de:0e1a,12d8:2304,1b21:0612,8086:8c20,1b21:1042 scsi_mod.use_blk_mq=1 default_hugepagesz=1G hugepagesz=1G hugepages=16 hugepagesz=2M hugepages=5 vt.default_red=0,170,0,170,118,170,0,170,85,255,85,255,85,255,85,255 vt.default_blu=0,0,0,0,134,170,170,170,85,85,85,85,255,255,255,255 threadirqs nmi_watchdog=0 initrd=microcode.cpio initrd=initramfs.cpio
[ 95123.045] Build Date: 06 November 2017  11:54:00AM
[ 95123.045]  
[ 95123.045] Current version of pixman: 0.34.0
[ 95123.045] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[ 95123.045] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 95123.045] (==) Log file: "/var/log/Xorg.0.log", Time: Wed Nov 29 00:50:25 2017
[ 95123.045] (==) Using config file: "/etc/X11/xorg.conf"
[ 95123.045] (==) Using config directory: "/etc/X11/xorg.conf.d"
[ 95123.045] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[ 95123.046] (==) ServerLayout "screen"
[ 95123.046] (==) No screen section available. Using defaults.
[ 95123.046] (**) |-->Screen "Default Screen Section" (0)
[ 95123.046] (**) |   |-->Monitor "<default monitor>"
[ 95123.047] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[ 95123.047] (**) |   |-->Device "i965"
[ 95123.047] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[ 95123.047] (**) Option "BlankTime" "0"
[ 95123.047] (**) Option "StandbyTime" "0"
[ 95123.047] (**) Option "SuspendTime" "0"
[ 95123.047] (**) Option "OffTime" "0"
[ 95123.047] (==) Automatically adding devices
[ 95123.047] (==) Automatically enabling devices
[ 95123.047] (==) Automatically adding GPU devices
[ 95123.047] (==) Max clients allowed: 256, resource mask: 0x1fffff
[ 95123.047] (**) FontPath set to:
	/usr/share/fonts/local,
	/usr/share/fonts/misc/,
	/usr/share/fonts/TTF/,
	/usr/share/fonts/OTF/,
	/usr/share/fonts/Type1/,
	/usr/share/fonts/100dpi/,
	/usr/share/fonts/75dpi/
[ 95123.047] (==) ModulePath set to "/usr/lib64/xorg/modules"
[ 95123.047] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[ 95123.047] (II) Loader magic: 0x5c59cc62fc40
[ 95123.047] (II) Module ABI versions:
[ 95123.047] 	X.Org ANSI C Emulation: 0.4
[ 95123.047] 	X.Org Video Driver: 23.0
[ 95123.047] 	X.Org XInput driver : 24.1
[ 95123.047] 	X.Org Server Extension : 10.0
[ 95123.050] (II) xfree86: Adding drm device (/dev/dri/card1)
[ 95123.063] (II) xfree86: Adding drm device (/dev/dri/card0)
[ 95123.085] (--) PCI: (0:0:2:0) 8086:0412:1849:0412 rev 6, Mem @ 0xf6400000/4194304, 0xc0000000/268435456, I/O @ 0x0000f000/64
[ 95123.088] (--) PCI:*(0:1:0:0) 10de:100a:1458:3628 rev 161, Mem @ 0xf5000000/16777216, 0xe0000000/134217728, 0xe8000000/33554432, I/O @ 0x0000e000/128, BIOS @ 0x????????/524288
[ 95123.088] (--) PCI: (0:18:0:0) 1002:6779:1787:5230 rev 0, Mem @ 0xd0000000/268435456, 0xf6f20000/131072, I/O @ 0x0000a000/256, BIOS @ 0x????????/131072
[ 95123.088] (II) LoadModule: "glx"
[ 95123.089] (II) Loading /usr/lib64/xorg/modules/extensions/libglx.so
[ 95123.098] (II) Module glx: vendor="X.Org Foundation"
[ 95123.098] 	compiled for 1.19.5, module version = 1.0.0
[ 95123.098] 	ABI class: X.Org Server Extension, version 10.0
[ 95123.098] (II) LoadModule: "intel"
[ 95123.098] (II) Loading /usr/lib64/xorg/modules/drivers/intel_drv.so
[ 95123.098] (II) Module intel: vendor="X.Org Foundation"
[... 317 lines cut ...]
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.40 (libva )
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 1.8.3
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD

```

Hagbard-Celin, 2017-11-29:
> vaGetDisplayDRM(int fd) requires that application provides a right fd.
And herein lies the problem. You let the driver be set directly in libva with LIBVA_DRIVER_NAME independent of witch application beeing used. But leave the issue of selecting the right fd to the application, leading to the current result where (almost) all are hard-coded with /dev/dri/renderD128, and people belive a pc with Nvidia+Intel or Radeon+Intel is a Optimus/Whatever-AMD-calls-it type Laptop. Please give us a option to override the device by env variable, like we can with driver, that is all that I am asking.

xhaihao, 2017-11-29:
> And this is what I get after running "grep -rl card0 | xargs sed -i 's/card0/card1/g'" and "grep -rl renderD128 | xargs sed -i 's/renderD128/renderD129/g'" on sources of libva, libva-utils and libva-intel-driver and recompile them:

You needn't change the source code, the option of --device should work for you, please check  
https://github.com/01org/intel-vaapi-driver/issues/186

xhaihao, 2017-11-29:
> But leave the issue of selecting the right fd to the application, leading to the current result where (almost) all are hard-coded with /dev/dri/renderD128, 

AFAIK Both FFmpeg/Libav and gstreamer-vaapi don't use hard-coded /dev/dri/renderD128.

xhaihao, 2017-11-29:
Could you run the commands below to check which kernel module creates renderD128 and renderD129

$> sudo cat /sys/kernel/debug/dri/128/name
$> sudo cat /sys/kernel/debug/dri/129/name

Hagbard-Celin, 2017-11-29:
Back on unpatched libva:

```
% unset LIBVA_DRIVER_NAME
% vainfo --display drm --device /dev/dri/renderD129
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/va/drivers/r600_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
% vainfo --display drm --device /dev/dri/renderD128
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/va/drivers/r600_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
```
% sudo cat /sys/kernel/debug/dri/128/name
Password: 
radeon dev=0000:12:00.0 unique=0000:12:00.0
% sudo cat /sys/kernel/debug/dri/129/name
i915 dev=0000:00:02.0 master=pci:0000:00:02.0 unique=0000:00:02.0

```

Hagbard-Celin, 2017-11-29:
> AFAIK Both FFmpeg/Libav and gstreamer-vaapi don't use hard-coded /dev/dri/renderD128.

Ok, I probably stand corrected. But I've yet to find a GUI-app that let me set this.

xhaihao, 2017-11-29:
> vainfo: VA-API version: 0.40 (libva )

The vainfo you are using isn't the latest one for libva1 because the version of libva isn't printed out in the log.  Please update your libva-utils from https://github.com/01org/libva-utils/releases/tag/1.8.3

Hagbard-Celin, 2017-11-29:
> The vainfo you are using isn't the latest one for libva1 because the version of libva isn't printed out in the log. Please update your libva-utils from https://github.com/01org/libva-utils/releases/tag/1.8.3

This is the version I use.
I've double checked, downloaded again and diffed against the source-archive I compiled from and they are identical.

Hagbard-Celin, 2017-11-29:
On another note, the file "/usr/lib64/va/drivers/r600_drv_video.so" actually does not exist, as I use the radon only for fbcon.

xhaihao, 2017-11-29:
r600 is a Mesa Gallium driver for radeon, please check the commit below for details: 

commit 2eef7235d429b1051bcaf75281868f40ee563c9d
Author: Mark Thompson <sw@jkqxz.net>           
Date:   Sat Feb 4 16:00:17 2017 +0000          
                                               
    drm: Fix Mesa driver names

Hagbard-Celin, 2017-11-29:
I'm just pointing out that the file does not exist, as I have not installed Gallium. Only Intel card is used for X11. Radeon is only used for fbcon and Nvidia only for KVM.

xhaihao, 2017-11-29:
> I've double checked, downloaded again and diffed against the source-archive I compiled from and they are identical.

I'm sorry for my bad memory. Actually --device for vainfo is not supported in 1.8.x, it only works in 2.0.x

xhaihao, 2017-11-29:
Please try https://github.com/01org/libva-utils/commit/f36a392a969a0b3045f2832495fc1ccfd4bf90e5

Hagbard-Celin, 2017-11-29:
I'll have to get back to it, getting too tired, middle of night here. Also capitulated and reported here: [https://trac.videolan.org/vlc/ticket/19173](https://trac.videolan.org/vlc/ticket/19173)

Hagbard-Celin, 2017-11-29:
> Please try 01org/libva-utils@f36a392

That works:
```

% unset LIBVA_DRIVER_NAME
% vainfo --display drm --device /dev/dri/renderD129
libva info: VA-API version 0.40.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib64/va/drivers/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_40
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.40 (libva )
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 1.8.3
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD

```

Hagbard-Celin, 2017-11-29:
Seems, like VLC got this on their todo-list. I stand completely corrected, you may close this.

xhaihao, 2017-11-30:
Closed as VLC will fix the fixed DRI device name

closed 2017-11-30 by xhaihao