intel/intel-vaapi-driver #297: VP9 decoding looks bad with DASH videos

op8867555, 2017-11-14:
Hi there

I' m getting gray boxes like #262 when playing back DASH VP9 videos.
It can be reproduced on chromium by changing video quality while playing back a VP9 youtube video. It doesn't happens with a fixed resolution video, so I open this as a separated issue. There are a few people can confirm this[^1].

It can be also reproduced on mpv by playing a WebM video contains video segments have different resolutions.
I tried construct such video by concatenating these VP9 samples[^2] with ffmepg. The gray boxes appears right after resolution changed on both chromium-vaapi and mpv.

Arch Linux (4.13.11) Intel Core i7-7500U / HD620
libva 2.0.0
intel-vaapi-driver 2.0.0

[^1]: https://aur.archlinux.org/packages/chromium-vaapi/
[^2]: http://dash-mse-test.appspot.com/media.html

xhaihao, 2017-11-14:
Please give a try with https://github.com/01org/intel-vaapi-driver/pull/296

op8867555, 2017-11-14:
I just tried v2.0.0 patched with #296. So far it hasn't happened again, I think #296 solved this issue.

maximbaz, 2017-11-14:
One other person and I both confirm that #296 fixes this issue - thanks!

closed 2017-11-28 by xhaihao