intel/intel-vaapi-driver #299: 4:2:2 or 4:0:0 support for encoding

sesse, 2017-11-16:
Hi,

I have multiple capture cards giving 4:2:2 video at a high rate that I would like to send on in realtime and in high quality. VA-API would be perfect for this (probably using intraframe-only H.264 at a high bitrate), except that the Intel driver seems to be locked to 4:2:0.

I considered making my own private format with three monochrome channels, ie., encode Y, Cb and Cr as three separate streams in monochrome. But 4:0:0 isn't supported either! It sounds like something that would be fairly easy to implement by just not encoding chroma; would it be possible to support something like this? Ideally, just native 4:2:2 would be the best, of course.

xhaihao, 2017-11-16:
You may give a try with jpeg encoding if you are interested in intraframe-only. JPEG encoder supports YUY2 and VYUY formats, please refer to https://github.com/01org/libva-utils/blob/master/encode/jpegenc.c

sesse, 2017-11-16:
H.264 is significantly better quality than JPEG, but sure, it's interesting. However, I'm on Haswell, and JPEG encoding didn't arrive before Skylake (well, Braswell, but for regular desktop chips…).

xhaihao, 2017-11-17:
We don't have plan to support h264 encoding with 4:2:2 or 4:0:0 format

closed 2017-11-17 by xhaihao