intel/intel-vaapi-driver #303: can't compile on clear linux

tuxutku, 2017-11-27:
i am trying to get those files:
        libva.so.1 
	libva-x11.so.1 
	libva-drm.so.1 

And those files exist:

utku@clr-449e9b2a44f8458bb4885604dc172a1c ~ $ find /usr -name libva*
/usr/lib64/libva-x11.so
/usr/lib64/libva.so
/usr/lib64/libva-x11.so.2.0.0
/usr/lib64/libva-drm.so.2.0.0
/usr/lib64/libva.so.2
/usr/lib64/totem/plugins/variable-rate/libvariable-rate.so
/usr/lib64/libva-drm.so
/usr/lib64/libva-wayland.so
/usr/lib64/libva-glx.so.2.0.0
/usr/lib64/libva-drm.so.2
/usr/lib64/libva-wayland.so.2.0.0
/usr/lib64/libva-glx.so.2
/usr/lib64/libva-wayland.so.2
/usr/lib64/libva.so.2.0.0
/usr/lib64/pkgconfig/libva-wayland.pc
/usr/lib64/pkgconfig/libva-glx.pc
/usr/lib64/pkgconfig/libva.pc
/usr/lib64/pkgconfig/libva-drm.pc
/usr/lib64/pkgconfig/libva-x11.pc
/usr/lib64/libva-glx.so
/usr/lib64/libva-x11.so.2

So i assume that if i compile libav and install to /usr/local, i can get these *.so.1 files
Terminal output when i tried to compile:
<pre>
utku@clr-449e9b2a44f8458bb4885604dc172a1c ~/Programlar/libva-master $ ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I m4 ${ACLOCAL_FLAGS}
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf
autoreconf: running: /usr/bin/autoheader
autoreconf: running: automake --add-missing --copy --no-force
configure.ac:166: installing './compile'
configure.ac:166: installing './config.guess'
configure.ac:166: installing './config.sub'
configure.ac:93: installing './install-sh'
configure.ac:93: installing './missing'
va/Makefile.am: installing './depcomp'
va/wayland/Makefile.am:68: warning: '%'-style pattern rules are a GNU make extension
va/wayland/Makefile.am:70: warning: '%'-style pattern rules are a GNU make extension
va/wayland/Makefile.am:72: warning: '%'-style pattern rules are a GNU make extension
va/wayland/Makefile.am:30: warning: source file '$(top_srcdir)/va/drm/va_drm_utils.c' is in a subdirectory,
va/wayland/Makefile.am:30: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
[... 274 lines cut ...]
os-testsuite
os-testsuite-phoronix
os-utils-gui
os-utils-gui-dev
perl-basic
perl-basic-dev
php-basic
python-basic
python-basic-dev
python-extras
python3-basic
ruby-basic
rust-basic
shells
storage-utils
storage-utils-dev
sysadmin-basic
sysadmin-basic-dev
telemetrics
user-basic
user-basic-dev
web-server-basic
xfce4-desktop
Current OS version: 19270
</pre>

tuxutku, 2017-11-27:
also compiling from 1.8 branch gave me this error:
<pre>
utku@clr-449e9b2a44f8458bb4885604dc172a1c ~/Programlar/intel-vaapi-driver-1.8-branch $ ./autogen.sh
ln: failed to create symbolic link '.git/hooks/pre-commit': No such file or directory
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal 
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: running: /usr/bin/autoconf
autoreconf: running: /usr/bin/autoheader
autoreconf: running: automake --add-missing --copy --no-force
configure.ac:73: installing './compile'
configure.ac:73: installing './config.guess'
configure.ac:73: installing './config.sub'
configure.ac:26: installing './install-sh'
configure.ac:26: installing './missing'
src/Makefile.am:116: warning: '%'-style pattern rules are a GNU make extension
src/Makefile.am: installing './depcomp'
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '1000' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... gcc-nm
checking the name lister (gcc-nm) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... dlltool
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... gcc-ranlib
[... 551 lines cut ...]
     ^~
In file included from /usr/include/va/va_backend.h:32:0,
                 from intel_driver.h:13,
                 from intel_batchbuffer.h:9,
                 from i965_decoder_utils.c:27:
/usr/include/va/va.h:345:5: note: declared here
     VAProfileH264Baseline va_deprecated_enum = 5,
     ^~~~~~~~~~~~~~~~~~~~~
i965_decoder_utils.c:966:9: warning: ‘num_slice_groups_minus1’ is deprecated [-Wdeprecated-declarations]
         if (pic_param->num_slice_groups_minus1 ||
         ^~
In file included from /usr/include/va/va_backend.h:32:0,
                 from intel_driver.h:13,
                 from intel_batchbuffer.h:9,
                 from i965_decoder_utils.c:27:
/usr/include/va/va.h:2168:27: note: declared here
     va_deprecated uint8_t num_slice_groups_minus1;
                           ^~~~~~~~~~~~~~~~~~~~~~~
make[3]: Leaving directory '/home/utku/Programlar/intel-vaapi-driver-1.8-branch/src'
make[2]: *** [Makefile:1341: all-recursive] Error 1
make[2]: Leaving directory '/home/utku/Programlar/intel-vaapi-driver-1.8-branch/src'
make[1]: *** [Makefile:654: all] Error 2
make[1]: Leaving directory '/home/utku/Programlar/intel-vaapi-driver-1.8-branch/src'
make: *** [Makefile:420: all-recursive] Error 1
</pre>

xhaihao, 2017-11-27:
1. The first part is libva related which is the duplicate of https://github.com/01org/libva/issues/152.  Could you file any libva issue on https://github.com/01org/libva/issues? 

2. please build libva 1.8.x first if you want to use intel-vaapi-driver 1.8.x. intel-vaapi-driver 1.8.x doesn't work with libva 2.0.x, sorry for the inconvenience.

xhaihao, 2017-11-27:
Please try the latest libva master

tuxutku, 2017-11-28:
I will try with optimization option -O1

tuxutku, 2017-11-28:
<pre>
utku@clr-449e9b2a44f8458bb4885604dc172a1c ~/Programlar/intel-vaapi-driver-1.8-branch $ ./autogen.sh
ln: failed to create symbolic link '.git/hooks/pre-commit': No such file or directory
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal 
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: running: /usr/bin/autoconf
autoreconf: running: /usr/bin/autoheader
autoreconf: running: automake --add-missing --copy --no-force
src/Makefile.am:116: warning: '%'-style pattern rules are a GNU make extension
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '1000' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... gcc-nm
checking the name lister (gcc-nm) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... dlltool
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... gcc-ranlib
checking command to parse gcc-nm output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
[... 388 lines cut ...]
                 ASSERT_RET(encode->last_packed_header_type == VAEncPackedHeaderSequence ||
                 ^~~~~~~~~~
In file included from /usr/include/va/va_backend.h:32:0,
                 from i965_output_dri.h:29,
                 from i965_drv_video.c:35:
/usr/include/va/va.h:1304:5: note: declared here
     VAEncPackedHeaderMiscMask va_deprecated_enum  = 0x80000000,
     ^~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:1063: libi965_drv_video_la-i965_drv_video.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/utku/Programlar/intel-vaapi-driver-1.8-branch/src'
make[2]: *** [Makefile:1341: all-recursive] Error 1
make[2]: Leaving directory '/home/utku/Programlar/intel-vaapi-driver-1.8-branch/src'
make[1]: *** [Makefile:654: all] Error 2
make[1]: Leaving directory '/home/utku/Programlar/intel-vaapi-driver-1.8-branch/src'
make: *** [Makefile:421: all-recursive] Error 1
utku@clr-449e9b2a44f8458bb4885604dc172a1c ~/Programlar/intel-vaapi-driver-1.8-branch $
</pre>

<pre>
utku@clr-449e9b2a44f8458bb4885604dc172a1c ~/Programlar/intel-vaapi-driver-1.8-branch $ echo $CFLAGS
-g -O1 -feliminate-unused-debug-types -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=32 -Wformat -Wformat-security -Wl,--copy-dt-needed-entries -m64 -fasynchronous-unwind-tables -Wp,-D_REENTRANT -ftree-loop-distribute-patterns -Wl,-z -Wl,now -Wl,-z -Wl,relro -fno-semantic-interposition -ffat-lto-objects -fno-signed-zeros -fno-trapping-math -fassociative-math -Wl,-sort-common
utku@clr-449e9b2a44f8458bb4885604dc172a1c ~/Programlar/intel-vaapi-driver-1.8-branch $ echo $CXXFLAGS
-g -O3 -feliminate-unused-debug-types -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector --param=ssp-buffer-size=32 -Wformat -Wformat-security -Wl,--copy-dt-needed-entries -m64 -fasynchronous-unwind-tables -Wp,-D_REENTRANT -ftree-loop-distribute-patterns -Wl,-z -Wl,now -Wl,-z -Wl,relro -fno-semantic-interposition -ffat-lto-objects -fno-signed-zeros -fno-trapping-math -fassociative-math -Wl,-sort-common -fvisibility-inlines-hidden
</pre>

uartie, 2017-11-28:
As @xhaihao pointed out, intel-vaapi-driver 1.8 is not compatible with libva 2.0 (VAAPI 1.0).  If you want to use libva 2.0, then you need to compile intel-vaapi-driver 2.0.  Otherwise, you need to use libva 1.8 with intel-vaapi-driver 1.8.

tuxutku, 2017-11-28:
i am trying to get libva.so.1

tuxutku, 2017-11-28:
and trying to compile 1.8 from 1.8 branch
messages are separate becouse i am texting from mobile

uartie, 2017-11-28:
Then you need libva 1.8 branch too.

tuxutku, 2017-11-28:
i don't understand 
libva.so.2 is allready exist, i assume its intel-vaapi-driver 2, the app i am compiled require libva.so.1 and i assume its intel-vaapi-driver 1.x (1.8 , 1.6 etc...),program didn't require libva.so.1 but after some updates it started to require libva.so.1 and i couldnt found the file, is intel-vaapi-driver 2 provide libav.so.1 too , or am i have to install 1.x
i couldn't compiled master branch and i assumed that i dont need to compile master branch any way  becouse it does not provide libva.so.1 anyway, then i cloned the 1.8 branch

note: i dont know how to get 1.8, and all i dont know if i have to install 1.x

tuxutku, 2017-11-28:
i don't know how to get libva.so.1 and i assume version 1.8 provides it, but i don't know. also libva.so.2 and libva*.so.2 files are exist and i assume newest version allready installed becouse i have the game and desktop bundless installed

xhaihao, 2017-11-28:
1. yes version 1.8 provides libva.so.1,
2. libva1 and libva2 can't coexist. Currently the include files for VAAPI was installed from libva2 on your system. I think you should re-install the development package for libva if you want to use version 1.8 or build libva version 1.8 from source code.

closed 2018-01-06 by tuxutku