intel/intel-vaapi-driver #307: Driver aborts when encoding H.264 at level 6 or higher

fhvwy, 2017-11-30:
Setting level_idc to 60 (for level 6) gives:
```
ffmpeg_g: i965_avc_encoder_common.c:226: i965_avc_get_max_mv_len: Assertion `0' failed.
Aborted
```

fritsch, 2017-12-13:
Level 6, 6.1 and 6.2 specify up to 8k surfaces. Currently all drivers are limited to a maximum of 4096 - does that need changing in above combination?

fhvwy, 2017-12-13:
@fritsch  The surface size constraint is already handled elsewhere - i965_CreateContext() blocks creating contexts for too-large streams.

(In any case, the 4096 limit is already insufficient for full support of level 4.2, which requires being able to handle a stream with width or height of up to 16\*floor(sqrt(8702\*8)) = 4208 pixels, and higher levels even bigger.  Only levels up to 4.1 are fully supported.)

commit 63e0375c (2018-01-10): Add support for H.264 levels 6, 6.1 and 6.2

Fixes #307.
closed 2018-01-10 by xhaihao