intel/intel-vaapi-driver #312: Low-power encoding does not support CBR

fthiery, 2017-12-06:
Using gstreamer-vaapi on an i3-6100; my application requires the performance offered by the low power encoding (fixed function), but also constant bitrate. The driver currently doesn't support CBR which is an issue for any HLS/DASH adaptive bitrate application, or any realtime communication oriented application which requires bitrate control.

Adding support for CBR with low-power encoding would be very useful.

https://github.com/01org/intel-vaapi-driver/blob/master/src/i965_device_info.c#L378

uartie, 2017-12-06:
https://github.com/01org/intel-vaapi-driver/blob/master/src/i965_device_info.c#L1052

fthiery, 2017-12-06:
Are you saying that it is actually doing CBR ?

uartie, 2017-12-06:
As long as the condition is met on L1051, then yes it should.  @xhaihao can probably give more details.

sreerenjb, 2017-12-06:
https://bugzilla.gnome.org/show_bug.cgi?id=789472#c7
Check cat /sys/kernel/debug/dri/0/i915_huc_load_status

uartie, 2017-12-06:
Can you verify if i915 HuC firmware is loaded?

Try setting `i915.enable_guc_loading=1` in the kernel boot command line.  You can see valid values with
`$ modinfo i915 | grep "enable_guc_loading"`

NOTE: GuC is required to be loaded for HuC to load and you'll need to ensure the firmware exists (`ls /lib/firmware/i915`)... https://01.org/linuxgraphics/downloads/firmware

Also look at `dmesg | egrep "drm|i915"` to see messages about GuC/HuC loading

fthiery, 2017-12-06:
Thanks, i managed to load the GuC with i915.enable_guc_loading=1, but not the HuC: i think my kernel is too old 

# uname -r
4.9.66-1-lts

# cat /sys/kernel/debug/dri/0/i915_guc_load_status
GuC firmware status:
	path: i915/skl_guc_ver6_1.bin
	fetch: SUCCESS
	load: SUCCESS
	version wanted: 6.1
	version found: 6.1
	header: offset is 0; size = 128
	uCode: offset is 128; size = 128640
	RSA: offset is 128768; size = 256

GuC status 0x800300ec:
	Bootrom status = 0x76
	uKernel status = 0x0
	MIA Core status = 0x3

Scratch registers:
	 0: 	0x0
	 1: 	0x0
	 2: 	0x0
	 3: 	0x5f5e100
	 4: 	0x600
	 5: 	0x0
	 6: 	0x0
	 7: 	0x8
	 8: 	0x11
	 9: 	0x0
	10: 	0x0
	11: 	0x0
	12: 	0x0
	13: 	0x0
	14: 	0x0
	15: 	0x700000

#  cat /sys/kernel/debug/dri/0/i915_huc_load_status
cat: /sys/kernel/debug/dri/0/i915_huc_load_status: No such file or directory

So what happens if i don't have the HuC loaded ? What is the bit rate control falling back to ?

uartie, 2017-12-06:
If no HuC, then you only get CQP https://github.com/01org/intel-vaapi-driver/blob/master/src/i965_device_info.c#L378

fthiery, 2017-12-06:
Many thanks, makes more sense now.  I'll upgrade my kernel and check

closed 2017-12-06 by uartie

fthiery, 2017-12-07:
Out of curiosity, why doesn't the following work ? Only passing the option on the boot cmdline seems to work for me:
# cat /etc/modprobe.d/i915.conf 
options i915.enable_guc_loading=1

ceyusa, 2017-12-07:
> Out of curiosity, why doesn't the following work ? Only passing the option on the boot cmdline seems to work for me:

because the format is 

```
options i915 enable_guc_loading=1 enable_guc_submission=1
```

fthiery, 2017-12-07:
(indeed...); but still no joy

```
$ cat /etc/modprobe.d/i915.conf 
options i915 enable_guc_loading=1 enable_guc_submission=1

$ sudo cat /sys/kernel/debug/dri/0/i915_huc_load_status
HuC firmware status:
	path: (null)
	fetch: NONE
	load: NONE
	version wanted: 0.0
	version found: 0.0
	header: offset is 0; size = 0
	uCode: offset is 0; size = 0
	RSA: offset is 0; size = 0

HuC status 0x00006000:
```

fthiery, 2017-12-07:
Looks like including i915 in my initrd was necessary. Strangely, not if using the boot cmdline parameter.

ceyusa, 2017-12-07:
strange. it works for me

just wrote a blog post about this issue

https://blogs.igalia.com/vjaquez/2017/12/07/enabling-huc-for-sklkbl-in-debiantesting/

fthiery, 2017-12-07:
Great, i did the same on the ArchLinux wiki.

Maybe i915 is included on debian in the initrd by default ?