intel/intel-vaapi-driver #315: VP9 decoding produces grey boxes (again)

maximbaz, 2018-01-06:
It looks like the issue #297 is back, when a video changes resolution, grey boxes start appearing, much like in #262. 

Just like in #297, it is reproduced when youtube changes video quality, but can also be reproduced with `mpv --hwdec=vaapi` by concatenating two videos of different resolution. [I made one such video](https://www.dropbox.com/s/hkl1rq3piwymzd2/vp9_concat_2_10.webm?dl=0), the merge happens somewhere around 02:10 (you have to download it to reproduce)

I'm not exactly sure updating what caused the bug, I reinstalled the OS and a few days later noticed that grey boxes are back.

Arch Linux, 4.14.11
HD Graphics 630

```
❯ vainfo
libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.0 (libva 2.0.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.0.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSlice
```

bwat47, 2018-01-07:
Can confirm here... I recall this was working after a libva-intel-driver update at some point, but now it's happening again. Running manjaro with libva-intel-driver 2.0.0-2

Kernel 4.14.11
Kaby Laky (Dell XPS 13 9360)

xhaihao, 2018-01-08:
You are using inter-vaapi-driver version 2.0.0 which doesn't fix this issue completely.  Could you try the master branch or apply the following commit to your code?

commit 9d66570032fb02b1e79a883af7697b035d700a8e
Author: Xiang, Haihao <haihao.xiang@intel.com>
Date:   Tue Oct 31 05:58:24 2017 +0800

    Add upper bound check in HCP_IND_OBJ_BASE_ADDR_STATE

maximbaz, 2018-01-08:
Oh my, I completely forgot that the 9d66570 didn't make it into the latest release, and when reinstalling OS I got the latest release 2.0.0 installed again. Sorry for the confusion, and yes, running the code from `master` fixes the issue!

Could you please consider releasing 2.0.1 with this fix? There was one other person who confirmed that the "grey boxes" recently returned, I think they also forgot that the fix is not in the official release yet.

closed 2018-01-08 by maximbaz