intel/intel-vaapi-driver #328: i3-7100U / no hardware acceleration of 2180p H.264 10-bit file.

38github, 2018-01-16:
I am having a problem getting vaapi to work with any Linux distro. I have an i3-7100U with HD 620 graphics and I have installed all libva, vaapi and intel packages in several distros and still mpv with hwdec=vaapi uses almost all CPU power. The latest I have tried it on is Ubuntu 18.04 and the following is what information I can give you about what I use to play the media.

```
$ vainfo
libva info: VA-API version 1.0.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.0 (libva 2.0.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.0.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
```

Video recorded with a Google Nexus 5X played using MPV:
```

$ mpv --hwdec=vaapi 2180p_h264_10bit.mp4
Playing: 2180p_h264_10bit.mp4
 (+) Video --vid=1 (*) (h264 3840x2160 28.751fps)
 (+) Audio --aid=1 --alang=eng (*) (aac 1ch 48000Hz)
AO: [pulse] 48000Hz mono 1ch float
VO: [opengl] 3840x2160 yuv420p
AV: 00:00:18 / 00:00:19 (98%) A-V:  0.000
 
 
Exiting... (End of file)
```

Do you have any idea why it does not work?

QuPengfei, 2018-01-16:
@38github, the VAAPI driver does not support AVC 10bit

38github, 2018-01-16:
Oh. Thank you for that information. Is it the same case with HEVC 10bit?

xhaihao, 2018-01-16:
HEVC 10bit decoding is supported on your KBL, please check the output of vainfo

> VAProfileHEVCMain10             : VAEntrypointVLD

closed 2018-01-16 by xhaihao

38github, 2018-01-16:
I reencoded some HEVC 10bit videos I have that did not use hardware acceleration but this time I (more) carefully configured the encode parameters in Handbrake and now the files are being accelerated. Thank you for your help! Weird though that AVC 10bit is hardware accelerated in Windows but not Linux but maybe Windows use another type of driver?

xhaihao, 2018-01-17:
Yes, widnows is using another driver.