intel/intel-vaapi-driver #342: vaExportSurfaceHandle possibly misinterprets surface size

pkerling, 2018-01-21:
Hi,

while investigating usage of the recently implemented `vaExportSurfaceHandle` in the Kodi media center we came across the problem that EGL import with `eglCreateImageKHR` of the surfaces decoded from HEVC10 failed with `EGL_BAD_ALLOC` in recent Mesa versions, while 8-bit HEVC and H.264 worked fine.

Testing the import of the Y plane of an example 4K (3840x2160) video decoded as P010, I traced the failure down to the following mesa code: https://github.com/mesa3d/mesa/blob/master/src/intel/isl/isl.c#L1325-L1335
It fails because the minimum pitch calculated is 15360, but the actual pitch is 7680 (which is 2*3840, i.e. the actual pitch is correct since the P010 Y plane has 2 bytes per pixel).

Now the wrong minimum pitch calculation seems to be a result of `intel-vaapi-driver` giving an erroneous result for the `width` and `height` fields of the `VADRMPRIMESurfaceDescriptor`, which I pass to `eglCreateImageKHR` as `EGL_WIDTH` and `EGL_HEIGHT`.
They are filled in here: https://github.com/01org/intel-vaapi-driver/blob/master/src/i965_drv_video.c#L6742-L6743
But `object_surface.width`is the *plane pitch in bytes* while it was my understanding that `VADRMPRIMESurfaceDescriptor.width` is the *surface width in *pixels*. Now the struct description does not explicitly spell this out, but at least the mesa vaapi drivers use pixels here, so it would be cool to agree on something.

Using `orig_width`/`orig_height` instead of `width`/`height` solves the problem for me.

xhaihao, 2018-01-22:
Yes, VADRMPRIMESurfaceDescriptor.width should be the surface width in pixels.

fhvwy, 2018-01-22:
Aside:  I don't think you want to use the surface width/height for this sort of use-case - you want the size of the image actually rendered on the surface, which may be smaller (especially when there is a cropping rectangle present).  This is probably why the problem wasn't noticed earlier - other programs using the API so far do not use these values because they only need the size of the image rendered on the surface, not the size of surface itself.

pkerling, 2018-01-22:
Thanks for the hint - under what conditions would that cropping occur? And are there downsides to just importing the whole surface to EGL using the parameters provided in VADRMPRIMESurfaceDescriptor?

If such a cropping rectangle means that the rendered image is placed at some location within the surface (i.e. it has an offset), wouldn't I still have to import the surface into EGL using the surface's full size and then actually use only part of it?

fhvwy, 2018-01-22:
The most common cropping case is when decoding video with height/width not divisible by the block size of the codec.  For example, in 1920x1080 video, 1080 is not divisible by 16 so in H.264 it is actually coded as 1920x1088 with eight lines of garbage data at the bottom of the frame.

As long as everything you do on the EGL side only uses the actual image region of the frame, then importing the whole surface to EGL should be ok.  But, note that even if you don't explicitly look outside the image the implementation may still be able to if any interpolation or filtering happens.

Unfortunately, there isn't currently any good answer for how to handle an offset from the top-left with GPU surfaces.  On the other hand, it is very rare - it can happen in H.264 and similar codecs, but not many decoders handle it properly so the feature is rarely used.  If you want to handle it fully then importing the whole surface and then being very careful which pixels you actually look at may well be the easiest method.

pkerling, 2018-01-22:
> But, note that even if you don't explicitly look outside the image the implementation may still be able to if any interpolation or filtering happens.

Fair point, but is setting `EGL_HEIGHT` (in the example) to 1080 instead of 1088 guaranteed to prevent that?

> Unfortunately, there isn't currently any good answer for how to handle an offset from the top-left with GPU surfaces. On the other hand, it is very rare

I'll happily ignore this then :-)

fhvwy, 2018-01-23:
> Fair point, but is setting EGL_HEIGHT (in the example) to 1080 instead of 1088 guaranteed to prevent that?

Yes - the GL implementation won't read outside the surface it knows exists.

closed 2018-01-31 by xhaihao

pkerling, 2018-01-31:
> Yes - the GL implementation won't read outside the surface it knows exists.

Cool -  I was worried it might use some side-channel information like it seems to do for the plane format modifiers.